esphome/issues #5598: Communication with INA226 failed!

OM-3kChcBjgFR2COfP, 2024-03-16:
### The problem

 Communication with INA226 failed errror

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.1

### What platform are you using?

ESP8266

### Board

NodeMCU lolin v3

### Example YAML snippet

```yaml
esphome:
  name: nodemcu-navien-ina226-pd100
  friendly_name: nodemcu-navien-ina226-pd100

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  password: "***"

wifi:
  networks:
  - ssid: "Keenetic-***"
    password: "***"
  - ssid: "Keenetic-***"
    password: "***"
  - ssid: "Keenetic-***"
    password: "***"

dallas:
  - pin: GPIO12
    update_interval: 120s

i2c:
  sda: D2
  scl: D1
  scan: true
  frequency: 400kHz // trying anything available
  id: bus_a

sensor:
  - platform: dallas
    address: 0x4e3ce10457af7b28
    name: "NAVIEN - 1"
  - platform: dallas
    address: 0x053ce104571e4428
    name: "NAVIEN - 2"
  - platform: ina226
    shunt_resistance: 0.1 ohm
    address: 0x40
    current:
      name: "INA226 Current"
      filters:
      - median:
          window_size: 5
          send_every: 5
          send_first_at: 1
    power:
      name: "INA226 Power"
    bus_voltage:
      name: "INA226 Bus Voltage"
    shunt_voltage:
      name: "INA226 Shunt Voltage"
    update_interval: 30s
```

### Anything in the logs that might be useful for us?

```txt
[09:17:58][C][logger:448]:   Level: DEBUG
[09:17:58][C][logger:449]:   Log Baud Rate: 115200
[09:17:58][C][logger:451]:   Hardware UART: UART0
[09:17:58][C][i2c.arduino:053]: I2C Bus:
[09:17:58][C][i2c.arduino:054]:   SDA Pin: GPIO4
[09:17:58][C][i2c.arduino:055]:   SCL Pin: GPIO5
[09:17:58][C][i2c.arduino:056]:   Frequency: 50000 Hz
[09:17:58][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[09:17:58][I][i2c.arduino:069]: Results from i2c bus scan:
[09:17:58][I][i2c.arduino:075]: Found i2c device at address 0x0C
[09:17:58][I][i2c.arduino:075]: Found i2c device at address 0x40
[09:17:58][C][dallas.sensor:075]: DallasComponent:
[09:17:58][C][dallas.sensor:076]:   Pin: GPIO12
[09:17:58][C][dallas.sensor:077]:   Update Interval: 120.0s
[09:17:58][W][dallas.sensor:080]:   Found no sensors!
[09:17:58][C][dallas.sensor:089]:   Device 'NAVIEN - 1'
[09:17:58][C][dallas.sensor:089]:     Device Class: 'temperature'
[09:17:58][C][dallas.sensor:089]:     State Class: 'measurement'
[09:17:58][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[09:17:58][C][dallas.sensor:089]:     Accuracy Decimals: 1
[09:17:58][C][dallas.sensor:097]:     Address: 0x4e3ce10457af7b28
[09:17:58][C][dallas.sensor:098]:     Resolution: 12
[09:17:58][C][dallas.sensor:089]:   Device 'NAVIEN - 2'
[09:17:58][C][dallas.sensor:089]:     Device Class: 'temperature'
[09:17:58][C][dallas.sensor:089]:     State Class: 'measurement'
[09:17:58][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[09:17:58][C][dallas.sensor:089]:     Accuracy Decimals: 1
[09:17:58][C][dallas.sensor:097]:     Address: 0x053ce104571e4428
[09:17:58][C][dallas.sensor:098]:     Resolution: 12
[09:17:58][C][ina226:081]: INA226:
[09:17:58][C][ina226:082]:   Address: 0x40
[09:17:58][E][ina226:085]: Communication with INA226 failed!
```

Masterz69, 2024-03-16:
Is it not working for 2 years (#3011), or on specific devices only.

OM-3kChcBjgFR2COfP, 2024-03-16:
> Is it not working for 2 years (#3011), or on specific devices only.

No, if i upload sketch from Arduino IDE to the same board - it works perfectly

closed 2024-08-23 by github-actions[bot]