esphome/issues #5600: IPv6 address text sensor not updated

Nardol, 2024-03-16:
### The problem

Some night ago, my ISP changed my IPv6/56 prefix.
My main TP-Link Deco M5 advertised a new /64 prefix and my device got a new address in this prefix.
But my `address_2` text sensor was not updated and still displays the old IPv6 address.

### Which version of ESPHome has the issue?

2024.3.0-dev (Mar 1 2024, 09:40:31)

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

wifi_info

### Example YAML snippet

```yaml
esphome:
  name: diffuseur-bureau

esp8266:
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key

network:
  enable_ipv6: true
  min_ipv6_addr_count: 2

text_sensor:
- platform: wifi_info
  ip_address:
    name: Adresse IPv4
    address_1:
      name: Adresse IPv6 locale
    address_2:
      name: Adresse IPv6 globale
```

### Additional information

Because (un?)fortunately, my prefix does not change often, I cannot wait enough to reproduce again and also test with an ESP32 and with the latest beta or dev version.
But when sending a ping the device is reachable at the new address so IPv6 works as expected, it looks it is only text sensors which are not updated.

HeMan, 2024-03-18:
I'll see if I can create a test environment where I can change prefixes, but since it's such a rare problem I'd go with the bad solution of turning it off and on again.

GoetzGoerisch, 2024-03-21:
Changing or dynamic IPv6-PD is a common thing for most residential and even businesses customers in Germany.

Therefore, this is a thing that needs to be considered.

GoetzGoerisch, 2024-03-24:
Inspired by the example above from @Nardol I have setup a textsensor as well.

``` YML
text_sensor:
- platform: wifi_info
  ip_address:
    name: Addresse IPv4
    address_1:
      name: IPv6 LL
    address_2:
      name: IPv6 GUA
    address_3:
      name: IPv6 ULA
```

Nevertheless after a reboot the link-local address is listed twice and therefore the text sensor IPv4 now contains the IPv6 LL address and so forth.

```
[08:10:40][C][wifi:416]:   IP Address: FE80::XXXX:XXFF:FE05:XXXX
[08:10:40][C][wifi:416]:   IP Address: 192.168.10.XXX
[08:10:40][C][wifi:416]:   IP Address: FE80::XXXX:XXFF:FE05:XXXX
[08:10:41][C][wifi:416]:   IP Address: 2A02:AAAA:BBBB:CCCC:XXXX:XXFF:FE05:XXXX
[08:10:41][C][wifi:416]:   IP Address: FD99:BBBB:CCCC:DDDD:XXXX:XXFF:FE05:XXXX
```

Nardol, 2024-07-23:
It should not be closed :slightly_smiling_face:

schildbach, 2026-01-16:
I think you cannot expect IPv6 addresses to remain in a specific order, or stay limited to a specific number. It's highly dynamic, due to the presence of various configuration protocols in the IPv6 space.

Likely if you want e.g. to show your GUA ESPHome would need some selector just for that.