esphome/issues #5601: Enabling IPv6 not possible with voice assistant

Nardol, 2024-03-16:
### The problem

I have set up a speaker with voice assistant, which works as fare as possible when I don't enable Ipv6.
But when enabling Ipv6, API connection is reset, making impossible to use the speaker.

### Which version of ESPHome has the issue?

2024.3.0b2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

network, wifi, voice_assistant

### Example YAML snippet

```yaml
---
esphome:
  name: enceinte-test
  friendly_name: "Enceinte test"
  area: Bureau
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.3.0b2

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:

api:

ota:
  password: !secret enceinte_test_ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
  - platform: restart
    id: restart_btn
    name: Restart

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    mode: mono

voice_assistant:
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 4
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_client_connected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.start_continuous:
  on_client_disconnected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.stop:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO39
      inverted: true
    name: Button
    disabled_by_default: true
    entity_category: diagnostic
    id: echo_button
    on_multi_click:
      - timing:
          - ON for at least 250ms
          - OFF for at least 50ms
        then:
          - if:
              condition:
                switch.is_off: use_wake_word
              then:
                - if:
                    condition: voice_assistant.is_running
                    then:
                      - voice_assistant.stop:
                    else:
                      - voice_assistant.start:
              else:
                - voice_assistant.stop
                - voice_assistant.start_continuous:
      - timing:
          - ON for at least 5s
        then:
          - button.press: restart_btn
      - timing:
          - ON for at least 10s
        then:
          - button.press: factory_reset_btn

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);

network:
  enable_ipv6: true
  #min_ipv6_addr_count: 2

text_sensor:
- platform: wifi_info
  ip_address:
    name: IPv4 address
    address_1:
      name: Local IPv6 address
    address_2:
      name: Global IPv6 address

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s

esp_adf:
```

### Additional information

When connecting with esphome logs, I regularly have connection reset by peer then it reconnects again and again...

HeMan, 2024-03-18:
It could be a memory issue. Could you by any chance connect it via serial/usb and see what the logs says there.

Nardol, 2024-03-18:
Yes I can, which log level would be the best for you and which components do you recommend me to let being more verbose?

In a first try I'm going to test in ERROR level to see if something could help.

Nardol, 2024-07-18:
I have made a try and I can now use IPv6 as expected. Experience" with my M5 stake does not become worse but it is simply because it cannot be worse :slightly_smiling_face: and I have the two expected addresses assigned.

closed 2024-07-18 by Nardol