esphome/issues #5602: AttributeError: 'NoneType' object has no attribute 'replace'

jclsn, 2024-03-16:
### The problem

Trying to compile a switch.yaml component and am getting. Had to take the dev version because of https://github.com/esphome/issues/issues/5486 and now this occurs.

```
INFO ESPHome 2024.4.0-dev
INFO Reading configuration speaker-switch.yaml...
WARNING 'lautsprecher_drehen': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome==2024.4.0.dev0', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 1061, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 1048, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 478, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/usr/lib/python3.11/site-packages/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/lib/python3.11/site-packages/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/usr/lib/python3.11/site-packages/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/usr/lib/python3.11/site-packages/esphome/components/gpio/switch/__init__.py", line 27, in to_code
    var = await switch.new_switch(config)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/components/switch/__init__.py", line 169, in new_switch
    await register_switch(var, config)
  File "/usr/lib/python3.11/site-packages/esphome/components/switch/__init__.py", line 164, in register_switch
    await setup_switch_core_(var, config)
  File "/usr/lib/python3.11/site-packages/esphome/components/switch/__init__.py", line 139, in setup_switch_core_
    await setup_entity(var, config)
  File "/usr/lib/python3.11/site-packages/esphome/cpp_helpers.py", line 106, in setup_entity
    add(var.set_object_id(sanitize(snake_case(CORE.friendly_name))))
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/esphome/helpers.py", line 362, in snake_case
    return value.replace(" ", "_").lower()
           ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'replace'
```

### Which version of ESPHome has the issue?

esphome-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

ssieb, 2024-03-16:
That should give a nicer validation error instead of crashing.  I think the problem is that your switch name is empty and you don't have the friendly name set.

jclsn, 2024-03-16:
@ssieb Ah yes, thank you! It compiles now

ssieb, 2024-03-16:
I'll keep this open.  That should give a validation error instead of crashing.