esphome/issues #5603: Senseair S8 unsigned ppm error

MallocArray, 2024-03-16:
### The problem

Similar to [#1899](https://github.com/esphome/issues/issues/1899) if the Senseair S8 reports a number less than 0, instead of a negative number, it jumps up to something like 65535 

A negative number isn't really an accurate reading, but neither is 65535 

As mentioned in the linked issue, it looks to be due to using uint16 instead of int16 for the ppm.

Either way, the user should trigger a manual calibration to get the readings more accurate, but a proper negative reading would be more logical than overflowing up to ~65535

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

senseair

### Example YAML snippet

```yaml
- platform: senseair
    # SenseAir S8 https://esphome.io/components/sensor/senseair.html
    # https://senseair.com/products/size-counts/s8-lp/
    co2:
      name: "CO2"
      id: co2
    id: senseair_s8
    uart_id: senseair_s8_uart
```
```

### Anything in the logs that might be useful for us?

```txt
[21:09:02][D][senseair:059]: SenseAir Received CO₂=65519ppm Status=0x00
[21:09:02][D][sensor:094]: 'CO2': Sending state 65519.00000 ppm with 0 decimals of accuracy
```

ssieb, 2024-03-16:
Please test the PR.

MallocArray, 2024-03-16:
I don't know how to have my local docker instance use this PR
I'm also unable to reproduce the Senseair S8 to report a negative number

ssieb, 2024-03-16:
> I don't know how to have my local docker instance use this PR

https://esphome.io/components/external_components

> I'm also unable to reproduce the Senseair S8 to report a negative number at this time

That's what I expected.  But at least verify that it doesn't break it.

MallocArray, 2024-03-16:
I was able to test it successfully, I think.  

I ended up with the device in my home where ppm is currently 1300 and then forced a calibration which reset it to think that was 400.
I then took the unit outside, where it is actually 400-500 which should force it to read around -900

This is what I see in the debug logs for senseair
```
[17:20:06][D][senseair:059]: SenseAir Received CO₂=0ppm Status=0x20
[17:20:06][D][sensor:093]: 'CO2': Sending state 0.00000 ppm with 0 decimals of accuracy
```
So it wasn't a negative number, but it also didn't overflow.  Is there maybe an implicit min_value of 0 for this sensor?

Either way, this appears to have addressed the issue.

ssieb, 2024-03-16:
Please try again.  Make sure you set the refresh time on the external component to something small so that it will pick up the changes.

MallocArray, 2024-03-17:
Looking good:
[19:40:58][D][senseair:059]: SenseAir Received CO₂=-187ppm Status=0x20
[19:40:58][D][sensor:093]: 'CO2': Sending state -187.00000 ppm with 0 decimals of accuracy

closed 2024-03-17 by jesserockz