esphome/issues #5604: Inkplate 10 partial update doesn't work

xaxix, 2024-03-17:
### The problem

My Inkplate 10 does not perform partial updates. A complete screen refresh is performed every time.

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Never worked

### What platform are you using?

ESP32

### Board

Inkplate 10 (pca6416a)

### Example YAML snippet

```yaml
display:
- platform: inkplate6
  id: inkplate_display_10
  model: inkplate_10
  greyscale: false
  partial_updating: true
  update_interval: 60s
  full_update_every: 30
  rotation: 0°

  ckv_pin: 32
  sph_pin: 33
  gmod_pin:
    pca6416a: pca6416a_hub
    number: 1
  gpio0_enable_pin:
    pca6416a: pca6416a_hub
    number: 8
  oe_pin:
    pca6416a: pca6416a_hub
    number: 0
  spv_pin:
    pca6416a: pca6416a_hub
    number: 2
  powerup_pin:
    pca6416a: pca6416a_hub
    number: 4
  wakeup_pin:
    pca6416a: pca6416a_hub
    number: 3
  vcom_pin:
    pca6416a: pca6416a_hub
    number: 5
```

xaxix, 2024-07-16:
Any updates here?

solarkennedy, 2024-10-23:
I have an inkplate 10 and confirm that it is partially updating.
I just wrote a blog post with it and it includes the whole yaml for reference: https://kyle.cascade.family/posts/a-baby-eink-dashboard/

Can you reflash with `VERBOSE` logging?
```
logger:
  level: VERBOSE
```

Then can you grep the logs for _any_ reference to `partial` to see if any of the code paths are being hit?
```
$ esphome logs inkplate.yaml | grep -i partial
[13:07:59][V][inkplate:516]: Partial update called
[13:08:00][V][inkplate:539]: Partial update buffer built after (52ms)
[13:08:00][V][inkplate:565]: Partial update loop k=0 (155ms)
[13:08:00][V][inkplate:565]: Partial update loop k=1 (203ms)
[13:08:00][V][inkplate:565]: Partial update loop k=2 (252ms)
[13:08:00][V][inkplate:565]: Partial update loop k=3 (299ms)
[13:08:00][V][inkplate:565]: Partial update loop k=4 (348ms)
[13:08:00][V][inkplate:573]: Partial update finished (659ms)
[13:08:00][V][inkplate:313]: Display finished (partial) (665ms)
```

Can you also grep for "Display finished" to verify if it really doing `full` updates every time?
Mine looks like this:
```
$ esphome logs inkplate.yaml | grep -i "Display finished"
...
[13:10:22][V][inkplate:313]: Display finished (partial) (668ms)
[13:10:30][V][inkplate:313]: Display finished (partial) (670ms)
[13:10:40][V][inkplate:313]: Display finished (partial) (669ms)
[13:10:50][V][inkplate:313]: Display finished (partial) (668ms)
[13:11:00][V][inkplate:313]: Display finished (partial) (667ms)
[13:11:10][V][inkplate:313]: Display finished (partial) (666ms)
[13:11:20][V][inkplate:313]: Display finished (partial) (665ms)
```

solarkennedy, 2024-10-23:
Probably the same as https://github.com/esphome/issues/issues/4730

xaxix, 2024-10-24:
Thanks to your input I was able to find the problem. Thank you very much! It was the following lines. After I commented them out it works.
```
#  - platform: template
#    name: "Inkplate Partial Updating"
#    lambda: return id(inkplate_display_10).get_partial_updating();
#    turn_on_action:
#      - lambda: id(inkplate_display_10).set_partial_updating(true);
#    turn_off_action:
#      - lambda: id(inkplate_display_10).set_partial_updating(false);
```

closed 2024-10-24 by xaxix