esphome/issues #5605: ESP32-S3 devboard get warnings about microphone and microphone is not working

n00btotal, 2024-03-17:
### The problem

When installing the YAML i get this:

`INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/voicer-two.yaml...
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-tflite-micro@None
INFO Compiling app...
Processing voicer-two (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/voicer-two/src/esphome/components/api/api_connection.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/api_server.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/list_entities.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/proto.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/voicer-two/src/esphome/components/api/user_services.o
Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/automation.o
Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/binary_sensor.o
Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/filter.o
Compiling .pioenvs/voicer-two/src/esphome/components/captive_portal/captive_portal.o
Compiling .pioenvs/voicer-two/src/esphome/components/esp32/core.o
Compiling .pioenvs/voicer-two/src/esphome/components/esp32/gpio.o
Compiling .pioenvs/voicer-two/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/voicer-two/src/esphome/components/esp32_rmt_led_strip/led_strip.o
Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/i2s_audio.o
Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.o
Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.o
Compiling .pioenvs/voicer-two/src/esphome/components/light/addressable_light.o
Compiling .pioenvs/voicer-two/src/esphome/components/light/automation.o
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp: In member function 'void esphome::i2s_audio::I2SAudioMicrophone::start_()':
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
   };
   ^
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
Compiling .pioenvs/voicer-two/src/esphome/components/light/esp_color_correction.o
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp: In static member function 'static void esphome::i2s_audio::I2SAudioSpeaker::player_task(void*)':
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
   };
   ^
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]`

When I run the code the microphone part is not working. (Almost except for some pins) identical code does works in an ordinary ESP32 (not S3 devboard).

### Which version of ESPHome has the issue?
[... 196 lines cut ...]
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp: In member function 'void esphome::i2s_audio::I2SAudioMicrophone::start_()':
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
   };
   ^
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
Compiling .pioenvs/voicer-two/src/esphome/components/light/esp_color_correction.o
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp: In static member function 'static void esphome::i2s_audio::I2SAudioSpeaker::player_task(void*)':
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
   };
   ^
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
```

fantnhu, 2024-03-20:
> ### A probléma
> A YAML telepítésekor az alábbi hibákat kapom a mikrofonnal és a hangszórókkal kapcsolatban. Csak ESP32-S3 devboardon. A mikrofon nem működik. "Ugyanaz" kód az ESP32 devboardon működik.
>

Hi! Before installing, press "Clean build Files" in ESPHome

BigBobbas, 2024-03-21:
change the channel for the microphone to left. it works set to right on a standards esp32 but should be set to left on and esp32 - s3 , the warnings aren't anything to worry about.

pimp1310, 2024-04-11:
> change the channel for the microphone to left. it works set to right on a standards esp32 but should be set to left on and esp32 - s3 , the warnings aren't anything to worry about.

i have the same error with channel : left and a GND bridge to L/R

`microphone:
  - platform: i2s_audio
    id: va_mic
    adc_type: external
    i2s_din_pin: GPIO4 #SD pin on the INMP441
    channel: left
    pdm: false
    i2s_audio_id: i2s_in
    bits_per_sample: 32 bit`

pimp1310, 2024-04-11:
> ### The problem
> When installing the YAML i get these errors below regarding microphone and speakers. Only on ESP32-S3 devboard. Microphone is not working. "Same" code on ESP32 devboard works.
> 
> ```
> INFO ESPHome 2024.2.2
> INFO Reading configuration /config/esphome/voicer-two.yaml...
> WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> INFO Generating C++ source...
> INFO Updating https://github.com/espressif/esp-tflite-micro@None
> INFO Compiling app...
> Processing voicer-two (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
> --------------------------------------------------------------------------------
> HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
>  - framework-espidf @ 3.40406.240122 (4.4.6) 
>  - tool-cmake @ 3.16.9 
>  - tool-ninja @ 1.10.2 
>  - toolchain-esp32ulp @ 2.35.0-20220830 
>  - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
>  - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
> Reading CMake configuration...
> Generating assembly for certificate bundle...
> Dependency Graph
> |-- noise-c @ 0.1.4
> Compiling .pioenvs/voicer-two/src/esphome/components/api/api_connection.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/api_frame_helper.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/api_pb2.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/api_pb2_service.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/api_server.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/list_entities.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/proto.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/subscribe_state.o
> Compiling .pioenvs/voicer-two/src/esphome/components/api/user_services.o
> Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/automation.o
> Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/binary_sensor.o
> Compiling .pioenvs/voicer-two/src/esphome/components/binary_sensor/filter.o
> Compiling .pioenvs/voicer-two/src/esphome/components/captive_portal/captive_portal.o
> Compiling .pioenvs/voicer-two/src/esphome/components/esp32/core.o
> Compiling .pioenvs/voicer-two/src/esphome/components/esp32/gpio.o
> Compiling .pioenvs/voicer-two/src/esphome/components/esp32/preferences.o
> Compiling .pioenvs/voicer-two/src/esphome/components/esp32_rmt_led_strip/led_strip.o
> Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/i2s_audio.o
> Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.o
> Compiling .pioenvs/voicer-two/src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.o
> Compiling .pioenvs/voicer-two/src/esphome/components/light/addressable_light.o
> Compiling .pioenvs/voicer-two/src/esphome/components/light/automation.o
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp: In member function 'void esphome::i2s_audio::I2SAudioMicrophone::start_()':
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
>    };
>    ^
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:62:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
> Compiling .pioenvs/voicer-two/src/esphome/components/light/esp_color_correction.o
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp: In static member function 'static void esphome::i2s_audio::I2SAudioSpeaker::player_task(void*)':
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
>    };
>    ^
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
[... cut ...]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
> src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:56:3: warning: missing initializer for member 'i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
> ```
> 
> ### Additional information
> _No response_

get you your mic to work?
i have exact the same problem and my mic is not working

FaGerstenlauer, 2024-06-15:
I have the same problem.

Has anyone solved the problem?

mistikdonbass, 2024-07-23:
I join you, the same mistake

BigBobbas, 2024-07-23:
do you have the latest version of ESPHome installed? 2024.7.1 ?  if you are on an older version, update ESPHome and try again. Also carry out a 'clean build files' to clear any existing build files from previous attempts.

MrDogYar, 2024-09-09:
I join you, the same mistake