esphome/issues #5608: Grow fingerprint sensor loses communication

Nadav-K, 2024-03-18:
### The problem

I have a "Grow" fingerprint sensor (R307/AS608/FPM10A) which is coupled with an ESP32-C3 Super Mini.
It's wired so that T-out goes to GPIO10, works fine most of the time.

Once every few hours/days the fingerprint sensor will stop responding to the ESP board. Setting VV logging I see that once I place my finger the board prints "No response received from reader". snipper below. any idea why this could happen?

I wanted to suggest adding an automation trigger in such a case, so that a recovery flow can be initiated from HomeAssistant or even the board itself..

```

[19:44:30][C][logger:447]: Logger:
[19:44:30][C][logger:448]:   Level: VERY_VERBOSE
[19:44:30][C][logger:449]:   Log Baud Rate: 115200
[19:44:30][C][logger:451]:   Hardware UART: USB_SERIAL_JTAG
[19:44:30][C][uart.idf:139]: UART Bus 0:
[19:44:30][C][uart.idf:140]:   TX Pin: GPIO21
[19:44:30][C][uart.idf:141]:   RX Pin: GPIO20
[19:44:30][C][uart.idf:143]:   RX Buffer Size: 256
[19:44:30][C][uart.idf:145]:   Baud Rate: 57600 baud
[19:44:30][C][uart.idf:146]:   Data Bits: 8
[19:44:30][C][uart.idf:147]:   Parity: NONE
[19:44:30][C][uart.idf:148]:   Stop bits: 1
[19:44:30][C][template.text_sensor:020]: Template Sensor 'Fingerprint State'
[19:44:30][C][status:034]: Status Binary Sensor '${friendly_name} Status'
[19:44:30][C][status:034]:   Device Class: 'connectivity'
[19:44:30][C][fingerprint_grow:537]: GROW_FINGERPRINT_READER:
[19:44:30][C][fingerprint_grow:538]:   System Identifier Code: 0x0000
[19:44:30][C][fingerprint_grow:540]:   Touch Sensing Pin: GPIO10
[19:44:30][C][fingerprint_grow:542]:   Sensor Power Pin: None
[19:44:30][C][fingerprint_grow:546]:   Idle Period to Sleep: Never
[19:44:30][C][fingerprint_grow:548]:   Update Interval: 0.8s
[19:44:30][C][fingerprint_grow:550]:   Fingerprint Count 'Fingerprint Count'
[19:44:30][C][fingerprint_grow:550]:     State Class: ''
[19:44:30][C][fingerprint_grow:550]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:550]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:550]:     Icon: 'mdi:fingerprint'
[19:44:30][C][fingerprint_grow:551]:     Current Value: 4
[19:44:30][C][fingerprint_grow:554]:   Status 'Fingerprint Status'
[19:44:30][C][fingerprint_grow:554]:     State Class: ''
[19:44:30][C][fingerprint_grow:554]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:554]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:555]:     Current Value: 0
[19:44:30][C][fingerprint_grow:558]:   Capacity 'Fingerprint Capacity'
[19:44:30][C][fingerprint_grow:558]:     State Class: ''
[19:44:30][C][fingerprint_grow:558]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:558]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:558]:     Icon: 'mdi:database'
[19:44:30][C][fingerprint_grow:559]:     Current Value: 150
[19:44:30][C][fingerprint_grow:562]:   Security Level 'Fingerprint Security Level'
[19:44:30][C][fingerprint_grow:562]:     State Class: ''
[19:44:30][C][fingerprint_grow:562]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:562]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:562]:     Icon: 'mdi:security'
[19:44:30][C][fingerprint_grow:563]:     Current Value: 3
[19:44:30][C][fingerprint_grow:566]:   Last Finger ID 'Fingerprint Last Finger ID'
[19:44:30][C][fingerprint_grow:566]:     State Class: ''
[19:44:30][C][fingerprint_grow:566]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:566]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:566]:     Icon: 'mdi:account'
[19:44:30][C][fingerprint_grow:567]:     Current Value: 1
[19:44:30][C][fingerprint_grow:570]:   Last Confidence 'Fingerprint Last Confidence'
[19:44:30][C][fingerprint_grow:570]:     State Class: ''
[19:44:30][C][fingerprint_grow:570]:     Unit of Measurement: ''
[19:44:30][C][fingerprint_grow:570]:     Accuracy Decimals: 0
[19:44:30][C][fingerprint_grow:570]:     Icon: 'mdi:account-check'
[19:44:30][C][fingerprint_grow:571]:     Current Value: 226
[19:44:30][C][mdns:115]: mDNS:
[19:44:30][C][mdns:116]:   Hostname: esp32newc3
[19:44:30][V][mdns:117]:   Services:
[... 421 lines cut ...]
[19:44:57][D][fingerprint_grow:125]: Scan and match
[19:44:57][D][fingerprint_grow:156]: Getting image 1
[19:44:58][E][fingerprint_grow:467]: No response received from reader
[19:44:58][D][fingerprint_grow:179]: Unknown Scan Error: 255
[19:44:58][W][component:214]: Component fingerprint_grow took a long time for an operation (2.02 s).
[19:44:58][W][component:215]: Components should block for at most 20-30ms.
[19:44:58][VV][scheduler:226]: Running interval '' with interval=5000 last_execution=86950275 (now=86955276)
[19:44:58][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86954740 (now=86957315)
[19:44:58][V][fingerprint_grow:021]: No touch sensing
[19:44:58][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86956990 (now=86957745)
[19:44:58][V][fingerprint_grow:021]: No touch sensing
[19:44:59][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86957740 (now=86958490)
[19:44:59][V][fingerprint_grow:021]: No touch sensing
[19:45:00][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86958490 (now=86959243)
[19:45:00][V][fingerprint_grow:021]: No touch sensing
[19:45:01][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86959240 (now=86959990)
[19:45:01][V][fingerprint_grow:021]: No touch sensing
[19:45:01][VV][scheduler:226]: Running interval '' with interval=5000 last_execution=86955275 (now=86960278)
[19:45:01][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86959990 (now=86960741)
[19:45:01][V][fingerprint_grow:021]: No touch sensing
[19:45:02][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86960740 (now=86961493)
[19:45:02][V][fingerprint_grow:021]: No touch sensing
[19:45:03][VV][scheduler:226]: Running interval 'update' with interval=750 last_execution=86961490 (now=86962240)
[19:45:03][V][fingerprint_grow:021]: No touch sensing
```

alexborro, 2024-03-20:
Hi @Nadav-K ,
That is an interesting issue.
First I will ask you to update the esphome version to the latest one (`2024.3.0` was just released a few hours ago). Remove the `external_components` session from your yaml file - you no longer need to take it from my branch.

Second, this board you are using has only one pair of UART, I'm not sure if there could by any conflict with the esphome internal logger. So this would be my next recommendation, to use another board if possible. I tested this component with a ESP32-WROOM based board and it works fine.

A third tip is to use the new [Sleep Mode](https://esphome.io/components/fingerprint_grow#sleep-mode) integrated in this latest version. It basically powers off the sensor when not being used. That forces a reset in the fingerprint device when you touch its sensor - might help getting rid of the serial comm problem. 

Let me know if I can help you with some further debugging.

tlissak, 2024-05-06:
Hi 

I found the solution 
The problem is the pin wiring issue 
Change the wiring as bellow :

....
uart:
  tx_pin: GPIO16 #Purple Green (GPIO16=D0)
  rx_pin: GPIO5 #Yellow (GPIO5=D1)
  baud_rate: 57600
....
status_led:
  pin: GPIO2 #D4 Empty No pin
....
fingerprint_grow:
  sensing_pin: GPIO4 #Blue (GPIO4=D2)