esphome/issues #5610: KSZ8081RNA missing

matwho, 2024-03-20:
### The problem

After upgrade from 2024.2.3 to 2024.3.0 Ethernet type KSZ8081RNA is missing with an error “Unknown value 'KSZ8081RNA', valid options are
'LAN8720', 'RTL8201', 'DP83848', 'IP101', 'JL1101',
' W5500'.”

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Womolin TI-CI-LIN Controller v2.1

### Component causing the issue

Ethernet

### Example YAML snippet

```yaml
ethernet:
  type: KSZ8081RNA
  mdc_pin: 23
  mdio_pin: 18
  clk_mode: GPIO16_OUT
```

### Anything in the logs that might be useful for us?

```txt
Unknown value 'KSZ8081RNA', valid options are
'LAN8720', 'RTL8201', 'DP83848', 'IP101', 'JL1101',
' W5500'.
```

matwho, 2024-03-28:
I think the problem was caused by esphome/esphome#4424

ssieb, 2024-03-28:
Try adding the linked PR as an external component.

closed 2024-03-28 by jesserockz