esphome/issues #5614: 2024.3.0 fails to build with wired ethernet and esp-idf 5.1.2

utemkin, 2024-03-20:
### The problem

After upgrade 2024.2.2 -> 2024.3.0 my project fails to build

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.1

### What platform are you using?

ESP32-IDF

### Board

wt32-eth01

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
esp32:
  board: wt32-eth01
  framework:
    type: esp-idf
    version: 5.1.2
    platform_version: 6.5.0

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/light/src/esphome/components/ethernet/ethernet_component.o
src/esphome/components/ethernet/ethernet_component.cpp: In member function 'virtual void esphome::ethernet::EthernetComponent::setup()':
src/esphome/components/ethernet/ethernet_component.cpp:191:26: error: 'ESP_MAC_ETH' was not declared in this scope
  191 |   esp_read_mac(mac_addr, ESP_MAC_ETH);
      |                          ^~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:191:3: error: 'esp_read_mac' was not declared in this scope
  191 |   esp_read_mac(mac_addr, ESP_MAC_ETH);
      |   ^~~~~~~~~~~~
*** [.pioenvs/light/src/esphome/components/ethernet/ethernet_component.o] Error 1
```

randybb, 2024-03-20:
ESPHome is using esp-idf 4.4 just for this reason - many components are not compatible with 5.0. And since you are overriding ESP-IDF version, this is not an issue of the latest version of ESPHome. It will be fixed, but we are not ready yet.

closed 2024-03-20 by randybb

utemkin, 2024-03-20:
Any ETA on switching default ESP_IDF to 5.x?