esphome/issues #5618: IPv6 implementation does not comply to RFC 7217 and 7721

GoetzGoerisch, 2024-03-21:
### The problem

The newly integrate and now working IPv6 implemenation generates its IID based on the MAC of the device which is considered to bad from a Security and Privacy standpoint.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.1

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

network

### Example YAML snippet

```yaml
esp8266:
  board: esp01_1m
  framework:
    version: recommended
    platform_version: 4.2.1

network:
    enable_ipv6: true
```

### Anything in the logs that might be useful for us?

```txt
[wifi:416]:   IP Address: FE80::AAAA:BBFF:FECC:DDDD
[wifi:416]:   IP Address: 2001:DB8::AAAA:BBFF:FECC:DDDD
][wifi:416]:   IP Address: FD00::AAAA:BBFF:FECC:DDDD
```

### Additional information

https://datatracker.ietf.org/doc/rfc7721/
https://datatracker.ietf.org/doc/rfc7217/

HeMan, 2024-03-21:
LwIP, which is the IP-stack used here, doesn't have privacy extension, and there seem to be no recent discussion about it either. I found a question from 2022, https://lists.nongnu.org/archive/html/lwip-users/2022-08/msg00007.html, but nothing more.

GoetzGoerisch, 2024-03-21:
Then at least this deviation should be clearly documented as this might have consequences for the infrastructure and the users privacy.

Source is here, if I'm not mistaken?
[network.rst](https://github.com/esphome/esphome-docs/blob/current/components/network.rst)

I would propose a PR is this is something you would merge?

closed 2024-03-21 by GoetzGoerisch