esphome/issues #5624: ESPHome Wake-on-LAN Button component creates erroneous error logs

kixtarter, 2024-03-22:
### The problem

The ESPHome Wake-on-LAN Button component creates an error log even when the magic packet was successfully sent.
This appears to be due to a bug in the logging code:
Current code:
```
  if (!begin_status || end_status) {
    ESP_LOGE(TAG, "Sending Wake-on-LAN Packet Failed!");
  }
```
This should be:
```
  if (!begin_status || !end_status) {
    ESP_LOGE(TAG, "Sending Wake-on-LAN Packet Failed!");
  }
```
For more background info see this community post: [Send WOL (Wake on LAN) magic packets from ESP8266](https://community.home-assistant.io/t/send-wol-wake-on-lan-magic-packets-from-esp8266/686295/8)

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.1

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini V4

### Example YAML snippet

```yaml
button:
  - platform: wake_on_lan
    name: "Start the HANUC"
    icon: mdi:lan-connect
    target_mac_address: XX:XX:XX:XX:XX:XX #redacted
```

### Anything in the logs that might be useful for us?

```txt
[13:22:46][D][button:010]: 'Start the HANUC' Pressed.
[13:22:46][I][wake_on_lan.button:030]: Sending Wake-on-LAN Packet...
[13:22:46][E][wake_on_lan.button:054]: Sending Wake-on-LAN Packet Failed!
```

### Additional information

Although this currently is only reported and tested for the ESP8266 platform it most probably is the same for the ESP32 platform.

plplaaa2, 2024-03-22:
There's another problem with wake on lan. Wake on lan doesn't work at all on libretiny esphome. It works well on ESP8266 with the same code. It seems that wake on lan is not supported yet on bk72xx.

kixtarter, 2024-03-23:
> There's another problem with wake on lan. Wake on lan doesn't work at all on libretiny esphome

This problem is unrelated to the logging problem, so you better create a separate issue or feature request for this.

palves, 2024-07-29:
Comment for activity.  (I ran into this a while back and got very confused then.)