esphome/issues #5626: AC Dimmer boot loop after update to 2024.3.0

leelists, 2024-03-22:
### The problem

The following script enter a boot loop with ESPHome 2024.3.0,
a rollback to ESPHome 2024.2.2 fix the issue.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.2

### What platform are you using?

ESP8266

### Board

d1mini

### Component causing the issue

ac_dimmer

### Example YAML snippet

```yaml
esphome:
  name: vmc-cave
  friendly_name: vmc_cave
  platformio_options:
    board_build.f_cpu: 160000000L  

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_pass

ota:
  password:  !secret ota_pass

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_pass
    priority: 2
  - ssid: !secret wifi2_ssid
    password: !secret wifi2_pass
    priority: 1
  output_power: 10
  
status_led:
  pin:
    number: GPIO2
    inverted: true

captive_portal:

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D7
    zero_cross_pin:
      number: D6
      mode:
        input: true
      inverted: yes
    min_power: 0.1
    zero_means_zero: true  

fan:
  - platform: speed
    output: dimmer1
    name: "VMC Speed"  
    speed_count: 5
```

### Anything in the logs that might be useful for us?

```txt
Unable to connect remotely to recover logs
```

ssieb, 2024-03-25:
What is the device?  Can you get serial logs?

leelists, 2024-03-26:
the device is an d1 mini (8266) with a [robbotdyn](https://www.amazon.com/GENUINE-RobotDYN-Programmable-Controller-Compatible/dp/B071X19VL1) ac dimmer.

unfortunately the device is embedded in difficult to access device so a don't have serial access to it, i will try the script on a naked d1mini and see if i can reproduce the error

closed 2024-08-15 by github-actions[bot]