esphome/issues #5627: Не удалось вызвать службу update/install. Error updating esphome-web-e3b81b.yaml via OTA; Try again in ESPHome dashboard for more information.

mr-aleks, 2024-03-23:
### The problem

The update notification constantly appears, although the latest version is already installed.
[image]
[image]
Error after:
[image]

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

Wemos mini

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-e3b81b
  friendly_name: ESPHome Web e3b81b

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Gs2zgbCxsHrkPo3xJ8ADa2h33F5MItbMksKdSnSL2fc="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Esphome-Web-E3B81B"
    password: "S6MOrUTGMTm7"

captive_portal:

# Example configuration entry
# Enable fallback hotspot (captive portal) in case wifi connection fails

i2c:
  sda: 4
  scl: 5
  scan: True

sensor:
  - platform: homeassistant
    id: sea_level
    entity_id: sensor.pogodaiklimat_pressure_at_sea_level
    internal: true
  - platform: homeassistant
    id: xiaomi2_temp
    entity_id: sensor.temperature_humidity_sensor_1306_temperature
    internal: true    
  - platform: aht10
    variant: AHT20
    temperature:
      name: "AHT20 Temperature"
    humidity:
      name: "AHT20 Humidity"
    update_interval: 60s

  - platform: bmp280
    temperature:
      name: "BMP280 Temperature"
      id: bmp280_temperature
      oversampling: 16x 
    pressure:
      name: "BMP280 Pressure"
      id: bmp280_pressure
    address: 0x77
    update_interval: 60s

  - platform: template
    name: "BMP280 Altitude"
    lambda: |-
      const float STANDARD_SEA_LEVEL_PRESSURE = 1034; //in hPa, see note
      return ((id(bmp280_temperature).state + 273.15) / 0.0065) *
        (powf((id(sea_level).state / id(bmp280_pressure).state), 0.190234) - 1); // in meter
    update_interval: 60s
    icon: 'mdi:signal'
    unit_of_measurement: 'm'

  - platform: homeassistant
    id: outside_temperature
    entity_id: sensor.esp_temperature
    internal: true

text_sensor:
  - platform: homeassistant
    id: sensor_door_wc
    entity_id: binary_sensor.0x00158d00019943cf_contact
    internal: true
time:
  - platform: homeassistant
    id: esptime
font:
  - file: 'Fonts/AnekDevanagari_Condensed-Regular.ttf'
    id: Anek48
    size: 48
  - file: 'Fonts/BebasNeueCyrillic.ttf'
    id: BebasCyr14
    size: 14
    glyphs: |-
          !"%()+=,-_.:°0123456789АБВГДЕЁЖЗИЙКЛМНОПРСТУФХЦЧЩЬЫЪЭЮЯABCDEFGHIJKLMNOPQRSTUVWXYZ абвгдеёжзийклмнопрстуфхцчщьыъэюяabcdefghijklmnopqrstuvwxyz'éèàòùç/&ôœìïöñ
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: D0
    address: 0x3C
    lambda: |-
      it.printf(1, 5, id(BebasCyr14), TextAlign::TOP_LEFT, "Внутри");
      it.line(50, 6, 58, 6);
      it.line(50, 6, 50, 14);
      it.line(58, 6, 58, 14);
      it.line(50, 14, 58, 14);

      it.line(51, 7, 57, 7);
      it.line(51, 7, 51, 13);
      it.line(57, 7, 57, 13);
      it.line(51, 13, 57, 13);

      if (id(sensor_door_wc).state == "off") {
        it.line(52, 8, 56, 8);
        it.line(52, 9, 56, 9);
        it.line(52, 10, 56, 10);
        it.line(52, 11, 56, 11);
        it.line(52, 12, 56, 12);
      }
      if (id(bmp280_temperature).has_state()) {
        it.printf(0, 60, id(Anek48), TextAlign::BASELINE_LEFT, "%.1f°", id(bmp280_temperature).state);
      }
      it.printf(127, 5, id(BebasCyr14), TextAlign::TOP_RIGHT, "Снаружи");
      if (id(xiaomi2_temp).has_state()) {
        it.printf(128, 60, id(Anek48), TextAlign::BASELINE_RIGHT, "%.1f°", id(xiaomi2_temp).state);
      }
# Отправка датчиков на народный мониторинг. Вписать свои mac, name, owner, lat, lon
http_request:
  timeout: 15s
interval:
  - interval: 5min
    then:
      http_request.post:
        url: https://narodmon.ru/json
        verify_ssl: false
        json:
          JsonObject device = root.createNestedArray("devices").createNestedObject();
          device["mac"] = "A4:C1:38:CA:13:06";
          device["name"] = "LYWSD03MMC";
          device["owner"] = "mr_aleks2@mail.ru";
          device["lat"] = 53.9161;
          device["lon"] = 27.5769;
          JsonArray sensors = device.createNestedArray("sensors");
          JsonObject temp = sensors.createNestedObject();
          temp["id"] = "T1";
          temp["value"] = id(xiaomi2_temp).state;
          temp["unit"] = "C";
          temp["name"] = "MI Температура";
```

### Anything in the logs that might be useful for us?

```txt
Регистратор: homeassistant.components.websocket_api.http.connection
Источник: components/websocket_api/commands.py:239
интеграция: Home Assistant WebSocket API (документация, проблемы)
Первое сообщение: 12:56:57 (2 сообщений)
Последнее сообщение: 13:06:20

[140250688879296] Error updating esphome-web-e3b81b.yaml via OTA; Try again in ESPHome dashboard for more information.
[140250688877632] Error updating esphome-web-e3b81b.yaml via OTA; Try again in ESPHome dashboard for more information.
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2319, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2356, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 905, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 161, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 465, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 184, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating esphome-web-e3b81b.yaml via OTA; Try again in ESPHome dashboard for more information.
```

ssieb, 2024-03-23:
Did you try what it's telling you to do?

mr-aleks, 2024-03-23:
> Did you try what it's telling you to do?

via dashboard esphome ?

[image]
```
Uploading: [=====                                                       ] 8% ERROR Error receiving acknowledge chunk OK: [Errno 104] Connection reset by peer
===== [ERROR] /config/esphome/esphome-web-e3b81b.yaml =====
```

the code did not change, although previously updates were always successful

mr-aleks, 2024-03-23:
Now the update has been successful, the notification in the system has disappeared, what it was is unclear
[image]

closed 2024-03-24 by mr-aleks