esphome/issues #5629: 2024.3.0 breaks existing configs with addition of project version update

excalbian, 2024-03-23:
### The problem

Starting with 2024.3.0 esphome won't compile most of my firmware that has been working fine for years. Seems like the implementation of adding `on_update` trigger to project version introduced a regression by limiting `PROJECT_VERSION` to 30 characters (29?).

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

20240.3.3 - N/A

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

core

### Example YAML snippet

```yaml
substitutions:
  devicename: big-tank
  devicename_upper: Big Tank
  platform: esp32
  board: esp32doit-devkit-v1
  project_name: excalbian.big-tank
  project_version: "v1.0.1 (${smart_tank_version} ${wifi_version} ${project_info_version} ${text_version} ${restart_version})"
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/api/api_frame_helper.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h: In member function 'virtual void esphome::ProjectUpdateTrigger::setup()':
src/esphome/core/defines.h:5:33: error: initializer-string for array of chars is too long [-fpermissive]
 #define ESPHOME_PROJECT_VERSION "v1.0.1 (tank-0.5.3 wifi-1.1.0 info-1.0.1 text-1.0.1 restart-1.0.1)"
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/defines.h:5:33: note: in definition of macro 'ESPHOME_PROJECT_VERSION'
 #define ESPHOME_PROJECT_VERSION "v1.0.1 (tank-0.5.3 wifi-1.1.0 info-1.0.1 text-1.0.1 restart-1.0.1)"
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/big-tank/src/main.cpp.o] Error 1
========================= [FAILED] Took 122.91 seconds =========================
```

### Additional information

I've been making use for years now of shared YAML (packages) to modularize my esphome configurations across 24+ devices. Where those ${*_version} come from the included packages so it's easy to tell what version of each package was being used in a given firmware.

With the  addition of the [on update trigger for project versions](https://github.com/esphome/esphome/pull/6298) it seems the code assumes / enforces a limit of 30 characters to the version string, causing my firmware to not be able to compile.

excalbian, 2024-03-23:
I was thinking this could be fixed with something as simple as:

```
uint32_t project_version_size = sizeof(ESPHOME_PROJECT_VERSION);
char previous_version[project_version_size];
char current_version[project_version_size] = ESPHOME_PROJECT_VERSION;
```

except that may not work, as there is no guarantee of the size of the previous version 😞. How can we test the change with a previous version that's longer than the current version to see how it's handled (or check that `pref.load` won't cause a buffer overflow)?

jesserockz, 2024-03-26:
@excalbian I guess there is nothing wrong with the way you have done it, but another way for your use case could be to create a diagnostic `text_sensor` with the static version inside each of your packages.

closed 2024-08-11 by github-actions[bot]