esphome/issues #5632: Networking broken on es8266 (minimal config included)

ghost58400, 2024-03-24:
### The problem

Network sockets (at least UDP) does not work anymore on esp8266 platform, but works perfectly on esp32 arduino and libretiny, it may be linked to the introduction of IPv6 recently.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

nodemcuv2, esp8285

### Component causing the issue

All UDP sockets (at least)

### Example YAML snippet

```yaml
esphome:
  name: test-udp

esp8266:
  board: nodemcuv2

#esp32:
#  board: esp32dev
#  framework:
#    type: arduino

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: test-udp
    password: !secret fallback_ap_password

captive_portal:

logger:
  level: VERY_VERBOSE

e131:
  method: multicast

light:
  - platform: partition
    name: "Partition Light"
    effects:
      - e131:
          universe: 1
    segments:
      - single_light_id: my_light_1
      - single_light_id: my_light_2

  - platform: monochromatic
    output: my_output
    id: my_light_1
  
  - platform: monochromatic
    output: my_output
    id: my_light_2

output:
  - platform: template
    id: my_output
    type: float
    write_action:
      - logger.log: "writing output"
```

### Anything in the logs that might be useful for us?

```txt
When using the above YAML on esp8266, nothing in the logs when issuing 
echo -n "hello" | nc -u -w0 test-udp.local 5568
on an other machine, but when flashing the same code on esp32 (uncommented esp32), the logs appears with the same command, indicating that esphome receives the UDP packet :
[V][e131:068]: Invalid packet received of size 5.
```

### Additional information

Tested on multiple devices including nodemcuv2, esp8285, esp32, libretiny. Esp32, both arduino and esp-idf, and libretiny works perfectly, but esp8266 (and esp8285) does not work.

Also, the code provided in https://esphome.io/cookbook/lambda_magic.html#send-udp-commands has the same behaviour as e131 UDP listener, it works on esp32 and libretiny but not on esp8266. This means that sockets (at least UDP) does not work on esp8266 platform, both in listening and sending packets.

I also tried to use the Arduino's WifiUDP library, it works well on esp8266 inside esphome, so the problem comes from socket implementation in esphome.

ssieb, 2024-03-25:
The socket library doesn't support UDP on ESP8266 or pico.

ghost58400, 2024-03-27:
Thank you for your insight, if it is normal, why is there no error message? The official components using UDP should not be marked compatible with esp8266
Ans also using the sockets library should trigger an error log if used with a custom component
The documentation of cookbook sending UDP packet should also precise that esp8266 is not working with this code, users can loose a lot of time by trying this code since it doesn't work and gives no error message

ssieb, 2024-04-02:
UDP was never working in the sockets library on 8266.  There's also the WifiUDP class, but it only works on Arduino.

ghost58400, 2024-05-20:
Hello, since esp8266 platform always uses Arduino, would it be possible to use WifiUDP inside Sockets on esp8266 platform ?