esphome/issues #5634: CSE7766 sensor constantly "... Received new state..."

mpflaga, 2024-03-25:
### The problem

Versions greater than 2023.12.9 appear to update and push sensor data back-to-back, spewing traffic, causing delays downstream. 
Through trial and error, I see the problem is introduced between 2023.12.9 & 2024.2.0b1. I suspect a specific commit, detailed below.

Filtering the log level for sensors helps reduce log noise. However, TCPdump shows the spewing of packets on 6053. 

The amount of traffic generated appears to affect the performance of web_server and is likely filling up HA's DB.
I have read some discord about S31's not reliably connecting with newer firmware yet working with old 23.12.9.

I first detected the anomaly when I updated a pilot S31 from 2023.12.6 to 2024.3.0. Then using PIP and venv I narrowed down the introduction of the anomaly to first present in 2024.2.0b1.

### Which version of ESPHome has the issue?

2024.2.0b1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.3.3, Supervisor 2024.03.0, Operating System 12.1

### What platform are you using?

ESP8266

### Board

SONOFF S31

### Component causing the issue

cse7766

### Example YAML snippet

```yaml
Below is the Package that I use for all my SONOFF S31's. Just add Network info.

# ESPHome Package for hardware setup of Shelly Plus 1
# example use:
#    substitutions:
#      esphome_name: living-room
#      friendly_name: Living Room
#    packages:
#      board: !include .sonoff-s31.yaml
#      network: !include .network.yaml

substitutions:
  model_name: "Sonoff S31"

esphome:
  name: ${esphome_name}
  friendly_name: ${friendly_name}

esp8266:
  board: esp01_1m

logger:
  baud_rate: 0 # (UART logging interferes with cse7766)

uart:
  rx_pin: RX
  baud_rate: 4800

switch:
  - platform: gpio
    name: "${model_name} Relay"
[... 623 lines cut ...]
[16:36:57][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[16:36:57][V][json:058]: Size after shrink 84 bytes
[16:36:57][V][sensor:043]: 'Sonoff S31 Power': Received new state 0.000000
[16:36:57][D][sensor:093]: 'Sonoff S31 Power': Sending state 0.00000 W with 1 decimals of accuracy
[16:36:57][V][sensor:043]: 'Sonoff S31 Daily Energy': Received new state 0.000000
[16:36:57][D][sensor:093]: 'Sonoff S31 Daily Energy': Sending state 0.00000 Wh with 3 decimals of accuracy
[16:36:57][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[16:36:57][V][json:058]: Size after shrink 88 bytes
[16:36:57][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[16:36:57][V][json:058]: Size after shrink 80 bytes
[16:36:57][V][sensor:043]: 'Sonoff S31 Current': Received new state 0.000000
[16:36:57][D][sensor:093]: 'Sonoff S31 Current': Sending state 0.00000 A with 1 decimals of accuracy
[16:36:57][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
...etc...
```

### Additional information

I am suspicious that the following commit may be cause.
```
    SHA-1: 8b2d76e8cebca0055b08319076f5cc8d94f25383
    * convert cse7766 to non-polling (#6095)
    Co-authored-by: Samuel Sieb <samuel@sieb.net>
```
I was not able to confirm this. My attempts to revert to just before this commit and run would not successfully compile.

mpflaga, 2024-03-25:
I had tried adding filtering throttle_average to each of the sensors, but it did not change anything.

ssieb, 2024-03-25:
You're using verbose logging.  Of course that commit is the cause, but you're not expected to use that log level for normal use.

closed 2024-03-25 by ssieb

mpflaga, 2024-03-25:
Thank you for the response. 
I see now that I had inadvertently associated the verbose log sensor reads to that of the API pushes, that I was seeing in the tcpdumps, during my investigation. This was incorrect. 

I relooked at it with the following snippet containing the correction as you pointed out and I now see the tcpdump behaving as expected.

```
...
logger:
  baud_rate: 0 # (UART logging interferes with cse7766)
  logs:
    sensor: DEBUG # (default)
sensor:
  - platform: cse7766
    apparent_power:
      name: "${model_name} Apparent Power"
      filters:
        - throttle_average: ${update_interval}
...
```
As detailed at [CSE7766 Power Sensor](https://esphome.io/components/sensor/cse7766.html)