esphome/issues #5636: Error compile  : UnknownPackageError: Could not find the package with 'platformio/tool-scons @ ~4.40600.0' 'linux_x86_64'

Jamiro75, 2024-03-25:
### The problem

Hi everyone !
I have been using esphome in my Linux desktop with docker for years without problems,
all versions >2024 give me this programming error
the latest version 2023.12.9 works.
below a sample yaml code, but the error appears in all my devices if I use platform >2024

### Which version of ESPHome has the issue?

2024.x.x

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

linux_x86_64'

### What platform are you using?

ESP8266

### Board

esp8266, esp32

### Component causing the issue

UnknownPackageError: Could not find the package with 'platformio/tool-scons @ ~4.40600.0' 

### Example YAML snippet

```yaml
yaml 
example

esphome:
  name: esp32-test
  platform: ESP32
  board: esp32dev
  name_add_mac_suffix: true
  
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxxxxxx"
```

### Anything in the logs that might be useful for us?

```txt
this error appear on >2024

INFO ESPHome 2024.3.0
INFO Reading configuration /config/esp-cam32.yaml...
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esp-cam32 (board: esp32cam; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-scons @ ~4.40600.0
INFO Installing platformio/tool-scons @ ~4.40600.0
UnknownPackageError: Could not find the package with 'platformio/tool-scons @ ~4.40600.0' requirements for your system 'linux_x86_64'

on 2023 is ok :

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esp32-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-test (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
RAM:   [=         ]  12.2% (used 39928 bytes from 327680 bytes)
Flash: [====      ]  42.6% (used 781141 bytes from 1835008 bytes)
========================= [SUCCESS] Took 2.23 seconds =========================
INFO Successfully compiled program.
```

ssieb, 2024-03-25:
Try rebuilding the container.  I assume your config files are stored outside of the container.

Jamiro75, 2024-03-26:
Oh yes, sure ..
When i've change the version i rebuild always the container but all the
2024.x.x give me the same error.
I try to clean up the building file ,  and in the working dir,  to have
only the yaml file but when compile the error go on.

jeff-h, 2024-04-27:
I am seeing a very similar error whenever I try to update or recompile for one of my ESP8266's:

`UnknownPackageError: Could not find the package with 'platformio/espressif8266 @ 4.2.1' requirements for your system 'linux_x86_64'`

Running ESPHome 2024.4.1, inside docker (using compose) under Linux x86. I've rebuilt the container without any change.

jeff-h, 2024-04-27:
Reverting to `2023.12.7 ` by changing my `docker-compose.yml` to use `image: ghcr.io/esphome/esphome:2023.12.7` has resolved the issue for now.

gpala7077, 2024-05-14:
I'm receiving the same error whenever I try to upgrade or install from the YAML.
[image]

gpala7077, 2024-05-14:
> I'm receiving the same error whenever I try to upgrade or install from the YAML. <img alt="image" width="490" src="https://private-user-images.githubusercontent.com/16492608/330271516-90009059-2a7f-4ef6-ac30-20d2349a0611.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTkwODUsIm5iZiI6MTcxNTY1ODc4NSwicGF0aCI6Ii8xNjQ5MjYwOC8zMzAyNzE1MTYtOTAwMDkwNTktMmE3Zi00ZWY2LWFjMzAtMjBkMjM0OWEwNjExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzNTMwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5ZTE4ZmQ5ZjgzNjFiMjgzNzk0ZDgxMzFlZDEzYjRjMzkzMmE1ZjMyOGMxZGZmOTRmZDJmMGQ5ZTgxZTIwNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.duinlmz2Bz9dOq8IIeuV8H5x5AFDbb5j589wTVsBYH4">
> 
> ```
>   esphome:
>     container_name: esphome
>     image: esphome/esphome:latest
>     volumes:
>       - /home/curiosity/smart_home/data/esphome/:/config   # Directory where home-assistant config files
>       - /etc/localtime:/etc/localtime:ro        # Expose local time
>     devices:
>       - /dev/ttyUSB0:/dev/ttyUSB0               # Expose USB inputs
>       - /dev/ttyUSB1:/dev/ttyUSB1               # Expose USB inputs
>     restart: always
>     privileged: true
>     network_mode: host
> ```

SOOOO..

 I DID A VERY BAD THING BUT IT WORKED....
 
 If you are docker user
 ```
 docker exec -it esphome /bin/bash
/usr/bin/python3 -m pip install -U platformio --break-system-packages
 ```

[image]

slovdahl, 2024-06-04:
YMMV, but I finally found a workaround.

My `docker-compose.yaml` has this mount:

```yaml
    volumes:
      - /home/homeassistant/esphome/config:/config
```

After removing the `.esphome/platformio` directory in the config folder it all started working.

```
rm -rf /home/homeassistant/esphome/config/.esphome/platformio
```

jeff-h, 2025-02-02:
Thanks @slovdahl — that fixed the issue for me!