esphome/issues #5637: ESPHome dashboard Python 3.12 - errors

mcpat-it, 2024-03-26:
### The problem

Installed esphome like always in a venv, up to Python 3.11 no problem, with 3.12,  If I click on logs I receive an error.

Also flashing is not possible.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.1

### What platform are you using?

ESP8266

### Board

d1_mini and esp32-c3-devkitm-1

### Component causing the issue

dashboard at logs and flashing (with another error)

### Example YAML snippet

```yaml
--
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.3.0
INFO Reading configuration config/slimmelezer.yaml...
Traceback (most recent call last):
  File "/opt/esphome.test/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/esphome.test/lib/python3.12/site-packages/esphome/__main__.py", line 1061, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/esphome.test/lib/python3.12/site-packages/esphome/__main__.py", line 1048, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/esphome.test/lib/python3.12/site-packages/esphome/__main__.py", line 474, in command_logs
    return show_logs(config, args, port)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/esphome.test/lib/python3.12/site-packages/esphome/__main__.py", line 376, in show_logs
    from esphome.components.api.client import run_logs
  File "/opt/esphome.test/lib/python3.12/site-packages/esphome/components/api/client.py", line 8, in <module>
    from aioesphomeapi import APIClient
  File "/opt/esphome.test/lib/python3.12/site-packages/aioesphomeapi/__init__.py", line 2, in <module>
    from .api_pb2 import (  # type: ignore[attr-defined] # noqa: F401
  File "/opt/esphome.test/lib/python3.12/site-packages/aioesphomeapi/api_pb2.py", line 17, in <module>
    from . import api_options_pb2 as api__options__pb2
  File "/opt/esphome.test/lib/python3.12/site-packages/aioesphomeapi/api_options_pb2.py", line 52, in <module>
    google_dot_protobuf_dot_descriptor__pb2.MethodOptions.RegisterExtension(needs_setup_connection)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'MethodOptions' has no attribute 'RegisterExtension'
```
```

### Additional information

Everything is running well with Python 3.11

ssieb, 2024-03-26:
I've been using Python 3.12 for quite a while with no issues.  I just tested the dashboard and that works fine too.  Make sure things are up to date.

mcpat-it, 2024-03-26:
I downloaded now the requirements.txt and checked, all packages up to date, and the error still exists... I also updated homeassistant with python 3.12, no errors, it's running.

mcpat-it, 2024-03-26:
@ssieb Now it's running, have an old raspi (don't want to setup all again like z-wave, ...), I did

```
pip uninstall protobuf
pip cache remove protobuf
pip install --no-binary :all: protobuf
```
It compiled and now esphome is running with python 3.12

closed 2024-03-26 by mcpat-it