esphome/issues #5639: ESP8266 switch not processing interlock and delay

BartKoopmans-private, 2024-03-26:
### The problem

When using a ESP8266MOD 12F from AZ-Delivery I cannot get the delay and interlock commands to work.
in the web interface and in Home Assistant the switches can switch on and off, but they don't switch off after 500ms or when the other switch is triggered.

When using the code on a ESP32-WROOM-32 it is working fine... 

the board will be used to switch a relay

the code used for the 8266:

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

AZ-Delivery D1 mini (ESP8266MOD 12-F)

### Component causing the issue

switch

### Example YAML snippet

```yaml
substitutions:
  name: esphome-web-test1
  friendly_name: relayboard1

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

switch:
  - platform: gpio
    name: "Open"
    id: relay_1
    pin: 0
    interlock: relay_2
   inverted: True
    on_turn_on:
      - delay: 500ms
      - switch.turn_off: relay_1
    restore_mode: RESTORE_DEFAULT_OFF
    
  - platform: gpio
    name: "Close"
    id: relay_2
    pin: 2
    interlock: relay_1
    inverted: True
    on_turn_on:
      - delay: 500ms
      - switch.turn_off: relay_2
    restore_mode: RESTORE_DEFAULT_OFF_
```

BartKoopmans-private, 2024-03-28:
nobody who can help?

ssieb, 2024-03-28:
If you want a quicker response, you should ask on discord.
You also didn't include any logs.

BartKoopmans-private, 2024-03-28:
Thanks, this can be closed now. This seem to be resolved in the new update that was released yesterday.

BartKoopmans-private, 2024-03-28:
Can be closed

closed 2024-03-28 by BartKoopmans-private