esphome/issues #5640: Bootloop after updates without any code change

roboraptor, 2024-03-26:
### The problem

Hi I have problem with newly built FW in ESPhome. 
I came arund by doing OTA update of my Air purifier and it stopped working.
So I went to my dashboard and it was offline. 
Then I opened the Air purifier and found my ESP board flashing like every 5seconds. 
So I connected serial port and got a lot of weird numbers and such.
I didnt know what it is so I went on google and realised those are som kind of exception errors and that my ESP board is bootlooping. 
I wanted to get my errors decoded co I know what is happening.
In the mean time I thought my program got too big but it is really just few lines. I disabled captive portal and builded it for esp01 (512K only) and it fitted and flashed well too. 
Then I got fiddling with my terminal to get the ELF file from the .pioenv, and I uploaded it to [Online Exception Stack Decoder](https://maximeborges.github.io/esp-stacktrace-decoder/) and tha output is in "anything useful" section.
Now after this I dont really know what else to do. 

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

ESP01-1M

### Example YAML snippet

```yaml
# https://community.home-assistant.io/t/ikea-fornuftig-in-home-assistant/318341
esphome:
  name: fornuftig
  friendly_name: Fornuftig

esp8266:
  board: esp01

# Enable logging
logger:
  level: NONE

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name} fallback hotspot"
    password: "fornuftig"

    
sensor:
  # Read LED from adc input 10 times per second
  - platform: adc
    pin: A0
    internal: true
    id: led_internal
    update_interval: 100ms
    accuracy_decimals: 1
    filters:
      - delta: 0.1
    on_value:
      if:
        condition:
          lambda: 'return x < 0.2;'
        then:
          - binary_sensor.template.publish:
              id: led_sensor
              state: OFF
        else:
          - binary_sensor.template.publish:
              id: led_sensor
              state: ON

binary_sensor:
  - platform: template
    id: led_sensor
    name: "${friendly_name} Filter"
    device_class: problem
  - platform: gpio
    internal: true
    id: in_d1
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - fan.turn_off:
          id: fan_internal
  - platform: gpio
    internal: true
    id: in_d5
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - fan.turn_on:
          id: fan_internal
          speed: 1
  - platform: gpio
    internal: true
    id: in_d6
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - fan.turn_on:
          id: fan_internal
          speed: 2
  - platform: gpio
    internal: true
    id: in_d7
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - fan.turn_on:
          id: fan_internal
          speed: 3

output:
  - platform: gpio
    pin: GPIO4
    id: gpio_d2
    inverted: true
  - platform: gpio
    pin: GPIO0
    id: gpio_d3
    inverted: true
  - platform: gpio
    pin: GPIO2
    id: gpio_d4
    inverted: true
  - platform: template
    id: fan_out_internal
    type: float
    write_action:
      if:
        condition:
          lambda: 'return state < 0.1;'
        then:
          - output.turn_off: gpio_d4
          - output.turn_off: gpio_d3
          - output.turn_off: gpio_d2

        else:
          - if:
              condition:
                lambda: 'return state < 0.4;'
              then:
                - output.turn_on:  gpio_d4
                - output.turn_off: gpio_d3
                - output.turn_off: gpio_d2

              else:
                - if:
                    condition:
                      lambda: 'return state < 0.7;'
                    then:
                      - output.turn_off: gpio_d4
                      - output.turn_on:  gpio_d3
                      - output.turn_off: gpio_d2

                    else:
                      - output.turn_off: gpio_d4
                      - output.turn_off: gpio_d3
                      - output.turn_on:  gpio_d2

fan:
  - platform: speed
    output: fan_out_internal
    id: fan_internal
    name: "${friendly_name} Fan"
    speed_count: 3
```

### Anything in the logs that might be useful for us?

```txt
0x4023e962: void std::__cxx11::basic_string, std::allocator >::_M_construct_aux(char*, char*, std::__false_type) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:249
0x4023e962: void std::__cxx11::basic_string, std::allocator >::_M_construct(char*, char*) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:268
0x4023e962: std::__cxx11::basic_string, std::allocator >::basic_string(std::__cxx11::basic_string, std::allocator > const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:453
0x40240000: tcp_bind at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c:750
0x4023ec2e: glue2esp_linkoutput at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:258
0x4023a30d: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:552
0x4023a2f4: _dtoa_r at /workdir/repo/newlib/newlib/libc/stdlib/dtoa.c:549
0x4023b663: __multiply at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:368
0x4023b6ab: __multiply at /workdir/repo/newlib/newlib/libc/stdlib/mprec.c:382
```

roboraptor, 2024-03-26:
I also used the original build for **ESP01-1M** board

**Example YAML snippet**
```
# https://community.home-assistant.io/t/ikea-fornuftig-in-home-assistant/318341
esphome:
  name: fornuftig
  friendly_name: Fornuftig

esp8266:
  board: esp01_1m

# Enable logging
logger:
  logs:
      # Make adc input less verbose
      adc: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name} fallback hotspot"
    password: "fornuftig"

captive_portal:
    
    
sensor:
  # Read LED from adc input 10 times per second
  - platform: adc
    pin: A0
    internal: true
    id: led_internal
    update_interval: 100ms
    accuracy_decimals: 1
    filters:
      - delta: 0.1
    on_value:
      if:
        condition:
          lambda: 'return x < 0.2;'
        then:
          - binary_sensor.template.publish:
              id: led_sensor
              state: OFF
        else:
          - binary_sensor.template.publish:
              id: led_sensor
              state: ON

binary_sensor:
  - platform: template
    id: led_sensor
    name: "${friendly_name} Filter"
    device_class: problem
  - platform: gpio
    internal: true
    id: in_d1
    pin:
[... 144 lines cut ...]
0x4023be30: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x4023bd6c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x4023be30: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x4023792c: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
0x4023be30: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x4023c02d: _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528
0x402370c4: _printf_float at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_float.c:310
0x40237900: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:235
0x4023bd6c: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x4023e284: std::__cxx11::basic_string, std::allocator >::_S_copy_chars(char*, char*, char*) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:395
0x4023e961: std::__cxx11::basic_string, std::allocator >::basic_string(std::__cxx11::basic_string, std::allocator > const&) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:453
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x40239531: _vsnprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c:71
0x4023dbfc: operator delete(void*, unsigned int) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_ops.cc:34
```

ssieb, 2024-03-26:
Paste the exception message and a few log lines from before that.

ssieb, 2024-03-26:
Also, you should use the new template fan or at least use the new speed trigger.
And why do you have a speed number when the fan already has a control for that?  I wouldn't be surprised if you're getting an infinite loop by doing that.

roboraptor, 2024-03-26:
Oh Im sorry, that is all I could recover. The original exception was like date and then the hexadecimal stack of numbers. I went that far I recovered the ELF file that I could translate it to ASCII. But there was literally no more info in the stack.

roboraptor, 2024-03-26:
> Also, you should use the new template fan or at least use the new speed trigger. And why do you have a speed number when the fan already has a control for that? I wouldn't be surprised if you're getting an infinite loop by doing that.

I didnt know that. I used the manual from someone who did it and it worked. Till now, when I tried to update it.

ssieb, 2024-03-26:
Can't you get the serial logs again?

ssieb, 2024-03-26:
What are the last esphome log lines before that?

roboraptor, 2024-03-26:
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00080210
~ld
[0;32m[I][logger:034]: Log initialized[0m
[0;35m[C][ota:483]: There have been 9 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][binary_sensor:034]: 'in_d1': Sending initial state OFF[0m
[0;36m[D][binary_sensor:034]: 'in_d5': Sending initial state OFF[0m
[0;36m[D][binary_sensor:034]: 'in_d6': Sending initial state OFF[0m
[0;36m[D][binary_sensor:034]: 'in_d7': Sending initial state OFF[0m
[0;36m[D][fan:120]: '${friendly_name} Fan' - Sending state:[0m
[0;36m[D][fan:121]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Setting:[0m
[0;36m[D][fan:024]:   State: OFF[0m
[0;36m[D][number:054]: '${friendly_name} Fan Speed' - Setting number value[0m
[0;36m[D][number:113]:   New number value: 0.000000[0m
[0;36m[D][number:012]: '${friendly_name} Fan Speed': Sending state 0.000000[0m
[0;36m[D][fan:021]: '${friendly_name} Fan' - Sett
User exception (panic/abort/assert)
```

ssieb, 2024-03-26:
Right, so that's what I was expecting.  You've created an infinite loop.  Remove the number.  If you need help with the config, come ask on the esphome discord server.

closed 2024-03-26 by ssieb

roboraptor, 2024-03-26:
Ok thank you.
I will try the template fan instead of speed