esphome/issues #5643: Component display took a long time for an operation (1368 ms).

jaytonic, 2024-03-26:
### The problem

I've a Waveshare 7.5" with a V2 sticker (and another one with 075BN-T7) with a waveshare  e-paper esp32 driver board.
I configured my display as shown on multiple different website.

Sometimes it does manage to refresh, but 3/4 of the time, it get errors.
When it display something, it's quite faded

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP32

### Board

Waveshare e-Paper ESP32 Driver board

### Component causing the issue

Waveshare e-paper 7.5"

### Example YAML snippet

```yaml
esphome:
  name: hallway_epaper
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hallway Fallback Hotspot"
    password: "FALLBACKDPSKGOESHERE"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# Example configuration entry
spi:
  clk_pin: 13
  mosi_pin: 14
    
font:
  - file: "segoeui.ttf"
    id: segoe
    size: 64

display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 7.50inv2
    update_interval: 30s
    lambda: |
     it.print(0, 0, id(segoe), "E-Paper is awesome!");

```
```

### Anything in the logs that might be useful for us?

```txt
[I][waveshare_epaper:2339]: Power on the display and hat
[E][waveshare_epaper:2284]: Timeout while displaying image!
[E][waveshare_epaper:2284]: Timeout while displaying image!
[W][component:232]: Component display took a long time for an operation (21388 ms).
[W][component:233]: Components should block for at most 30 ms.
[I][waveshare_epaper:2339]: Power on the display and hat
[E][waveshare_epaper:2284]: Timeout while displaying image!
```

ssieb, 2024-03-26:
You're flagging the wrong line.  The `Timeout while displaying image!` is likely the problem.  It sounds like an issue with the display or connection.  I suggest you come ask on the esphome discord server instead.

jaytonic, 2024-03-26:
I moved to the 7.50inv2alt, and now everything seems better, I don't have timeout anymore, the screen is more vivid.
If I check the logs, I still have some logs that doesn't look too good:

```
[I][waveshare_epaper:2339]: Power on the display and hat
[W][component:232]: Component display took a long time for an operation (1369 ms).
[W][component:233]: Components should block for at most 30 ms.
```

ssieb, 2024-03-26:
Did you see the pinned issue when went to create this one?
https://github.com/esphome/issues/issues/4717
[image]

closed 2024-03-26 by ssieb