esphome/issues #5644: When using wt32-eth01 as board and esp-idf as framework type, ethernet is not working on a WT32-ETH01

KlausHans, 2024-03-26:
### The problem

If i use wt32-eth01 as board and esp-idf as framework type, ethernet is not working on a WT32-ETH01. The leds on the ethernet connector don't light up, nothing.
If i use wt32-eth01 as board  and arduino as framework type it works.
If i use esp32dev as board  and esp-idf as framework type it also works.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP32-IDF

### Board

wt32-eth01

### Component causing the issue

esp32

### Example YAML snippet

```yaml
substitutions:
  name: bluetooth-proxy-eg
  friendly_name: Bluetooth Proxy EG
  board: wt32-eth01 #wt32-eth01 not working together with esp-idf

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: ${board}
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms

bluetooth_proxy:
  active: true
```

KlausHans, 2024-03-27:
Just found this https://github.com/esphome/esphome/discussions/4207#discussioncomment-7665609
So, someone else confirms the problem.

g-kiss, 2024-04-06:
If you have ESP32-ETH01 (with 8MB flash), use this:
```
esphome:
  platformio_options:
    board_build.flash_mode: dio
    board_build.flash_size: keep
```

WalterDSU, 2024-04-29:
> If you have ESP32-ETH01 (with 8MB flash), use this:
> 
> ```
> esphome:
>   platformio_options:
>     board_build.flash_mode: dio
>     board_build.flash_size: keep
> ```

wowwwww, it works now. very good to know this. thanks  for sharing.

Chris112, 2024-05-12:
> If you have ESP32-ETH01 (with 8MB flash), use this:
> 
> ```
> esphome:
>   platformio_options:
>     board_build.flash_mode: dio
>     board_build.flash_size: keep
> ```

I have a wts-eth01 v1.4 (aliexpress claims 32mb flash) and I added this to my esphome config and it fixed the boot loop issue too, thanks xo

darkxst, 2024-05-14:
> If you have ESP32-ETH01 (with 8MB flash), use this:

ESP32-ETH01 has 4MB Flash

It would appear that some pins required for qio flash are muxed with ethernet MAC. 

This probably needs to be fixed here:
https://github.com/platformio/platform-espressif32/blob/022e6044b468d0db65d502f5a16a25f446e94f5e/boards/wt32-eth01.json#L12

halogen28372, 2024-11-08:
I'm having a similar problem.

darkxst, 2024-11-08:
@halogen28372 just override the setting to  `"flash_mode": dio` in your config. I submitted an upstream patch for this, but it was never merged.

xtypebox, 2024-11-12:
Hello, I have a  ESP32-ETH01 (with 8MB flash) and I use the: 

```
esphome:
  platformio_options:
    board_build.flash_mode: dio
    board_build.flash_size: keep
```

But still no luck.

`[10:35:50][I][logger:354]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][ethernet:029]: Setting up Ethernet...
[D][esp-idf:000]: E (625) esp.emac: emac_esp32_init(349): reset timeout

[D][esp-idf:000]: E (625) esp_eth: esp_eth_driver_install(214): init mac failed

[E][ethernet:106]: ETH driver install error: (263) ESP_ERR_TIMEOUT
[E][component:113]: Component ethernet was marked as failed.`

peff74, 2025-01-27:
the are some WT32-eth01 out there with 8MB flash, until now, they have always been in version 1.4

WalterDSU, 2025-01-28:
esphome updated yaml 5 month ago to fix this issue, do you check it out?

https://github.com/esphome/bluetooth-proxies/tree/main/wt32

ammgws, 2025-04-05:
Confirmed those settings worked for my wt32-eth01 device

```yaml
esphome:
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: wt32-eth01
  framework:
    type: esp-idf
```

KlausHans, 2025-12-07:
I just tried it without any added options:
```
esphome:
  name: growatt-mod-5000tl3-xh
  friendly_name: Growatt MOD 5000TL3-XH

esp32:
  board: wt32-eth01
  framework:
    type: esp-idf

...
```
And it works!
I didn't tried it with my bluetooth proxy, though.
Can someone confirm it?