esphome/issues #5645: Homeassitant adoption of device fails for ESP32C3 "404 not found"

Rattkener, 2024-03-27:
### The problem

I am running Homeassistant OS locally. No HTTPS. 
I have used the esphome web utility to flash my ESP32 C3, given it proper Wi-Fi creds, and have verified operation by connecting directly to the ESP. 
Next, I swing back over to HA, see it is discovered in the ESPHome AddOn, and attempt to adopt. However it fails with this in the logs:

2024-03-26 22:30:32,818 ERROR Error while fetching https://raw.githubusercontent.com/esphome/firmware/v2/esphome-web/esp32c3.yaml: 404 Client Error: Not Found for url: https://raw.githubusercontent.com/esphome/firmware/v2/esphome-web/esp32c3.yaml

Testing on OOB device shows this does in fact provide a 404. And without any additional information I am led to believe this is the only cause of the issue. 

I have installed the most recent update (pushed ~1hr ago) and rebooted with the same issue. 

### Which version of ESPHome has the issue?

2024.3.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

SeeedStudio (3 e's are intentional)

### Component causing the issue

XIAO-ESP32c3

### Anything in the logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2024.3.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.1  (aarch64 / odroid-n2)
 Home Assistant Core: 2024.3.3
 Home Assistant Supervisor: 2024.03.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[22:23:57] INFO: Configuring timezone (America/New_York)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[22:23:59] INFO: Waiting for ESPHome dashboard to come up...
[22:23:59] INFO: Starting ESPHome dashboard...
2024-03-26 22:24:01,657 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[22:24:01] INFO: Starting NGINX...
[22:24:02] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-03-26 22:26:32,832 INFO 200 GET /devices (0.0.0.0) 2.28ms
2024-03-26 22:30:27,948 INFO 304 GET / (0.0.0.0) 26.15ms
2024-03-26 22:30:28,019 INFO 304 GET /devices (0.0.0.0) 1.54ms
2024-03-26 22:30:28,921 INFO 304 GET /secret_keys (0.0.0.0) 4.62ms
2024-03-26 22:30:32,818 ERROR Error while fetching https://raw.githubusercontent.com/esphome/firmware/v2/esphome-web/esp32c3.yaml: 404 Client Error: Not Found for url: https://raw.githubusercontent.com/esphome/firmware/v2/esphome-web/esp32c3.yaml
2024-03-26 22:30:32,822 WARNING 422 POST /import (0.0.0.0) 510.83ms
2024-03-26 22:30:33,022 INFO 304 GET /devices (0.0.0.0) 1.41ms
2024-03-26 22:44:05,900 INFO 304 GET / (0.0.0.0) 1.12ms
2024-03-26 22:44:05,952 INFO 304 GET /devices (0.0.0.0) 1.59ms
2024-03-26 22:44:11,190 INFO 304 GET /devices (0.0.0.0) 1.25ms
2024-03-26 22:44:17,188 INFO 304 GET /devices (0.0.0.0) 1.20ms
```

ssieb, 2024-03-28:
Try again now.

Rattkener, 2024-03-28:
That worked TYVM

ssieb, 2024-03-28:
We don't really understand what happened there (or how it was even working before), but Jesse created the tag now.

closed 2024-03-28 by ssieb