esphome/issues #5646: problems with connecting to Pi Pico

mbirrell66, 2024-03-27:
### The problem

after the last update, I have a problem connecting to my Pi Pico devices. My ESP32 devices are connecting as expected. Please guys, can you test your updates before deploying them? This is an ongoing pain for me. One device I have got is an automatic chicken door and although the foxes love the idea of it failing to close, I would rather it close. 

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

RP2040

### Board

Pi Pico

### Component causing the issue

api

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.3.1
INFO Reading configuration /config/esphome/hendoor.yaml...
INFO Starting log output from hendoor.local using esphome API
WARNING Can't connect to ESPHome API for hendoor.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to hendoor.local in the background
```

### Additional information

I have uninstalled EspHome and reinstalled it. This time that doesn't work.

ssieb, 2024-03-27:
Updates are tested before releasing.  Please come to the esphome discord server for help with this.
(and "latest" is not a version)

mbirrell66, 2024-03-30:
The fix is remove every device I have that is a Pi Pico and re-create it every x time you do an update to ESPHome - just like paying Russian roulette lol. I know that none of you guys are professional programmers so it would be very unfair of me to expect that level of compliance. Keep up the good work because I love what you have created.

ssieb, 2024-03-30:
Other people don't have that problem and you never actually tried to resolve the issue with us.

mbirrell66, 2024-03-30:
I did as you asked and got no reply. Other people have had this problem and it was resolved by an update at least 2 times that I know of. Like I said it isn't a problem, because we are paying peanuts to have even this support. Without it, I would have had to do a lot of research and coding on my own. For me backups usually save me but not this time.

closed 2024-03-30 by mbirrell66