esphome/issues #5647: Panic with W5500 ethernet SPI on esp32cam

npdgm, 2024-03-27:
### The problem

An esp32cam board is wired with short leads to a FeatherWing ethernet board with an W5500 controller.
There is no SD card inserted on the esp32cam, so the HSPI bus is free.
ESPHome will crash on boot right after `Setting up Ethernet...`.

### Which version of ESPHome has the issue?

2024.3.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
esphome:
  name: esp32cam

esp32:
  board: esp32cam
  framework:
    type: arduino

logger:
  level: DEBUG

ethernet:
  type: W5500
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12
  cs_pin: GPIO15
```

### Anything in the logs that might be useful for us?

```txt
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:158]: Log initialized
[I][app:029]: Running through setup()...
[C][ethernet:034]: Setting up Ethernet...
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8ba3  PS      : 0x00060230  A0      : 0x800d93f2  A1      : 0x3ffb2540  
A2      : 0x00000003  A3      : 0x00000000  A4      : 0x00000087  A5      : 0x00000020  
A6      : 0x00000000  A7      : 0x3ffaef54  A8      : 0xc0100003  A9      : 0x00013ffc  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000087  A13     : 0x3ffb58cc  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000003  LBEG    : 0x40089734  LEND    : 0x4008973f  LCOUNT  : 0x00000000  

Backtrace:0x400d8ba0:0x3ffb25400x400d93ef:0x3ffb2570 0x400e919a:0x3ffb25a0 0x400e5846:0x3ffb25e0 0x400d3d4d:0x3ffb2640 0x40124681:0x3ffb2730 0x401246e5:0x3ffb2750 0x400d61a8:0x3ffb2770 0x400d72f4:0x3ffb27a0 0x400d8656:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

### Additional information

Boot is successful when setting up the generic SPI port instead or `ethernet`:

```yaml
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12
```

```
[I][logger:158]: Log initialized
[I][app:029]: Running through setup()...
[D][spi:039]: Setting up SPI bus...
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2024.3.0 compiled on Mar 27 2024, 23:21:05
[C][logger:166]: Logger:
[C][logger:167]:   Level: DEBUG
[C][logger:169]:   Log Baud Rate: 115200
[C][logger:170]:   Hardware UART: UART0
[C][spi:068]: SPI bus:
[C][spi:069]:   CLK Pin: GPIO14
[C][spi:070]:   SDI Pin: GPIO12
[C][spi:071]:   SDO Pin: GPIO13
[C][spi:076]:   Using HW SPI: SPI
```

Changing the SPI speed for `ethernet` did not help. For example I tried:
```yaml
ethernet:
  type: W5500
  clock_speed: 8Mhz
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12
  cs_pin: GPIO15
```

ismadiluk, 2024-03-30:
For me Esp32cam works well using GPIO02 as cs_pin

npdgm, 2024-03-31:
> For me Esp32cam works well using GPIO02 as cs_pin

Wow I feel stupid, thanks a lot it does work with other GPIOs as CS!
GPIO02 was preventing flashing over serial and I had to disconnect it temporarily, so I switched cs_pin to GPIO16 and that's better.
Cheers!

closed 2024-03-31 by npdgm