esphome/issues #5649: Unable to call dfrobot_sen0395.settings from within a number template set_action call

bakerkj, 2024-03-28:
### The problem

I am trying to use the relatively new dfrobot_sen0395 component. When I try to call dfrobot_sen0395.setting from within a number template set_action code generation completes, but the compiler fails to compile the generated code.

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

not relevant

### Component causing the issue

dfrobot_sen0395

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32dev

uart:
  id: uart_bus
  tx_pin: GPIO13
  rx_pin: GPIO14
  baud_rate: 115200

dfrobot_sen0395:
  - id: mmwave_one
    uart_id: uart_bus

number:
  - platform: template
    name: sensitivity
    min_value: 0
    max_value: 9
    initial_value: 7
    step: 1
    set_action:
      - dfrobot_sen0395.settings:
          id: mmwave_one
          sensitivity: 5
```

### Anything in the logs that might be useful for us?

```txt
This is the error I get when building...

(venv):~/esphome$ esphome run test.yaml                                                                                      
INFO ESPHome 2024.3.1                                                                                                                                     
INFO Reading configuration test.yaml...                                                                                                                   
INFO Generating C++ source...                                           
INFO Compiling app...                                                       
[... 219 lines cut ...]
Compiling .pioenvs/test/FrameworkArduino/StreamString.cpp.o
*** [.pioenvs/test/src/main.cpp.o] Error 1
============================================================================================================================================ [FAILED] Took 6.47 seconds ============================================================================================================================================

This is an extract of the main.cpp code which is failing:
  template__templatenumber = new template_::TemplateNumber();
  template__templatenumber->set_update_interval(60000);
  template__templatenumber->set_component_source("template.number");
  App.register_component(template__templatenumber);
  App.register_number(template__templatenumber);
  template__templatenumber->set_name("sensitivity");
  template__templatenumber->set_object_id("sensitivity");
  template__templatenumber->set_disabled_by_default(false);
  template__templatenumber->traits.set_min_value(0.0f);
  template__templatenumber->traits.set_max_value(9.0f);
  template__templatenumber->traits.set_step(1.0f);
  template__templatenumber->traits.set_mode(number::NUMBER_MODE_AUTO);
  template__templatenumber->set_optimistic(false);
  template__templatenumber->set_initial_value(7.0f);
  automation = new Automation<float>(template__templatenumber->get_set_trigger());
  dfrobot_sen0395_dfrobotsen0395settingsaction = new dfrobot_sen0395::DfrobotSen0395SettingsAction<float>();
  dfrobot_sen0395_dfrobotsen0395settingsaction->set_parent(mmwave_one);
  dfrobot_sen0395_dfrobotsen0395settingsaction->set_sensitivity(5);
  automation->add_actions({dfrobot_sen0395_dfrobotsen0395settingsaction});
```

bakerkj, 2024-04-17:
@niklasweber is there a chance you could help me with this?

Thanks!

niklasweber, 2024-04-17:
Hi @bakerkj, thank you for posting the issue. This seems strange. It seems to be related to the number component. When I use the yaml config you provided, I get the same error. But it works when I call the settings action from a button component:
```
button:
  - platform: template
    name: sensitivity
    on_press:
      - dfrobot_sen0395.settings:
          id: mmwave_one
          sensitivity: 5
```
I'm not too familiar with esphome templates. Somehow when using the number component, the template has two float values, which is not compatible with the action I defined in the component. @jesserockz do you have an idea why the settings action works for the button component, but not for the number component? Do I need to define another class in automation.h?

closed 2026-10-16 by bdraco