esphome/issues #5654: no "statistics" for some sensors

532910, 2024-03-29:
### The problem

weight_miscale has statistics

weight_user1, weight_user2, impedance_user1, impedance_user1 has no statistics

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

sonoff MiniR4

### Component causing the issue

template

### Example YAML snippet

```yaml
The second template from https://esphome.io/components/sensor/xiaomi_miscale
```

532910, 2024-03-29:
another example

`RSSI` HAS statistics, `Battery` HAS NOT:

```
sensor:
  - name: 'RSSI'
    entity_category: diagnostic
    platform: ble_client
    ble_client_id: ble_id
    type: rssi
    update_interval: '5s'

  - name: 'Battery'
    device_class: 'battery'
    unit_of_measurement: '%'
    platform: ble_client
    ble_client_id: ble_id
    type: characteristic
    service_uuid: '00000001-$uuid_base'
    characteristic_uuid: '00000041-$uuid_base'
    lambda: return (x[1]*0x100 + x[0]);
    notify: true
```

DAVe3283, 2024-03-30:
It looks like `ble_client` sensors don't have a `state_class` by default, since they could be returning any kind of information. You can try adding the appropriate state class explicitly:

```diff
 sensor:
   - name: 'RSSI'
     entity_category: diagnostic
     platform: ble_client
     ble_client_id: ble_id
     type: rssi
     update_interval: '5s'

   - name: 'Battery'
     device_class: 'battery'
+    state_class: measurement
     unit_of_measurement: '%'
     platform: ble_client
     ble_client_id: ble_id
     type: characteristic
     service_uuid: '00000001-$uuid_base'
     characteristic_uuid: '00000041-$uuid_base'
     lambda: return (x[1]*0x100 + x[0]);
     notify: true
```

532910, 2024-03-30:
Indeed, adding `state_class: measurement` to all these sensors make statistics works. Thank you, @DAVe3283!

I believe https://esphome.io/components/sensor/xiaomi_miscale should be updated before closing this issue.

closed 2024-08-12 by github-actions[bot]