esphome/issues #5655: Support newer Radon Eye devices

untergeek, 2024-03-29:
### The problem

In `radon_eye_listener.cpp` it _only_ listens for devices that start with `FR:R`. However, my brand new Radon Eye (firmware 3.0.1) has a device ID that starts with `FR:I`.

I edited `radon_eye_listener.cpp` to look for `FR:I` and my device was found easily.
```
[D][radon_eye_ble:016]: Found Radon Eye RD200 device Name: FR:IB[REDACTED] (MAC: REDACTED)
```

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

nodemcu-32s

### Component causing the issue

radon_eye_ble

untergeek, 2024-07-29:
Is there any way to make this easier? I am going to hook up a Python BLE client to the device to get the data directly. I may end up making a PR at some point.

jesserockz, 2024-07-29:
The listener is bery simple as you have found by editing it. I would happily merge a PR that adds checking for `FR:I` as well if you make one :pray:

Vituco, 2024-09-23:
Hello. I am having same exact problem but my device start with FR:H

Which version of ESPHome has the issue?
2024.9.1

What type of installation are you using?
HomeAssist OS 

Which version of Home Assistant has the issue?
2024.9.2

What platform are you using?
ESP32-IDF

Board:
esp32_s3

Component causing the issue
radon_eye_ble

baldisos, 2024-10-08:
I opened a Pull Request for that, i've added the "FR:I" and "FR:H" variant as possible Radon Eye devices.
[https://github.com/baldisos/esphome/blob/radon_eye_ble/esphome/components/radon_eye_ble/radon_eye_listener.cpp](url)

Feel free to test it with your device by adding the external component to your config:

```
external_components:
  - source: github://baldisos/esphome@radon_eye_ble
    components: [ radon_eye_ble ]`
```

untergeek, 2024-10-09:
I tried this, and it's doing something more, but perhaps not everything expected:

```
[15:33:54][W][radon_eye_rd200:141]: Connection in progress
[15:35:36][D][esp32_ble_tracker:270]: Starting scan...
[15:38:45][D][esp-idf:000][BTU_TASK]: W (4991866) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x8

[15:38:45][D][esp-idf:000][BTU_TASK]: W (4991876) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8

[15:38:45][D][esp-idf:000][BTU_TASK]: W (4991879) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x8

[15:38:45][D][esp-idf:000][BTU_TASK]: W (4991881) BT_APPL: gattc_conn_cb: if=6 st=0 id=6 rsn=0x8

[15:38:45][D][esp-idf:000][BTU_TASK]: W (4991884) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8

[15:38:45][D][esp32_ble_client:172]: [3] [08:D1:F9:3B:D2:5A] ESP_GATTC_DISCONNECT_EVT, reason 8
[15:38:45][W][radon_eye_rd200:021]: Disconnected!
[15:38:45][D][esp32_ble_client:110]: [3] [08:D1:F9:3B:D2:5A] ESP_GATTC_CLOSE_EVT
[15:38:45][D][esp32_ble_client:110]: [3] [08:D1:F9:3B:D2:5A] Found device
[15:38:45][D][esp32_ble_tracker:671]: Found device 08:D1:F9:3B:D2:5A RSSI=-51
[15:38:45][D][esp32_ble_tracker:692]:   Address Type: PUBLIC
[15:38:45][D][esp32_ble_tracker:694]:   Name: 'FR:IB06RE000109'
[15:38:45][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[15:38:45][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[15:38:45][D][esp-idf:000][BTU_TASK]: E (4992060) BT_BTM: BTM_BleScan scan not active

[15:38:45][D][esp-idf:000][BTU_TASK]: W (4992064) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

[15:38:45][I][esp32_ble_client:067]: [3] [08:D1:F9:3B:D2:5A] 0x00 Attempting BLE connection
[15:38:46][D][esp32_ble_client:110]: [3] [08:D1:F9:3B:D2:5A] ESP_GATTC_CONNECT_EVT
[15:38:46][D][esp32_ble_client:110]: [3] [08:D1:F9:3B:D2:5A] ESP_GATTC_OPEN_EVT
[15:38:46][I][radon_eye_rd200:015]: Connected successfully!
[15:38:46][D][esp32_ble_tracker:270]: Starting scan...
[15:38:46][D][esp32_ble_client:110]: [3] [08:D1:F9:3B:D2:5A] ESP_GATTC_SEARCH_CMPL_EVT
[15:38:46][I][esp32_ble_client:227]: [3] [08:D1:F9:3B:D2:5A] Connected
[15:38:46][W][radon_eye_rd200:030]: No sensor read characteristic found at service 00001523-1212-EFDE-1523-785FEABCD123 char 00001525-1212-EFDE-1523-785FEABCD123
[15:38:46][D][esp32_ble_client:188]: [3] [08:D1:F9:3B:D2:5A] cfg_mtu status 0, mtu 517
```

closed 2024-10-09 by jesserockz

ckuethe, 2024-10-09:
Better than it was, but I also see the `[04:07:27][W][radon_eye_rd200:030]: No sensor read characteristic found at service 00001523-1212-EFDE-1523-785FEABCD123 char 00001525-1212-EFDE-1523-785FEABCD123`

I can get a good data dump from my RD200V3 (fw 3.0.1) using https://github.com/sormy/radoneye-reader and it seems that different service and characteristic UUIDs are in use: [radoneye-dumper.py](https://github.com/sormy/radoneye-reader/blob/main/radoneye-dumper.py#L11)

baldisos, 2024-10-09:
That's a problem within the radon_eye_rd200 component tho. I only fixed the search for the prefixes within the _radon_eye_ble_ component. With the general problem of not being able to read data, i'm stuck myself at the moment.