esphome/issues #5657: "ERROR Could not match idedata" when attempting upload

GH15ADF, 2024-03-30:
### The problem

I completely reset my Windows 10 laptop so there would be no conflicts with any previous app installs (e.g., VS Code and PlatformIO extension, Python versions), downloaded and installed Python from https://www.python.org/downloads/. Followed the two pip3 commands as per https://esphome.io/guides/installing_esphome.html. 

From there, I plugged in my ESP32 board to a USB port and ran esphome upload ac-presence.yaml

The error from the "log" information is what is observed. I have tried this on two other machines and in Python virtual environments. All behave in the same way. I **CAN** upload this configuration to that board when connected directly to my Home Assistant server. However, I need this to be useful in a remote location for some development and testing of my ESPHome configuration/application.

### Which version of ESPHome has the issue?

ESPHome 2024.3.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

esphome upload 

### Example YAML snippet

```yaml
esphome:
  name: ac-presence
  friendly_name: ac-presence

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "some key"

ota:
  password: "who knows"

wifi:
  ssid: "mywifi"
  password: "apassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ac-Presence Fallback Hotspot"
    password: "something"

captive_portal:

remote_receiver:
  pin: GPIO32
  dump: all
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.3.1
INFO Reading configuration ac-presence.yaml...
Found multiple options for uploading, please choose one:
  [1] COM4 (USB-SERIAL CH340 (COM4))
  [2] Over The Air (ac-presence.local)
(number): 1
Usage: platformio run [OPTIONS]
Try 'platformio run -h' for help.

Error: Invalid value for '-d' / '--project-dir': Path '.esphome\\build\\ac-presence' does not exist.
ERROR Could not match idedata, please report this error
ERROR Stdout: ************************************************************************************************************************
If you like PlatformIO, please:
- star it on GitHub > https://github.com/platformio/platformio-core
- follow us on LinkedIn to stay up-to-date on the latest project news > https://www.linkedin.com/company/platformio/
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
************************************************************************************************************************

ERROR
```

binaryfrog, 2024-04-01:
Have you tried running `esphome run ac-presence.yaml` first?

GH15ADF, 2024-04-01:
Resolved, please close this ticket! Many thanks, my mistake! 
I went straight from a new manual install, through the Wizard and directly to upload. Never even thought about trying the other commands.

closed 2024-05-25 by GH15ADF