esphome/issues #5659: Boot loop

brianf21, 2024-03-31:
### The problem

[esp-web-tools-logs.txt](https://github.com/esphome/issues/files/14817129/esp-web-tools-logs.txt)

I am getting a boot loop. 

### Which version of ESPHome has the issue?

I am using ESPHome-econet version 1.23.0. I am not sure which version of ESPHome.

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

Atom RS-485 https://docs.m5stack.com/en/atom/atomic485

### Component causing the issue

mdns failed during boot

### Anything in the logs that might be useful for us?

```txt
[W][mdns:019]: mDNS init failed: ESP_ERR_INVALID_STATE
[E][component:114]: Component mdns was marked as failed.
[E][mdns:159]: Error set: unspecified

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x400837a5:0x3ffb23800x4008b6b5:0x3ffb23a0 0x40090bed:0x3ffb23c0 0x40105969:0x3ffb24f0 0x40115ee1:0x3ffb2520 0x40115f41:0x3ffb2540 0x40105015:0x3ffb2590 0x400e2b71:0x3ffb25b0 0x400e2baa:0x3ffb25d0 0x400e1849:0x3ffb25f0 0x401690b9:0x3ffb2630 0x40169155:0x3ffb2650 0x400e670c:0x3ffb2670 0x400ec0c2:0x3ffb26a0 0x400f080a:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[W][mdns:019]: mDNS init failed: ESP_ERR_INVALID_STATE
[E][component:114]: Component mdns was marked as failed.
[E][mdns:159]: Error set: unspecified

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x400837a5:0x3ffb23800x4008b6b5:0x3ffb23a0 0x40090bed:0x3ffb23c0 0x40105969:0x3ffb24f0 0x40115ee1:0x3ffb2520 0x40115f41:0x3ffb2540 0x40105015:0x3ffb2590 0x400e2b71:0x3ffb25b0 0x400e2baa:0x3ffb25d0 0x400e1849:0x3ffb25f0 0x401690b9:0x3ffb2630 0x40169155:0x3ffb2650 0x400e670c:0x3ffb2670 0x400ec0c2:0x3ffb26a0 0x400f080a:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[... 564 lines cut ...]
[E][component:114]: Component mdns was marked as failed.
[E][mdns:159]: Error set: unspecified

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x400837a5:0x3ffb23400x4008b6b5:0x3ffb2360 0x40090bed:0x3ffb2380 0x40105969:0x3ffb24b0 0x40115ee1:0x3ffb24e0 0x40115f41:0x3ffb2500 0x40105015:0x3ffb2550 0x400e2b71:0x3ffb2570 0x400e2baa:0x3ffb2590 0x400e1849:0x3ffb25b0 0x401690b9:0x3ffb25f0 0x40169155:0x3ffb2610 0x400e670c:0x3ffb2630 0x400e1b84:0x3ffb2660 0x400e9bad:0x3ffb26a0 0x400f080a:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

### Additional information

I was updating the firmware via this web interface. https://esphome-econet.github.io/install/

ssieb, 2024-03-31:
You are installing an unknown version of esphome with external components.  You need to file the issue with them first.

closed 2024-03-31 by ssieb

brianf21, 2024-03-31:
It was already open with them. https://github.com/esphome-econet/esphome-econet/issues/379

ssieb, 2024-04-01:
If you can reproduce it without the external component or have some way to verify that it's not caused by that or their compile, then it's a valid issue for here.