esphome/issues #5663: IR Receiver dump Midea/Coolix error

dimon90, 2024-04-01:
### The problem

The receiver can no longer reset Midea AC codes. He worked about 6 months ago, in the fall. I haven't used it since then, and now, after many updates, the codes are not recognized.
Only the OFF command can be recognized.

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

remote_receiver IR Midea dump

### Example YAML snippet

```yaml
esphome:
  name: childrenroom_climate
esp8266:
  board: d1_mini

logger:

api:
    encryption:
        key: !secret apikey

ota:
    password: !secret apipass

wifi:
  ssid: "IoT"
  password: !secret iotpass
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "childrenroom_climate_recovery"
    password: !secret iotpass

#captive_portal:

web_server:
  port: 80

remote_transmitter:
  pin: D8
  carrier_duty_percent: 80%

remote_receiver:
  id: ir_reciver
  pin:
    number: D2
    inverted: true
    mode:
      input: true
      pullup: true
  # high 55% tolerance is recommended for some remote control units
  tolerance: 55%
  dump: midea
  
climate:
  - platform: midea_ir
    name: "childrenroom_ac"
    sensor: childrenroom_tempsensor_temperature
    use_fahrenheit: false
    receiver_id: ir_reciver

sensor:
  - platform: homeassistant
    id: childrenroom_tempsensor_temperature
    entity_id: sensor.childrenroom_tempsensor_temperature
```

### Anything in the logs that might be useful for us?

```txt
# I continuously pressed On and Off. As you can see from the logs, only the OFF command is recognized. I also tried all possible commands from the remote control - no luck

[21:21:06][D][climate:396]: 'childrenroom_ac' - Sending state:
[21:21:06][D][climate:399]:   Mode: OFF
[21:21:06][D][climate:404]:   Fan Mode: AUTO
[21:21:06][D][climate:410]:   Preset: NONE
[21:21:06][D][climate:416]:   Swing Mode: OFF
[21:21:06][D][climate:419]:   Current Temperature: nan°C
[21:21:06][D][climate:425]:   Target Temperature: 24.00°C
[21:21:06][I][remote.coolix:104]: Received Coolix: 0xB27BE0
[21:21:12][D][climate:396]: 'childrenroom_ac' - Sending state:
[21:21:12][D][climate:399]:   Mode: OFF
[21:21:12][D][climate:404]:   Fan Mode: AUTO
[21:21:12][D][climate:410]:   Preset: NONE
[21:21:12][D][climate:416]:   Swing Mode: OFF
[21:21:12][D][climate:419]:   Current Temperature: nan°C
[21:21:12][D][climate:425]:   Target Temperature: 24.00°C
[21:21:12][I][remote.coolix:104]: Received Coolix: 0xB27BE0
[21:21:24][D][climate:396]: 'childrenroom_ac' - Sending state:
[21:21:24][D][climate:399]:   Mode: OFF
[21:21:24][D][climate:404]:   Fan Mode: AUTO
[21:21:24][D][climate:410]:   Preset: NONE
[21:21:24][D][climate:416]:   Swing Mode: OFF
[21:21:24][D][climate:419]:   Current Temperature: nan°C
[21:21:24][D][climate:425]:   Target Temperature: 24.00°C
[21:21:24][I][remote.coolix:104]: Received Coolix: 0xB27BE0
```

petergerard, 2024-04-24:
I’ve also been experiencing this issue. I tried to find where in the code a change may have happened but haven’t had much luck. I’m assuming there is a library dependency where a bug was introduced. Also don’t know which version of esphome this broke in, since I also don’t use the remotes often.

dimon90, 2024-06-06:
I'm fixed this issue by downloading docker image with older version of ESPHome 2023-05 and compiling .bin file. Now it's working.

`[13:49:25][D][climate:378]: 'bedroom_climate_ac' - Sending state:
[13:49:25][D][climate:381]:   Mode: COOL
[13:49:25][D][climate:386]:   Fan Mode: AUTO
[13:49:25][D][climate:392]:   Preset: NONE
[13:49:25][D][climate:398]:   Swing Mode: OFF
[13:49:25][D][climate:401]:   Current Temperature: nan°C
[13:49:25][D][climate:407]:   Target Temperature: 23.00°C
[13:49:29][D][climate:378]: 'bedroom_climate_ac' - Sending state:
[13:49:29][D][climate:381]:   Mode: OFF
[13:49:29][D][climate:386]:   Fan Mode: AUTO
[13:49:29][D][climate:392]:   Preset: NONE
[13:49:29][D][climate:398]:   Swing Mode: OFF
[13:49:29][D][climate:401]:   Current Temperature: nan°C
[13:49:29][D][climate:407]:   Target Temperature: 23.00°C
[13:49:30][D][climate:378]: 'bedroom_climate_ac' - Sending state:
[13:49:30][D][climate:381]:   Mode: COOL
[13:49:30][D][climate:386]:   Fan Mode: AUTO
[13:49:30][D][climate:392]:   Preset: NONE
[13:49:30][D][climate:398]:   Swing Mode: OFF
[13:49:30][D][climate:401]:   Current Temperature: nan°C
[13:49:30][D][climate:407]:   Target Temperature: 23.00°C
[13:49:48][D][climate:378]: 'bedroom_climate_ac' - Sending state:
[13:49:48][D][climate:381]:   Mode: OFF
[13:49:48][D][climate:386]:   Fan Mode: AUTO
[13:49:48][D][climate:392]:   Preset: NONE
[13:49:48][D][climate:398]:   Swing Mode: OFF
[13:49:48][D][climate:401]:   Current Temperature: nan°C
[13:49:48][D][climate:407]:   Target Temperature: 23.00°C
[13:49:48][D][sensor:093]: 'bedroom_climate_uptime': Sending state 34.04400 s with 0 decimals of accuracy
[13:49:49][D][climate:378]: 'bedroom_climate_ac' - Sending state:
[13:49:49][D][climate:381]:   Mode: COOL
[13:49:49][D][climate:386]:   Fan Mode: AUTO
[13:49:49][D][climate:392]:   Preset: NONE
[13:49:49][D][climate:398]:   Swing Mode: OFF
[13:49:49][D][climate:401]:   Current Temperature: nan°C
[13:49:49][D][climate:407]:   Target Temperature: 23.00°C`

mmiller7, 2024-09-15:
Not sure if I'm affected by this, just got a new whynter heat pump which seems to use the mieda_ir codes to control it.  Can't seem to get the receiver working and I can't figure out why.

petergerard, 2024-09-16:
Right, this seems to affect any unit using `midea_ir` platform (which is apparently used by a number of brands). The bug was introduced some time after the 2023-05 version. Unfortunately I don't have a straightforward way to go back to that version, due to other compatibility issues, so I've hid my remotes in a drawer.