esphome/issues #5664: RPI_DPI_RGB display will not compile

LilDucky, 2024-04-02:
### The problem

I am attempting to configure the RPI_DPI_RGB display driver for the ESP32-8048S070 development board. This has an ESP32-S3, 7 inch RGB display, GT911 touch screen and a plethora of other handy features.

**The problem**
When compiling a basic YAML configuration I get the following error.

```
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp: In member function 'virtual void esphome::rpi_dpi_rgb::RpiDpiRgb::setup()':
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:10:3: error: 'esp_lcd_rgb_panel_config_t' was not declared in this scope
   esp_lcd_rgb_panel_config_t config{};
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:10:3: note: suggested alternative: 'esp_lcd_panel_init'
   esp_lcd_rgb_panel_config_t config{};
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
   esp_lcd_panel_init
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:11:3: error: 'config' was not declared in this scope
   config.flags.fb_in_psram = 1;
   ^~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:11:3: note: suggested alternative: 'confstr'
   config.flags.fb_in_psram = 1;
   ^~~~~~
   confstr
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:35:19: error: 'esp_lcd_new_rgb_panel' was not declared in this scope
   esp_err_t err = esp_lcd_new_rgb_panel(&config, &this->handle_);
                   ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:35:19: note: suggested alternative: 'esp_lcd_panel_set_gap'
   esp_err_t err = esp_lcd_new_rgb_panel(&config, &this->handle_);
                   ^~~~~~~~~~~~~~~~~~~~~
                   esp_lcd_panel_set_gap
*** [.pioenvs\displaytest\src\esphome\components\rpi_dpi_rgb\rpi_dpi_rgb.o] Error 1
Compiling .pioenvs\displaytest\src\esphome\core\helpers.o
```

### Which version of ESPHome has the issue?

latest and dev. Clean install of python, ESPHome etc did not change the issue

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

ESP32-8048S070

### Component causing the issue

RPI_DPI_RGB display driver

### Example YAML snippet

```yaml
substitutions:
  esphome_name: displaytest

esphome:
  name: "${esphome_name}"
  
esp32:
  board: esp32dev
  variant: esp32s3
[... 225 lines cut ...]
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:10:3: error: 'esp_lcd_rgb_panel_config_t' was not declared in this scope
   esp_lcd_rgb_panel_config_t config{};
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:10:3: note: suggested alternative: 'esp_lcd_panel_init'
   esp_lcd_rgb_panel_config_t config{};
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
   esp_lcd_panel_init
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:11:3: error: 'config' was not declared in this scope
   config.flags.fb_in_psram = 1;
   ^~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:11:3: note: suggested alternative: 'confstr'
   config.flags.fb_in_psram = 1;
   ^~~~~~
   confstr
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:35:19: error: 'esp_lcd_new_rgb_panel' was not declared in this scope
   esp_err_t err = esp_lcd_new_rgb_panel(&config, &this->handle_);
                   ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/rpi_dpi_rgb/rpi_dpi_rgb.cpp:35:19: note: suggested alternative: 'esp_lcd_panel_set_gap'
   esp_err_t err = esp_lcd_new_rgb_panel(&config, &this->handle_);
                   ^~~~~~~~~~~~~~~~~~~~~
                   esp_lcd_panel_set_gap
*** [.pioenvs\displaytest\src\esphome\components\rpi_dpi_rgb\rpi_dpi_rgb.o] Error 1
Compiling .pioenvs\displaytest\src\esphome\core\helpers.o
================================================================ [FAILED] Took 77.70 seconds ================================================================
```

LilDucky, 2024-04-06:
Built a docker instance of ESPhome and get the same issue trying to build.

Does not look platform related to me.

clydebarrow, 2024-04-06:
You need to choose a board that has an ESP32-S3. `esp32dev` is an ESP32, not an S3. The `variant` setting does not override that.

closed 2024-04-06 by clydebarrow

martijnbuts, 2024-04-09:
Did you test the config again with the esp32-s3 board and does the screen work now?

martijnbuts, 2024-04-16:
> > @LilDucky Did you test the config again with the esp32-s3 board and does the screen work now? I can compile the config with esp32: board: esp32-s3-devkitc-1 variant: esp32s3 framework: type: esp-idf platform_version: 6.6.0 version: 5.1.2 #board: ESP32-8048S070 sdkconfig_options:
> > but i do not get a correct screen image [esp32-large-display (2).txt](https://github.com/esphome/issues/files/14920342/esp32-large-display.2.txt)
> 
> It compiles for me but I get a band of random pixels across the bottom 10mm of the display.

Please try this config, it works for me
https://github.com/clowrey/esphome-sunton-esp32-8048s070c