esphome/issues #5667: Similar to #3396, having issues with PlatformIO being incompatible

26tajeen, 2024-04-02:
### The problem

Made the fatal mistake of running an upgrade command on esphome and it broke the installation of my Catalina Mac.

### Which version of ESPHome has the issue?

Version: 2022.2.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

ESP32-POE

### Component causing the issue

CLI

### Example YAML snippet

```yaml
n/a
```

### Anything in the logs that might be useful for us?

```txt
macbookpro:~ $ esphome run podesp.yaml 
INFO Reading configuration podesp.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/__main__.py", line 369, in command_run
    exit_code = compile_program(args, config)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/__main__.py", line 186, in compile_program
    rc = platformio_api.run_compile(config, CORE.verbose)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/platformio_api.py", line 106, in run_compile
    return run_platformio_cli_run(config, verbose)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/platformio_api.py", line 102, in run_platformio_cli_run
    return run_platformio_cli(*command, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/platformio_api.py", line 93, in run_platformio_cli
    patch_structhash()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/esphome/platformio_api.py", line 23, in patch_structhash
    from platformio.commands.run import helpers, command
ModuleNotFoundError: No module named 'platformio.commands.run'
macbookpro:~ stevejones$ esphome run podesp.yaml 
macbookpro:~ stevejones$ pip3 pip3 install esphome -U
ERROR: unknown command "pip3"
macbookpro:~ stevejones$ pip3 install esphome -U
Requirement already satisfied: esphome in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (2022.2.6)
Requirement already satisfied: voluptuous==0.12.2 in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (from esphome) (0.12.2)
[... 539 lines cut ...]
esphome 2022.2.6 requires click==8.0.3, but you have click 8.1.7 which is incompatible.
esphome 2022.2.6 requires zeroconf==0.38.3, but you have zeroconf 0.38.4 which is incompatible.
Successfully installed zeroconf-0.38.4
macbookpro:~ stevejones$ esphome run podesp.yaml 
INFO Reading configuration podesp.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
***************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
***************************************************************************************************************************************************************************

Processing podesp (board: esp32-poe; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.5.0
Error: Please upgrade to the PlatformIO Core 6
macbookpro:~ stevejones$ esphome version
Version: 2022.2.6
macbookpro:~ stevejones$
```

sbrk-r, 2024-04-05:
> 2022.2.6
Is there a reason you are running a 2+ year old release?

26tajeen, 2024-04-06:
I’ve never manually tried to update it until now and it’s never prompted me to update it or updated itself automatically.

It’s precisely because I thought “oh, I’ll update esphome cli - might as well” that my installation is now temporarily broken. It was working fine before I tried the update.

closed 2024-08-20 by github-actions[bot]