esphome/issues #5669: offset filter not working with aht platform

cuthulino, 2024-04-03:
### The problem

I am using the following config for reading an aht sensor.
It needs a little offset but it is not working with this platform.
```
sensor:
  - platform: aht10
    id: aht_in
    i2c_id: bus_in
    variant: AHT20
    temperature:
      id: ga1_t_in
      name: temp in
      filters:
        - offset: 0.7
      on_value:
        then:
          - lambda: |-
              id(knxd).group_write_2byte_float("2/4/230", x);
    humidity:
      id: ga1_h_in
      name: hum in
      on_value:
        then:
          - lambda: |-
              id(knxd).group_write_2byte_float("2/5/230", x);
    update_interval: 120s
```

Testing it with a template sensor works without problems:
```
  - platform: template
    name: test
    lambda: "return id(ga1_t_in).state;"
    filters:
      - offset: 10
```

I already read that sometimes, after flashing via the cable, the offset works.
I tried it but it doesnt work.

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

Espressif ESP-32 / ESP-WROOM-32

### Component causing the issue

sensor/aht10

ssieb, 2024-04-03:
How can you tell it isn't working?

cuthulino, 2024-04-04:
Adding or removing the filter does not alter the sensor value.
I tried with a high, good recognizable offset value and the output value in esphome logs and homeassistant do not change.

ssieb, 2024-04-07:
Add an `on_raw_value:` trigger to print the raw value and show the logs.

cuthulino, 2024-04-17:
After adding the logger lines i got this output:
```
[E][main:101]: The temperature sensor reports raw_value 22.8 
[E][main:095]: The temperature sensor reports value 23.5 
```

So the offset is set.
Strange is the reported Temp in Homeassistant, it stepped up after I added the `on_raw_value` line:
[image]

closed 2024-04-17 by cuthulino