esphome/issues #5670: Button activate alone 1 minute after press

macledefa, 2024-04-03:
### The problem

The problem is that after every press of the button, the code works well, turning on my relay. But exactly 1 minute after, the relay activate for like 6 / 7 seconds on its own.

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

esp8285

### Additional information

I am using this board :

ssieb, 2024-04-03:
What do the logs show?

macledefa, 2024-04-04:
Basically nothing, even in VERY_VERBOSE level ...

`[11:06:53][VV][api.service:576]: on_ping_request: PingRequest {}
[11:06:53][VV][api.service:043]: send_ping_response: PingResponse {}
[11:07:11][VV][api.service:576]: on_ping_request: PingRequest {}
[11:07:11][VV][api.service:043]: send_ping_response: PingResponse {}
[11:07:29][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=80816 (now=140816)
[11:07:42][VV][api.service:784]: on_button_command_request: ButtonCommandRequest {
  key: 46977646
}
[11:07:42][D][button:010]: 'Allumer mon PC' Pressed.
[11:07:42][VV][scheduler:032]: set_timeout(name='', timeout=500)
[11:07:42][VV][scheduler:225]: Running timeout '' with interval=500 last_execution=154244 (now=154747)
[11:07:51][VV][api.service:576]: on_ping_request: PingRequest {}
[11:07:51][VV][api.service:043]: send_ping_response: PingResponse {}
[11:07:53][VV][api.connection:132]: Sending keepalive PING...
[11:07:53][VV][api.service:037]: send_ping_request: PingRequest {}
[11:07:53][VV][api.service:585]: on_ping_response: PingResponse {}
[11:08:29][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=140816 (now=200818)
[11:08:31][VV][api.service:576]: on_ping_request: PingRequest {}
[11:08:31][VV][api.service:043]: send_ping_response: PingResponse {}
[11:08:53][VV][api.service:576]: on_ping_request: PingRequest {}
[11:08:53][VV][api.service:043]: send_ping_response: PingResponse {}
[11:09:11][VV][api.service:576]: on_ping_request: PingRequest {}
[11:09:11][VV][api.service:043]: send_ping_response: PingResponse {}`

The relay as turn on 1 minute after my button press, at 11:08:42

ssieb, 2024-04-07:
You're using an output which doesn't log anything.  Try using a switch instead.

macledefa, 2024-04-10:
Ok, i've update my YAML to use this code :

'''YAML
switch:
  - platform: template
    name: "Allumer mon PC"
    turn_on_action:
      - logger.log: "Switch Turned On!"
      - output.turn_on: out_relay
      - delay: 0.5s
      - output.turn_off: out_relay
    turn_off_action:
      - logger.log: "Switch Turned Off!"
'''

but, with the same problem, I still don't have much info in the log (still in VERY_VERBOSE level) :

'''YAML
[11:54:46][VV][api.service:043]: send_ping_response: PingResponse {}
[11:55:01][VV][api.service:664]: on_switch_command_request: SwitchCommandRequest {
  key: 46977646
  state: YES
}
[11:55:01][D][switch:012]: 'Allumer mon PC' Turning ON.
[11:55:01][D][main:249]: Switch Turned On!
[11:55:01][VV][scheduler:032]: set_timeout(name='', timeout=500)
[11:55:02][VV][scheduler:225]: Running timeout '' with interval=500 last_execution=179142 (now=179646)
[11:55:08][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=125367 (now=185375)
[11:55:23][VV][api.connection:132]: Sending keepalive PING...
[11:55:23][VV][api.service:037]: send_ping_request: PingRequest {}
[11:55:23][VV][api.service:585]: on_ping_response: PingResponse {}
[11:55:26][VV][api.service:576]: on_ping_request: PingRequest {}
[11:55:26][VV][api.service:043]: send_ping_response: PingResponse {}
[11:56:03][VV][api.service:576]: on_ping_request: PingRequest {}
[11:56:03][VV][api.service:043]: send_ping_response: PingResponse {}
[11:56:06][VV][api.service:576]: on_ping_request: PingRequest {}
[11:56:06][VV][api.service:043]: send_ping_response: PingResponse {}
[11:56:08][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=185367 (now=245371)
'''

ssieb, 2024-04-10:
That's not what I meant.  The `button` was correct.  But use a `switch` instead of an `output`.

macledefa, 2024-04-13:
Sorry I didn't understood correctly !

I've changed my code to this : 

```YAML
button:
  - platform: template
    name: "Allumer mon PC"
    id: sw_pc_power
    on_press:
      - switch.turn_on: out_switch
      - delay: 0.5s
      - switch.turn_off: out_switch

switch:
  - platform: gpio
    internal: True
    id: out_switch
    pin: 12
    on_turn_on:
    - logger.log: "Switch Turned On!"
    on_turn_off:
    - logger.log: "Switch Turned Off!"
```
but it changed nothing with my problem, and still nothing in the log ...

```YAML
[11:26:43][VV][api.service:576]: on_ping_request: PingRequest {}
[11:26:43][VV][api.service:043]: send_ping_response: PingResponse {}
[11:26:49][VV][api.service:784]: on_button_command_request: ButtonCommandRequest {
  key: 46977646
}
[11:26:49][D][button:010]: 'Allumer mon PC' Pressed.
[11:26:49][D][switch:012]: 'out_switch' Turning ON.
[11:26:49][D][switch:055]: 'out_switch': Sending state ON
[11:26:49][D][main:275]: Switch Turned On!
[11:26:49][VV][scheduler:032]: set_timeout(name='', timeout=500)
[11:26:50][VV][scheduler:225]: Running timeout '' with interval=500 last_execution=89178 (now=89683)
[11:26:50][D][switch:016]: 'out_switch' Turning OFF.
[11:26:50][D][switch:055]: 'out_switch': Sending state OFF
[11:26:50][D][main:281]: Switch Turned Off!
[11:27:01][VV][api.connection:132]: Sending keepalive PING...
[11:27:01][VV][api.service:037]: send_ping_request: PingRequest {}
[11:27:01][VV][api.service:585]: on_ping_response: PingResponse {}
[11:27:23][VV][api.service:576]: on_ping_request: PingRequest {}
[11:27:23][VV][api.service:043]: send_ping_response: PingResponse {}
[11:27:34][VV][scheduler:225]: Running interval '' with interval=60000 last_execution=73769 (now=133774)
[11:28:01][VV][api.connection:132]: Sending keepalive PING...
[11:28:01][VV][api.service:037]: send_ping_request: PingRequest {}
[11:28:01][VV][api.service:585]: on_ping_response: PingResponse {}
[11:28:03][VV][api.service:576]: on_ping_request: PingRequest {}
[11:28:03][VV][api.service:043]: send_ping_response: PingResponse {}
```

ssieb, 2024-04-14:
Then it's not something coming from esphome.