esphome/issues #5678: OTA Never Ever Works on ESP32-S3

luckylinux, 2024-04-06:
### The problem

I tried several times OTA and it Never Ever works (at least for me and on an ESP32-S3).

Looking at some threads here it is, most likely, a WiFi AP Configuration issue:
- https://github.com/esphome/issues/issues/3318
- https://github.com/esphome/issues/issues/734
- https://community.home-assistant.io/t/wth-does-esphome-wireless-update-fail-so-often/471646/27
- ...

I am running my ESP32 off a Homemade AP off a Rock 5B ARM64 SBC with Alfa AWUS1900 USB-Wifi Adapter (RTL8814AU) with Hostapd + DNSmasq.

The devices have no problem being connected and staying connected. Ping works.

I also tried to Reboot the ESP32 (Normal Mode or Safe Mode) prior to performing the OTA.

Nope. It still won't work.

Host DNS Resolution works correctly (the Hostname resolves to the correct IP address).

I tried version: 1 and version: 2 for the OTA section, both time out.

Web Server OTA is not supported on esp-idf framework so that's out of the window too.

This is what I have in the OTA section:
```
ota:
  safe_mode: true
  version: 2 # Also tried version: 1, same result
  password: !secret ota_password
```

Anything that can be done to get OTA to work ?

The error message just reports:
```
ERROR Error receiving acknowledge version: timed out
```

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

Atom S3 Lite

### Component causing the issue

ota

### Anything in the logs that might be useful for us?

```txt
ERROR Error receiving acknowledge version: timed out
```

ssieb, 2024-04-06:
What does the device log show when you try to update it?

luckylinux, 2024-04-06:
@ssieb 

How should I see it ?

It's offline since I tried OTA ... it didn't come back up.

The only thing I can see on MQTT is with
`mosquitto_sub -i UbuntuClient -h 192.168.4.10 -p 1883 -t inverter/deye-a/debug`

I get only:
```
[W][component:233]: Components should block for at most 30 ms.
```

`web_server` is currently NOT enabled.

Or do you mean trying to perform OTA when it's plugged in via USB port ?

ssieb, 2024-04-06:
You didn't say anything about it not coming back up.  If the ota update fails, it should continue with the original version.
You also didn't provide the config.  Are you saying you don't have the api configured and only have mqtt?
But if it's not coming up at all, then you will need to get serial logs to find out what's happening.

luckylinux, 2024-04-06:
@ssieb

The complete config YAML is available in the "Example YAML snippet` Section.

I do not have a single file, it's a packaged version.

Thus to reproduce the easiest is on Debian/Ubuntu (or other GNU/Linux Distribution):
```
# The issue appears within several projects but this is an easy replicate method
# The other projects also include BLE and many other files to be compiled
# So this is much faster to replicate in terms of compilation time
git clone https://github.com/luckylinux/esphome-for-deye.git
cd esphome-for-deye
cp secrets.sh.example secrets.sh
./build.sh
# Select the desired configuration (anything you want)
```

luckylinux, 2024-04-06:
Could it be a low-RAM issue ?

The Atom S3 Lite (unlike many ESP32 / ESP32-S3 modules e.g. -WROVER or -WROOM) does NOT have an external PSRAM (0 MB).

It has a ESP32-S3FN8 SOC 8 MB Flash and approx. integrated 320 kB RAM.

Although according to Wikipedia it should have integrated 520 kiB RAM:
https://en.wikipedia.org/wiki/ESP32#ESP32-S3

```
[sudo] password for USER: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3.11-venv is already the newest version (3.11.6-3).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Requirement already satisfied: esphome in ./venv/lib/python3.11/site-packages (2024.3.1)
Requirement already satisfied: cryptography==42.0.2 in ./venv/lib/python3.11/site-packages (from esphome) (42.0.2)
Requirement already satisfied: voluptuous==0.14.2 in ./venv/lib/python3.11/site-packages (from esphome) (0.14.2)
Requirement already satisfied: PyYAML==6.0.1 in ./venv/lib/python3.11/site-packages (from esphome) (6.0.1)
Requirement already satisfied: paho-mqtt==1.6.1 in ./venv/lib/python3.11/site-packages (from esphome) (1.6.1)
Requirement already satisfied: colorama==0.4.6 in ./venv/lib/python3.11/site-packages (from esphome) (0.4.6)
Requirement already satisfied: icmplib==3.0.4 in ./venv/lib/python3.11/site-packages (from esphome) (3.0.4)
Requirement already satisfied: tornado==6.4 in ./venv/lib/python3.11/site-packages (from esphome) (6.4)
Requirement already satisfied: tzlocal==5.2 in ./venv/lib/python3.11/site-packages (from esphome) (5.2)
Requirement already satisfied: tzdata>=2021.1 in ./venv/lib/python3.11/site-packages (from esphome) (2023.4)
Requirement already satisfied: pyserial==3.5 in ./venv/lib/python3.11/site-packages (from esphome) (3.5)
Requirement already satisfied: platformio==6.1.13 in ./venv/lib/python3.11/site-packages (from esphome) (6.1.13)
Requirement already satisfied: esptool==4.7.0 in ./venv/lib/python3.11/site-packages (from esphome) (4.7.0)
Requirement already satisfied: click==8.1.7 in ./venv/lib/python3.11/site-packages (from esphome) (8.1.7)
Requirement already satisfied: esphome-dashboard==20240319.0 in ./venv/lib/python3.11/site-packages (from esphome) (20240319.0)
Requirement already satisfied: aioesphomeapi==23.1.1 in ./venv/lib/python3.11/site-packages (from esphome) (23.1.1)
Requirement already satisfied: zeroconf==0.131.0 in ./venv/lib/python3.11/site-packages (from esphome) (0.131.0)
Requirement already satisfied: python-magic==0.4.27 in ./venv/lib/python3.11/site-packages (from esphome) (0.4.27)
Requirement already satisfied: ruamel.yaml==0.18.6 in ./venv/lib/python3.11/site-packages (from esphome) (0.18.6)
Requirement already satisfied: kconfiglib==13.7.1 in ./venv/lib/python3.11/site-packages (from esphome) (13.7.1)
Requirement already satisfied: pyparsing>=3.0 in ./venv/lib/python3.11/site-packages (from esphome) (3.1.1)
Requirement already satisfied: argcomplete>=2.0.0 in ./venv/lib/python3.11/site-packages (from esphome) (3.2.1)
Requirement already satisfied: aiohappyeyeballs>=2.3.0 in ./venv/lib/python3.11/site-packages (from aioesphomeapi==23.1.1->esphome) (2.3.2)
Requirement already satisfied: async-interrupt>=1.1.1 in ./venv/lib/python3.11/site-packages (from aioesphomeapi==23.1.1->esphome) (1.1.1)
Requirement already satisfied: protobuf>=3.19.0 in ./venv/lib/python3.11/site-packages (from aioesphomeapi==23.1.1->esphome) (4.25.2)
Requirement already satisfied: chacha20poly1305-reuseable>=0.12.1 in ./venv/lib/python3.11/site-packages (from aioesphomeapi==23.1.1->esphome) (0.12.1)
Requirement already satisfied: noiseprotocol<1.0,>=0.3.1 in ./venv/lib/python3.11/site-packages (from aioesphomeapi==23.1.1->esphome) (0.3.1)
Requirement already satisfied: cffi>=1.12 in ./venv/lib/python3.11/site-packages (from cryptography==42.0.2->esphome) (1.16.0)
Requirement already satisfied: bitstring>=3.1.6 in ./venv/lib/python3.11/site-packages (from esptool==4.7.0->esphome) (4.1.4)
Requirement already satisfied: ecdsa>=0.16.0 in ./venv/lib/python3.11/site-packages (from esptool==4.7.0->esphome) (0.18.0)
Requirement already satisfied: reedsolo<1.8,>=1.5.3 in ./venv/lib/python3.11/site-packages (from esptool==4.7.0->esphome) (1.7.0)
Requirement already satisfied: intelhex in ./venv/lib/python3.11/site-packages (from esptool==4.7.0->esphome) (2.3.0)
Requirement already satisfied: bottle==0.12.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (0.12.25)
Requirement already satisfied: marshmallow==3.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (3.20.2)
Requirement already satisfied: pyelftools==0.30 in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (0.30)
Requirement already satisfied: requests==2.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (2.31.0)
Requirement already satisfied: semantic-version==2.10.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (2.10.0)
Requirement already satisfied: tabulate==0.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (0.9.0)
Requirement already satisfied: ajsonrpc==1.2.* in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (1.2.0)
Requirement already satisfied: starlette<0.36,>=0.19 in ./venv/lib/python3.11/site-packages (from platformio==6.1.13->esphome) (0.31.1)
[... cut ...]
Wrote 0x12f0d0 bytes to file /home/USER/ESPHome/esphome-for-deye/.esphome/build/deye-c/.pioenvs/deye-c/firmware-factory.bin, ready to flash to offset 0x0
================================================================= [SUCCESS] Took 70.21 seconds =================================================================
INFO Successfully compiled program.
INFO Resolving IP address of deye-c.solar.MYDOMAIN.TLD
INFO  -> 172.22.20.3
INFO Uploading .esphome/build/deye-c/.pioenvs/deye-c/firmware.bin (1175760 bytes)
ERROR Error receiving acknowledge version: timed out

```

luckylinux, 2024-04-06:
Otherwise could it be that `deep_sleep` is enabled by default ? I don't have anything in the YAML file in that regards ...

It doesn't state anything in the docs, but it seems you HAVE to declare that block yourself:
https://esphome.io/components/deep_sleep.html#esp32-wakeup-cause

luckylinux, 2024-04-06:
I tried pinging the ESP32 from a Desktop computer sitting in the House to the ESP32 in the Garage through a Power Line adapter:

Ubuntu Desktop <--> ~ 10 meters Cat6 Cable <--> Power Line Adapter <--> ~ 10 meters Power Cord <--> Power Line Adapter <--> ~ 2 meters Cat 6 Cable <--> Rock 5B running MQTT/dnsmasq/hostapd <--> Alfa AWUS1900 USB-Wifi Adapter (RTL8814AU) <--> ESP32

`sudo ping -l 1492 deye-a.solar.MYDOMAIN.TLD` yields

```
[sudo] password for USER: 
PING deye-a.solar.MYDOMAIN.TLD (172.22.20.1) 56(84) bytes of data.
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=1 ttl=254 time=8.87 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=2 ttl=254 time=9.82 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=3 ttl=254 time=9.82 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=4 ttl=254 time=11.1 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=5 ttl=254 time=11.1 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=6 ttl=254 time=15.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=7 ttl=254 time=16.2 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=8 ttl=254 time=21.2 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=9 ttl=254 time=22.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=10 ttl=254 time=22.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=11 ttl=254 time=35.9 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=12 ttl=254 time=35.9 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=13 ttl=254 time=36.9 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=14 ttl=254 time=18.8 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=15 ttl=254 time=6.69 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=16 ttl=254 time=17.8 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=17 ttl=254 time=6.49 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=18 ttl=254 time=6.59 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=19 ttl=254 time=6.63 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=20 ttl=254 time=6.46 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=21 ttl=254 time=6.66 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=22 ttl=254 time=6.78 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=23 ttl=254 time=7.65 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=24 ttl=254 time=6.40 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=25 ttl=254 time=17.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=26 ttl=254 time=8.81 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=27 ttl=254 time=13.0 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=28 ttl=254 time=6.81 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=29 ttl=254 time=17.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=30 ttl=254 time=6.90 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=31 ttl=254 time=6.27 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=32 ttl=254 time=6.97 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=33 ttl=254 time=6.71 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=34 ttl=254 time=7.17 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=35 ttl=254 time=6.69 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=36 ttl=254 time=6.84 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=37 ttl=254 time=8.43 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=38 ttl=254 time=41.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=39 ttl=254 time=6.67 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=40 ttl=254 time=13.6 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=41 ttl=254 time=6.35 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=42 ttl=254 time=19.5 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=43 ttl=254 time=6.55 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=44 ttl=254 time=8.48 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=45 ttl=254 time=12.6 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=46 ttl=254 time=9.14 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=47 ttl=254 time=7.78 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=48 ttl=254 time=7.47 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=49 ttl=254 time=7.71 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=50 ttl=254 time=41.9 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=51 ttl=254 time=7.48 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=52 ttl=254 time=87.2 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=53 ttl=254 time=28.4 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=54 ttl=254 time=6.79 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=55 ttl=254 time=6.75 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=56 ttl=254 time=6.74 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=57 ttl=254 time=8.42 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=58 ttl=254 time=8.65 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=59 ttl=254 time=19.1 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=60 ttl=254 time=7.62 ms
64 bytes from 172.22.20.1 (172.22.20.1): icmp_seq=61 ttl=254 time=6.97 ms
[... 416 lines cut ...]
64 bytes from 192.168.4.10: icmp_seq=231 ttl=64 time=3.48 ms
64 bytes from 192.168.4.10: icmp_seq=232 ttl=64 time=3.47 ms
64 bytes from 192.168.4.10: icmp_seq=233 ttl=64 time=3.50 ms
64 bytes from 192.168.4.10: icmp_seq=234 ttl=64 time=3.42 ms
64 bytes from 192.168.4.10: icmp_seq=235 ttl=64 time=3.35 ms
64 bytes from 192.168.4.10: icmp_seq=236 ttl=64 time=3.50 ms
64 bytes from 192.168.4.10: icmp_seq=237 ttl=64 time=2.65 ms
[... cut ...]
64 bytes from 192.168.4.10: icmp_seq=245 ttl=64 time=3.29 ms
64 bytes from 192.168.4.10: icmp_seq=246 ttl=64 time=3.21 ms
64 bytes from 192.168.4.10: icmp_seq=247 ttl=64 time=3.47 ms
64 bytes from 192.168.4.10: icmp_seq=248 ttl=64 time=1.78 ms
64 bytes from 192.168.4.10: icmp_seq=249 ttl=64 time=3.62 ms
64 bytes from 192.168.4.10: icmp_seq=250 ttl=64 time=3.02 ms
^C
--- 192.168.4.10 ping statistics ---
250 packets transmitted, 250 received, 0% packet loss, time 237004ms
rtt min/avg/max/mdev = 1.725/3.386/8.194/0.795 ms, pipe 13
```

luckylinux, 2024-04-06:
But note that OTA was also failing when directly connected from within the garage (so EXCLUDING any of this Power Line adapters etc).

luckylinux, 2024-04-07:
I will need to cross-check on these specific devices, but I could replicate the PROBLEM on a Raspberry Pi 2 (armv7l) with the same hostapd+dnsmasq configuration and the same Alfa AWUS1900 USB-Wifi Adapter (RTL8814AU).

For those having issues with OTA failing to upload, like at 4% - 10%  or so, the FIRST thing to check if your network connectivity, configuration, routing, firewall, etc. That was also what the other GitHub Issues / Home Assistant Forums seemed to suggest.

I lost 2+ days trying to sort out why OTA would fail, the watchdog getting triggered, then the device BRICKS without even resetting itself.

In some cases cycling USB Power Supply is enough to make it recover. But in some/most cases, you need to USB flash AGAIN.

The network connectivity seems fine, although if you ping long enough you can see the latency increasing dramatically.

But there is NOT really a network "failure" per se ...

[CODE]root@MYHOSTNAME:~# ping -c 20 esp32-s3-base.local

root@MYHOSTNAME:~# ping -c 20 esp32-s3-base.local

PING esp32-s3-base.local (172.27.30.1) 56(84) bytes of data.

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=1 ttl=254 time=87.4 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=2 ttl=254 time=8.65 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=3 ttl=254 time=31.1 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=4 ttl=254 time=55.2 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=5 ttl=254 time=86.5 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=6 ttl=254 time=102 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=7 ttl=254 time=30.7 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=8 ttl=254 time=53.6 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=9 ttl=254 time=77.2 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=10 ttl=254 time=102 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=11 ttl=254 time=25.2 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=12 ttl=254 time=45.5 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=13 ttl=254 time=1810 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=14 ttl=254 time=787 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=15 ttl=254 time=296 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=16 ttl=254 time=421 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=17 ttl=254 time=750 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=18 ttl=254 time=466 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=19 ttl=254 time=285 ms

64 bytes from esp32-s3-base.local (172.27.30.1): icmp_seq=20 ttl=254 time=102 ms

--- esp32-s3-base.local ping statistics ---

20 packets transmitted, 20 received, 0% packet loss, time 19036ms

rtt min/avg/max/mdev = 8.649/281.088/1809.820/418.520 ms, pipe 2

root@MYHOSTNAME:~# traceroute esp32-s3-base.local

traceroute to esp32-s3-base.local (172.27.30.1), 30 hops max, 60 byte packets

 1  192.168.1.7 (192.168.1.7)  0.513 ms  0.431 ms  0.436 ms

 2  192.168.4.31 (192.168.4.31)  0.825 ms  0.884 ms  0.989 ms

 3  esp32-s3-base.local (172.27.30.1)  62.587 ms  66.768 ms  66.756 ms

root@MYHOSTNAME:~# tcptraceroute esp32-s3-base.local

Running:

    traceroute -T -O info esp32-s3-base.local

traceroute to esp32-s3-base.local (172.27.30.1), 30 hops max, 60 byte packets

 1  192.168.1.7 (192.168.1.7)  0.450 ms  0.450 ms  0.399 ms

 2  192.168.4.31 (192.168.4.31)  1.086 ms  0.880 ms  0.734 ms

 3  esp32-s3-base.local (172.27.30.1) <rst,ack>  53.865 ms  58.118 ms  61.274 ms

[/CODE]

Out of desperation, I tried to flash the firmware from the Raspberry Pi that is running the AP, so it's DIRECTLY in the correct subnet for the esp32 device that it's trying to flash. Upload the firmware on the Raspberry Pi server first, you do NOT want to wait for all the time it takes for it to compile (Raspberry Pi 2 here ...):

[CODE=bash]#!/bin/bash

# Activate Python venv

source ~/ESPHome/venv/bin/activate

# Flash Device

esphome upload esp32-s3-base.yaml --file firmware.bin --device esp32-s3-base.local[/CODE]

Root cause was to enable (on the Rock 5B in the Garage or the Raspberry Pi that I have in the House for some warmer-environment testing) in /etc/sysctl.conf (for Debian/Raspberry Pi OS/Ubuntu and similar GNU/Linux Distributions):

[CODE]net.ipv4.ip_forward=1[/CODE]

​That's probably causing some serious flooding of the different subnets which are now "merged" into one, without NAT/Masquerading or other stuff to take care of these conflicts.

In past versions this was done by iptables, but nowadays it got replaced by nftables, which is very different to configure.

That's what I need to figure out right now ....

luckylinux, 2024-04-07:
The difference (while still probably having the issue) can be seen in `iperf3`.

Directly to AP using the main LAN address:
```
root@MYHOSTNAME:~# iperf3 -c 192.168.4.31 -P 1
Connecting to host 192.168.4.31, port 5201
[  5] local 192.168.13.84 port 33698 connected to 192.168.4.31 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  12.9 MBytes   108 Mbits/sec    0    308 KBytes       
[  5]   1.00-2.00   sec  10.5 MBytes  88.1 Mbits/sec    0    327 KBytes       
[  5]   2.00-3.00   sec  11.2 MBytes  94.4 Mbits/sec    0    362 KBytes       
[  5]   3.00-4.00   sec  10.7 MBytes  89.7 Mbits/sec    0    419 KBytes       
[  5]   4.00-5.00   sec  11.3 MBytes  94.9 Mbits/sec    0    419 KBytes       
[  5]   5.00-6.00   sec  8.70 MBytes  73.0 Mbits/sec    0    419 KBytes       
[  5]   6.00-7.00   sec  9.57 MBytes  80.3 Mbits/sec    0    440 KBytes       
[  5]   7.00-8.00   sec  8.70 MBytes  73.0 Mbits/sec    0    440 KBytes       
[  5]   8.00-9.00   sec  8.70 MBytes  73.0 Mbits/sec    0    440 KBytes       
[  5]   9.00-10.00  sec  8.70 MBytes  73.0 Mbits/sec    0    440 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   101 MBytes  84.7 Mbits/sec    0             sender
[  5]   0.00-10.02  sec  98.8 MBytes  82.7 Mbits/sec                  receiver

iperf Done.
```

To the AP usig the AP IP address (on the ESP32 "side" so to speak):
```
root@MYHOSTNAME:~# iperf3 -c 172.27.1.1 -P 1
Connecting to host 172.27.1.1, port 5201
[  5] local 192.168.13.84 port 48100 connected to 172.27.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   201 KBytes  1.64 Mbits/sec    2   1.41 KBytes       
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   201 KBytes   164 Kbits/sec    5             sender
[  5]   0.00-10.00  sec  65.0 KBytes  53.3 Kbits/sec                  receiver

iperf Done.
```

closed 2024-08-20 by github-actions[bot]