esphome/issues #5681: sn74hc595 over spi with wemos d1 mini lite: Pin xx is used in multiple places / Either "data_pin" and "clock_pin" must be set or "spi_id" must be set.

alex-mextner, 2024-04-06:
### The problem

I tried many config variations. since obviously incorrect one from documentation through many other variants from the network.

even reading the source code and help of chatgpt 4 were not helpful.

source of error 1: https://github.com/esphome/esphome/blob/release/esphome/pins.py#L119
source of error 2: https://github.com/esphome/esphome/blob/master/esphome/components/sn74hc595/__init__.py#L63 <br/>
  btw, look message is too generic and hides possible details of what exactly wrong.
  
Incorrect documentation ("Over SPI"): https://www.esphome.io/components/sn74hc595#over-spi

What I tried:
1
```
spi:
sn74hc595:
  # ...
```

2
```
spi:
  id: spi_bus
  clk_pin: D5  # SPI Clock pin, GPIO 14 on ESP8266 (Wemos D1 Mini Lite)
  mosi_pin: D7  # SPI MOSI pin, GPIO 13 on ESP8266 (Wemos D1 Mini Lite)

sn74hc595:
  - id: sn74hc595_shift_register
    spi_id: spi_bus
    latch_pin: D6  # Use GPIO 12 (D6) for the latch_pin to avoid conflict
    oe_pin: D0  # Use GPIO 16 (D0) for oe_pin to avoid conflict, if needed
    sr_count: 1  # Number of daisy-chained SN74HC595s
```

3 adding `allow_other_uses: true` to sn74hc595

4. option 2 without spi_id

5. using GPIOn pin numbers as alternative

6. using different pins, avoiding setting oe_pin

7 many other trials with config...

----

# Why spi
Actually it works over gpio, however it's really unreliable, so i'm trying spi hoping best.
I checked out the signal diagram using the logical analyser and the signal looks wrong, inconsistent, non-symmetric

I'll attach the picture.

Also, i tried different options with wiring and configuring OE pin. disable it and wire to GND, enable and pull-up, and more. I restarted the device every time.

# Ideas
Looks, It's issue of wemos configuration incompatibility near to pin mapping under the hood.
[image]

### Which version of ESPHome has the issue?

2024.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini Lite

### Component causing the issue

sn74hc595

### Example YAML snippet

```yaml
esphome:
  name: esp-alert-matrix
  friendly_name: esp-alert-matrix

esp8266:
  board: d1_mini_lite

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "ohKuoEK0RjAAv5jmCDSD11o4X71C6CMgk8uxVa96AJY="

ota:
  password: "621930d359b3de71e76bc1d91dbdfa40"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Alert-Matrix"
    password: "wpS4WoxZaQi7"

captive_portal:

# spi:
#   id: spi_bus1
#   clk_pin: D5
#   mosi_pin: D7
#   # Optional: miso_pin: D6

sn74hc595:
  - id: 'sn74hc595_hub'
    latch_pin: D2
    oe_pin: D6
    sr_count: 1
    # spi_id: spi_bus1
    data_pin: D1
    clock_pin: D3

# Individual outputs
switch:
  - platform: gpio
    name: "LED0"
    pin:
      sn74hc595: sn74hc595_hub
      number: 0 # pin
      inverted: false
```

alex-mextner, 2024-04-06:
also, I searched the internet intensively, one of the links:
https://community.home-assistant.io/t/esphome-compilation-error-74hc595/649622

d1 mini lite pinout for reference:
https://www.wemos.cc/en/latest/d1/d1_mini_lite.html

alex-mextner, 2024-04-06:
Slightly more detailed error I got when have tried to compile:
```
INFO ESPHome 2024.3.1
INFO Reading configuration /config/esphome/esp-alert-matrix.yaml...
Failed config

sn74hc595: [source /config/esphome/esp-alert-matrix.yaml:36]
  
  Pin 4 is used in multiple places.
  - [source /config/esphome/esp-alert-matrix.yaml:36]
    id: sn74hc595_hub
    latch_pin: 
      number: 4
      mode: 
        output: True
        input: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    oe_pin: 
      number: 12
      mode: 
        output: True
        input: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    sr_count: 1
    spi_id: spi_bus1
Pin 4 is used in multiple places

Pin 12 is used in multiple places

Pin 12 is used in multiple places
```

alex-mextner, 2024-04-06:
[image]

alex-mextner, 2024-04-06:
Speaking about gpio wiring.
I said it's unreliable and diagram looks messy. 
[image]

ssieb, 2024-07-30:
Your description is incomprehensible.  The view of the bit-banging output is expected and looks ok.  The timings are completely fine.  Try explaining again without going all over the place.  One config, one issue.