esphome/issues #5687: Enable ESP32c6 to display chip temperature correctly, it is currently unable to display

jjcszxh, 2024-04-08:
### The problem

Enable ESP32c6 to display chip temp
[image]
erature correctly, it is currently unable to display

IDF document：https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/temp_sensor.html

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32C6

### Component causing the issue

internal_temperature

### Example YAML snippet

```yaml
- platform: internal_temperature
    name: ${node_internal_temperature_name}
    update_interval: 60s
```

sbrk-r, 2024-04-08:
I found a similar issue with S2, theory at the time was it was an upstream problem, I haven't tried since then. Can you get some log output?

https://github.com/esphome/issues/issues/4500

jjcszxh, 2024-04-08:
This is my yaml configuration
esphome:
  name: ${node_name}

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 8MB
  framework:
    platform_version: 6.5.0
    type: esp-idf
    version: 5.1.2
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      #CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      #CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      #CONFIG_OPENTHREAD_ENABLED: n
      #CONFIG_ENABLE_WIFI_STATION: y
      #CONFIG_USE_MINIMAL_MDNS: y

logger:

  level: ${node_logger_level}

api:
  reboot_timeout: 0s
  encryption:
    key: ${node_api_key}

ota:
  safe_mode: true
  password: ${node_ota_password}

wifi:
  ssid: ${node_wifi_ssid}
  password: ${node_wifi_password}   
  manual_ip:    
    static_ip: ${node_manual_ip_static_ip}
    gateway: ${node_manual_ip_gateway}
    subnet: ${node_manual_ip_subnet}
    dns1: 223.5.5.5
    dns2: 114.114.114.114
  ap:
    ssid: ${node_ap_ssid}
    password: ${node_ap_password}

captive_portal:

I use very verbose, logger

debug [05:42:19][D][internal_temperature:052]: Ignoring invalid temperature (success=0, value=nan)

May I ask how to debug in detail

Is there an issue with the upstream version of IDF,

DAVe3283, 2024-04-09:
I played with this on my C6 and found the ESP-IDF function call is returning the no data sentinel, so ESPHome correctly discards it.

The bug will have to be handled on the ESP-IDF side from what I can tell.

closed 2024-08-22 by github-actions[bot]