esphome/issues #5689: W5500 w/ESP32-S3-N16R8 [W][ethernet:237]: Connecting via ethernet failed! Re-connecting... [V][ethernet:424]: DHCP Client Status: 0

NonaSuomy, 2024-04-09:
### The problem

Trying to connect ESP32-S3-N16R8 to a W5500 Lite module (USR-ES1) also tried a full-sized W5500 also had the same issue just DHCP looking for an address over and over again.

Tried to set manual IP and got error 2 instead of 0

Tried different pins listed in the pull request still no go

https://github.com/esphome/esphome/pull/4424

Tried in Arduino on the board seems to have a different issue: 
```
E (1121) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (1123) w5500.mac: emac_w5500_init(795): vefiry chip ID failed
E (1124) esp_eth: esp_eth_driver_install(228): init mac failed
```
Which some said it was because some module take too long and require a 10ms delay after reset. It seems a few fixes went in for this but not sure if that meshes with esphome.
https://github.com/espressif/esp-idf/pull/11337/commits/177f70c056e87b48117646e79b7eca8d920cfd4d
https://github.com/espressif/esp-idf/issues/11331

Tried to do a TCPDUMP but it never requested for DHCP on the network.
`sudo tcpdump -v -i any port 67 or port 68 -e -n -vv`
Tested plugging in a laptop to the that port and it tossed up the dhcp of that unit in the dump
```
21:26:00.952592 enp8s0 B   ifindex 4 de:ad:be:ef:01 ethertype IPv4 (0x0800), length 376: (tos 0x0, ttl 128, id 55867, offset 0, flags [none], proto UDP (17), length 356)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from de:ad:be:ef:01, length 328, xid 0x187f1e34, Flags [none] (0x0000)
          Client-Ethernet-Address de:ad:be:ef:01
          Vendor-rfc1048 Extensions
            Magic Cookie 0x#######
            DHCP-Message (53), length 1: Request
            Client-ID (61), length 7: ether de:ad:be:ef:01
            Requested-IP (50), length 4: 10.0.42.145
            Server-ID (54), length 4: 10.0.42.40
            Hostname (12), length 15: "DESKTOP-#######"
            FQDN (81), length 18: "DESKTOP-#######"
            Vendor-Class (60), length 8: "MSFT 5.0"
            Parameter-Request (55), length 14:
              Subnet-Mask (1), Default-Gateway (3), Domain-Name-Server (6), Domain-Name (15)
              Router-Discovery (31), Static-Route (33), Vendor-Option (43), Netbios-Name-Server (44)
              Netbios-Node (46), Netbios-Scope (47), Unknown (119), Classless-Static-Route (121)
              Classless-Static-Route-Microsoft (249), Unknown (252)
            END (255), length 0
```
Guessing because the w5500 module never fully comes up to make the DHCP request.

Tried changing the clock from 30 default to 25,20,16,10MHz.

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Issue with ESPHome not HA

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3-N16R8 https://www.aliexpress.com/item/1005006266375800.html

### Component causing the issue

Ethernet/W5500
[... 195 lines cut ...]

### Additional information

Tested both modules on an Arduino Uno and both worked file like so:

```
Starting WebServer on mac index = 17
Connected! IP address: 10.0.42.192
W6100 => Speed: 100 MB, Duplex: FULL DUPLEX, Link status: LINK
===
Using mac index = 17
Connected! IP address: 10.0.42.192
W6100 => Speed: 100 MB, Duplex: FULL DUPLEX, Link status: LINK
===
Using mac index = 17

Host: 10.0.42.192
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9
```
![image](https://github.com/esphome/issues/assets/1906575/6abab58f-98ae-4887-a89e-0f87469a7ed1)

jjcszxh, 2024-04-09:
Could it be a problem with your use of DuPont cables

NonaSuomy, 2024-04-09:
Same wires above worked fine on the Arduino test above. I even attached longer ones to the ones I was using for the S3 to swap from female to male and it worked just fine. Went from 10CM female to female ones, attached to those another 20CM male to male, Is the S3 more sensitive?

NonaSuomy, 2024-04-10:
Just to show it's not the wires I custom handmade my own 70mm dupont wires, still same issue.

          

```
[V][esp32.preferences:059]: nvs_get_blob('233825507'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[C][ota:483]: There have been 0 suspected unsuccessful boot attempts.
[VV][esp32.preferences:040]: s_pending_save: key: 233825507, len: 4
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[VV][esp32.preferences:123]: Checking if NVS data 233825507 has changed
[V][esp32.preferences:163]: nvs_get_blob('233825507'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][esp32.preferences:126]: sync: key: 233825507, len: 4
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][scheduler:063]: set_interval(name='', interval=60000, offset=15372)
[VV][scheduler:063]: set_interval(name='update', interval=60000, offset=29769)
[D][binary_sensor:034]: 'Status': Sending initial state OFF
[C][ethernet:034]: Setting up Ethernet...
[VV][esp-idf:000]: I (483) w5500.mac: version=0

[VV][esp-idf:000]: I (495) system_api: Base MAC address is not set

[VV][esp-idf:000]: I (496) system_api: read default base MAC address from EFUSE

[VV][esp-idf:000]: I (498) esp_eth.netif.netif_glue: 3c:84:27:c7:14:9b

[VV][esp-idf:000]: I (499) esp_eth.netif.netif_glue: ethernet attached to netif

[V][ethernet:367]: [Ethernet event] ETH started (num=0)
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
[VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=4294938004 (now=30708)
[V][sensor:043]: '_Uptime': Received new state 30.709000
[D][sensor:094]: '_Uptime': Sending state 30.70900 s with 0 decimals of accuracy
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
[VV][scheduler:226]: Running interval '' with interval=60000 last_execution=4294952400 (now=45104)
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
```
Plug in the network cable:
```
[VV][esp-idf:000]: E (84971) w5500.mac: w5500_send_command(123): send command timeout
[VV][esp-idf:000]: E (84972) w5500.mac: emac_w5500_start(279): issue OPEN command failed
[VV][esp-idf:000]: E (84973) w5500.mac: emac_w5500_set_link(422): w5500 start failed
[VV][esp-idf:000]: E (84974) esp_eth: eth_on_state_changed(121): ethernet mac set link failed
[VV][esp-idf:000]: E (84975) w5500.phy: w5500_update_link_duplex_speed(84): change link failed
[VV][esp-idf:000]: E (84976) w5500.phy: w5500_get_link(108): update link duplex speed failed
```
```
[VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=30708 (now=90708)
[V][sensor:043]: '_Uptime': Received new state 90.709000
[D][sensor:094]: '_Uptime': Sending state 90.70900 s with 0 decimals of accuracy
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
[VV][scheduler:226]: Running interval '' with interval=60000 last_execution=45104 (now=105104)
[W][ethernet:237]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:424]: DHCP Client Status: 0
```

Doing a reset I see this not sure if it's something?
```
[I][ethernet:220]: Starting ethernet connection
[W][ethernet:152]: Warning set: unspecified
```

NonaSuomy, 2024-04-11:
I think it has something to do with trying to use GPIO09 as the CS Pin or just generally using some of the FSPI pins in the wrong configuration. Need to further test.

I switched to using the UART0 and JTAG pins which were not in use and the interface came up.
```yaml
  clk_pin: GPIO42       # MTMS
  mosi_pin: GPIO44      # U0RXD
  miso_pin: GPIO40      # MTDO
  cs_pin: GPIO39        # MTCK
  interrupt_pin: GPIO41 # MTDI
  reset_pin: GPIO43     # U0TXD
```
[image]

nikito7, 2024-08-20:
With S3 N4R2 I had same issue.
In tasmota it works fine (4 pins).

It's interrupt_pin mandatory?

halogen28372, 2024-11-08:
I'm having a similar problem.