esphome/issues #5690: Compile ESP32c6, latest version of idf framework error reported

jjcszxh, 2024-04-09:
### The problem

I want to use the latest version of the idf framework to compile esp32c6
But I encountered an error message
This error seems to be an issue with the data type

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32-C6-DevKitC-1-N8

### Example YAML snippet

```yaml
esphome:
  name: ${node_name}

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 8MB
  framework:
    platform_version: 6.6.0
    type: esp-idf
    version: 5.2.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y

logger:
  level: ${node_logger_level}

api:
  reboot_timeout: 0s
  encryption:
    key: ${node_api_key}

ota:
  safe_mode: true
  password: ${node_ota_password}

wifi:
  ssid: ${node_wifi_ssid}
  password: ${node_wifi_password}   
  manual_ip:    
    static_ip: ${node_manual_ip_static_ip}
    gateway: ${node_manual_ip_gateway}
    subnet: ${node_manual_ip_subnet}
    dns1: 223.5.5.5
    dns2: 114.114.114.114
  ap:
    ssid: ${node_ap_ssid}
    password: ${node_ap_password}

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esp32-c6/src/esphome/components/logger/logger_esp32.o
Compiling .pioenvs/esp32-c6/src/main.o
Archiving .pioenvs/esp32-c6/esp-idf/app_trace/libapp_trace.a
Archiving .pioenvs/esp32-c6/esp-idf/app_update/libapp_update.a
Archiving .pioenvs/esp32-c6/esp-idf/bootloader_support/libbootloader_support.a
Archiving .pioenvs/esp32-c6/esp-idf/cmock/libcmock.a
Compiling .pioenvs/esp32-c6/console/argtable3/arg_end.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_file.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_hashtable.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_int.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_lit.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_rem.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_rex.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_str.o
Compiling .pioenvs/esp32-c6/console/argtable3/arg_utils.o
Compiling .pioenvs/esp32-c6/console/argtable3/argtable3.o
Compiling .pioenvs/esp32-c6/cxx/cxx_exception_stubs.o
src/esphome/components/logger/logger_esp32.cpp: In member function 'void esphome::logger::Logger::pre_setup()':
src/esphome/components/logger/logger_esp32.cpp:134:27: error: invalid conversion from 'int' to 'uart_port_t' [-fpermissive]
  134 |         this->uart_num_ = -1;
      |                           ^~
      |                           |
      |                           int
src/esphome/components/logger/logger_esp32.cpp:139:27: error: invalid conversion from 'int' to 'uart_port_t' [-fpermissive]
  139 |         this->uart_num_ = -1;
      |                           ^~
      |                           |
      |                           int
Compiling .pioenvs/esp32-c6/cxx/cxx_guards.o
Compiling .pioenvs/esp32-c6/driver/gpio/gpio.o
Compiling .pioenvs/esp32-c6/driver/gpio/gpio_glitch_filter_ops.o
Compiling .pioenvs/esp32-c6/driver/gpio/rtc_io.o
Compiling .pioenvs/esp32-c6/driver/spi/spi_bus_lock.o
Compiling .pioenvs/esp32-c6/driver/deprecated/adc_legacy.o
Compiling .pioenvs/esp32-c6/driver/deprecated/adc_dma_legacy.o
Compiling .pioenvs/esp32-c6/driver/parlio/parlio_common.o
Compiling .pioenvs/esp32-c6/driver/parlio/parlio_tx.o
Compiling .pioenvs/esp32-c6/driver/gpio/dedic_gpio.o
Compiling .pioenvs/esp32-c6/driver/gpio/gpio_pin_glitch_filter.o
Compiling .pioenvs/esp32-c6/driver/gpio/gpio_flex_glitch_filter.o
Compiling .pioenvs/esp32-c6/driver/gpio/gpio_etm.o
Compiling .pioenvs/esp32-c6/driver/gptimer/gptimer.o
Compiling .pioenvs/esp32-c6/driver/gptimer/gptimer_priv.o
Compiling .pioenvs/esp32-c6/driver/deprecated/timer_legacy.o
Compiling .pioenvs/esp32-c6/driver/gptimer/gptimer_etm.o
Compiling .pioenvs/esp32-c6/driver/i2c/i2c.o
Compiling .pioenvs/esp32-c6/driver/i2c/i2c_master.o
Compiling .pioenvs/esp32-c6/driver/i2c/i2c_common.o
Compiling .pioenvs/esp32-c6/driver/i2c/i2c_slave.o
Compiling .pioenvs/esp32-c6/driver/i2s/i2s_common.o
Compiling .pioenvs/esp32-c6/driver/i2s/i2s_platform.o
Compiling .pioenvs/esp32-c6/driver/i2s/i2s_std.o
Compiling .pioenvs/esp32-c6/driver/deprecated/i2s_legacy.o
Compiling .pioenvs/esp32-c6/driver/i2s/i2s_pdm.o
Compiling .pioenvs/esp32-c6/driver/i2s/i2s_tdm.o
Compiling .pioenvs/esp32-c6/driver/ledc/ledc.o
Compiling .pioenvs/esp32-c6/driver/mcpwm/mcpwm_cap.o
Compiling .pioenvs/esp32-c6/driver/mcpwm/mcpwm_cmpr.o
Compiling .pioenvs/esp32-c6/driver/mcpwm/mcpwm_com.o
*** [.pioenvs/esp32-c6/src/esphome/components/logger/logger_esp32.o] Error 1
========================= [FAILED] Took 18.27 seconds =========================
```

### Additional information

[image]

jjcszxh, 2024-04-09:
根据代码分析，我感觉是esp32c6，uart的端口问题

jjcszxh, 2024-04-09:
https://github.com/esphome/esphome/pull/6487

SeByDocKy, 2024-04-19:
Got also the same error with an ESP32 S3 coupled with Weika component

plastik30, 2025-03-21:
I'm also having this issue on an ESP32-S3... any solutions ?

ximex, 2025-05-31:
@room pls retest if this is still an issue. with `2025.5.1` many of the issues with `esp32c6` should be solved with default config (without setting `version`, `platform_version`)