esphome/issues #5691: TSL2591 integration time parameter is not templatable

bakerkj, 2024-04-09:
### The problem

I am unable to use a template to configure the `integration_time` parameter of the the TSL2591. I'd like to be able to do this so I can adjust the integration time depending on the sensing needs (low integration time for bright light; high integration time for dim light).

Thanks!

### Which version of ESPHome has the issue?

2024.3.2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.2

### What platform are you using?

ESP32

### Board

not applicable

### Component causing the issue

tsl2591

### Example YAML snippet

```yaml
esphome:
  name: testing
  platform: ESP32
  board: wemos_d1_mini32

i2c:
  sda: 21
  scl: 22

sensor:
  - platform: tsl2591
    address: 0x29
    update_interval: 1s
    integration_time: !lambda return 100;
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.3.2
INFO Reading configuration /config/esphome/illuminance-sensor.yaml...
Failed config

sensor.tsl2591: [source /config/esphome/illuminance-sensor.yaml:11]
  platform: tsl2591
  address: 41
  update_interval: 1s
  
  This option is not templatable!.
  integration_time: !lambda |-
    return 100;
```

bakerkj, 2024-04-15:
@wjcarpenter, is this something you could help with?

wjcarpenter, 2024-04-16:
It might be possible to make that parameter templatable (though I've never done that, so it's new territory for me). There is a slight logical paradox in there, though, because you want to set the integration time based on the light readings, but the light readings depend on the integration time (and other parameters). Probably the best your could do is set the integration time for the _next_ reading iteration, which might be OK for your use case.

OTOH, there is a C++ API for directly setting the integration time (and APIs for the other parameters). You could use those right now. For example, in the on_value automation for the TSL2591, use a lambda to decide if you want to modify the integration time and then do so if desired.

ssieb, 2024-04-16:
Can't you just use the automatic gain control option?

bakerkj, 2024-04-17:
> Can't you just use the automatic gain control option?

yep - I am using automatic gain control.

I had arbitrarily chosen 200ms integraiton times, but even with automatic gain control, turning the gain down to 1, the sensor was still maxing out in bright conditions. But I'd like to be able to adjust the integration time back up in low light conditions to gain accuracy without reprogramming.

bakerkj, 2024-04-17:
Thanks @wjcarpenter!

> Probably the best your could do is set the integration time for the _next_ reading iteration, which might be OK for your use case.

Yeah this would work for my needs.

> OTOH, there is a C++ API for directly setting the integration time (and APIs for the other parameters). You could use those right now. For example, in the on_value automation for the TSL2591, use a [...]

I'll definitely have a look at the C++ API.

wjcarpenter, 2024-04-20:
It seems a bit specialized to me. But if you are interested in it and have the skills, you could work up a PR for it.