esphome/issues #5693: Bluetooth proxy stopped working with ld2410

mitrokun, 2024-04-10:
### The problem

Last year I created a standard bluetooth proxy that is used with the ld2410c. The esp-idf platform is used. With the first esphome update of 2024, I updated the proxy firmware and the device stopped connecting.
A cyclic error log is attached to the information.
I had to flash a backup of the December version, which works well. 
I try to update again with every major update, but the result is the same.
Why is this happening and how can it be fixed?

### Which version of ESPHome has the issue?

2024.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4

### What platform are you using?

ESP32

### Board

wroom-32 320KB RAM, 4MB

### Example YAML snippet

```yaml
esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: True
```

### Anything in the logs that might be useful for us?

```txt
[22:48:28][D][esp32_ble_tracker:266]: Starting scan...
[22:48:28][I][bluetooth_proxy:278]: [0] [0D:27:2E:4D:04:9D] Connecting v3 with cache
[22:48:28][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[22:48:28][I][esp32_ble_client:067]: [0] [0D:27:2E:4D:04:9D] 0x00 Attempting BLE connection
[22:48:30][D][esp-idf:000]: W (172105) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:48:31][D][esp-idf:000]: W (173635) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:48:32][D][esp-idf:000]: W (174638) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:48:33][D][esp-idf:000]: W (175162) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[22:48:33][D][esp-idf:000]: W (175167) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x3e

[22:48:33][D][esp-idf:000]: W (175171) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x3e

[22:48:33][D][esp-idf:000]: W (175175) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[22:48:33][D][esp32_ble_client:172]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_DISCONNECT_EVT, reason 62
```

mitrokun, 2024-04-10:
If I change the framework to arduino on new versions, the connection is established and everything works.
But it is confusing that there are 10 reports per second in the logs. Is this normal?
```
[04:48:28][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:29][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:30][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT
[04:48:31][D][esp32_ble_client:110]: [0] [0D:27:2E:4D:04:9D] ESP_GATTC_NOTIFY_EVT

```

closed 2024-08-09 by mitrokun

Czara, 2024-08-18:
Hello

I have a same issue today.

`INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-9719cc.yaml...
INFO Starting log output from 192.168.5.116 using esphome API
INFO Successfully connected to esp32-bluetooth-proxy-baie @ 192.168.5.116 in 0.058s
INFO Successful handshake with esp32-bluetooth-proxy-baie @ 192.168.5.116 in 0.116s
[18:08:39][I][app:100]: ESPHome version 2024.7.3 compiled on Aug 18 2024, 17:52:13
[18:08:39][C][wifi:599]: WiFi:
[18:08:39][C][wifi:427]:   Local MAC: 24:6F:28:97:19:CC
[18:08:39][C][wifi:432]:   SSID: [redacted]
[18:08:39][C][wifi:435]:   IP Address: 192.168.5.116
[18:08:39][C][wifi:439]:   BSSID: [redacted]
[18:08:39][C][wifi:440]:   Hostname: 'esp32-bluetooth-proxy-baie'
[18:08:39][C][wifi:442]:   Signal strength: -58 dB ▂▄▆█
[18:08:39][C][wifi:446]:   Channel: 6
[18:08:39][C][wifi:447]:   Subnet: 255.255.255.0
[18:08:39][C][wifi:448]:   Gateway: 192.168.5.1
[18:08:39][C][wifi:449]:   DNS1: 1.1.1.1
[18:08:39][C][wifi:450]:   DNS2: 1.0.0.1
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][C][logger:185]: Logger:
[18:08:39][C][logger:186]:   Level: DEBUG
[18:08:39][C][logger:188]:   Log Baud Rate: 115200
[18:08:39][C][bluetooth_proxy:088]: Bluetooth Proxy:
[18:08:39][C][bluetooth_proxy:089]:   Active: YES
[18:08:39][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
[18:08:39][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[18:08:39][C][factory_reset.button:011]: Factory Reset Button 'Factory reset'
[18:08:39][C][factory_reset.button:011]:   Icon: 'mdi:restart-alert'
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][C][esp32_ble:383]: ESP32 BLE:
[18:08:39][C][esp32_ble:385]:   MAC address: 24:6F:28:97:19:CE
[18:08:39][C][esp32_ble:386]:   IO Capability: none
[18:08:39][C][esp32_ble_tracker:653]: BLE Tracker:
[18:08:39][C][esp32_ble_tracker:654]:   Scan Duration: 300 s
[18:08:39][C][esp32_ble_tracker:655]:   Scan Interval: 320.0 ms
[18:08:39][C][esp32_ble_tracker:656]:   Scan Window: 30.0 ms
[18:08:39][C][esp32_ble_tracker:657]:   Scan Type: ACTIVE
[18:08:39][C][esp32_ble_tracker:658]:   Continuous Scanning: True
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][C][mdns:116]: mDNS:
[18:08:39][C][mdns:117]:   Hostname: esp32-bluetooth-proxy-baie
[18:08:39][C][esphome.ota:073]: Over-The-Air updates:
[18:08:39][C][esphome.ota:074]:   Address: esp32-bluetooth-proxy-baie.local:3232
[18:08:39][C][esphome.ota:075]:   Version: 2
[18:08:39][C][safe_mode:018]: Safe Mode:
[18:08:39][C][safe_mode:020]:   Boot considered successful after 60 seconds
[18:08:39][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:08:39][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[18:08:39][C][api:139]: API Server:
[18:08:39][C][api:140]:   Address: esp32-bluetooth-proxy-baie.local:6053
[18:08:39][C][api:142]:   Using noise encryption: YES
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:39][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT
[18:08:40][D][esp32_ble_client:110]: [0] [AD:22:B2:34:B8:14] ESP_GATTC_NOTIFY_EVT`