esphome/issues #5696: EKTF2232-Touchscreen dimensions not properly transformed

droans, 2024-04-11:
### The problem

I'm using this touchscreen controller with the Inkplate 6 Plus with dimensions of 1024x758. By default, the display is in landscape instead of portrait, meaning the X max touchpoint is 1024 and the Y max is 758.

Because I want it to be vertical, I set the display to be flipped 270 degrees. This is completed properly and the dimensions for any page are properly translated.

However, between 2024.2 and 2024.3, the touch point dimensions are no longer being properly transformed. X will always keep the default value of 1024 and Y will always keep the default value of 758.

I have tried all combinations of transform options for the touchscreen. The values I receive are:

| swap_xy  | mirror_x | mirror_y | Top Left X/Y | Top Right X/Y | Bottom Left X/Y | Bottom Right X/Y |
|----------|----------|----------|--------------|---------------|-----------------|------------------|
| true     | true     | true     | 0,0          | 0,758         | 1024,0          | 1024,758         |
| true     | true     | false    | 1024,0       | 1024,758      | 0,0             | 0,758            |
| true     | false    | true     | 0,758        | 0,0           | 1024,758        | 1024,0           |
| true     | false    | false    | 1024,758     | 1024,0        | 0,758           | 0,0              |
| false    | true     | true     | 0,0          | 1024,0        | 0,758           | 1024,758         |
| false    | true     | false    | 0,758        |               | 0,0             | 1024,0           |
| false    | false    | true     | 1024,0       | 0,0           | 1024,758        | 0,758            |
| false    | false    | false    | 1024,758     | 0,758         | 1024,0          | 0,0              |
| expected |          |          | 0,0          | 758,0         | 0,1024          | 758,1024        |

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.04

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

EKTF2232

### Example YAML snippet

```yaml
pca6416a:
  - id: pca6416a_hub
    address: 0x20

touchscreen:
  - platform: ektf2232
    interrupt_pin: GPIO36
    rts_pin: 
      pca6416a: pca6416a_hub
      number: 10
    transform:
      swap_xy: true
      mirror_x: true
      mirror_y: true
    on_touch:
      - logger.log:
          format: "touch x=%d, y=%d"
          args: ['touch.x', 'touch.y']

display:
  - platform: inkplate6
    id: main_display
    greyscale: false
    partial_updating: true
    update_interval: never
    model: inkplate_6_plus
    rotation: 270
    ckv_pin: 32
    sph_pin: 33
    gmod_pin:
      pca6416a: pca6416a_hub
      number: 1
    gpio0_enable_pin:
      pca6416a: pca6416a_hub
      number: 8
    oe_pin:
      pca6416a: pca6416a_hub
      number: 0
    spv_pin:
      pca6416a: pca6416a_hub
      number: 2
    powerup_pin:
      pca6416a: pca6416a_hub
      number: 4
    wakeup_pin:
      pca6416a: pca6416a_hub
      number: 3
    vcom_pin:
      pca6416a: pca6416a_hub
      number: 5
    pages: 
      - id: page1
        <<: !include pages/page1.yaml
```

nielsnl68, 2024-04-11:
Interesting, could we talk on discord a bit more:
https://discord.com/channels/429907082951524364/1076516361011998761

droans, 2024-04-12:
I don't have the ability to stay active enough during troubleshooting for Discord. Is there information I could give you here to help?

nielsnl68, 2024-04-12:
That is no problem, I just want all touchscreen issues to be combined at one place. Slow chat is okay.

For now could you press all 4 corners Top-left, top-right, bottom-left and bottom-right. and write down the values you get.
after that could you change: `args: ['touch.x', 'touch.y']` into `args: ['touch.x_raw', 'touch.y_raw']` and press the same sequence.

For the display, it is better use the **transform** settings there as well, because **rotation** is a software based procedure and with **transform** you modify the display it self. BTW. when you swap x and y then you also need to swap the dimensions.

droans, 2024-04-25:
Sorry for the delay, work has been keeping me busy. 

Running ESPHome 2024.4.1. Only change to above config is to remove `swap_xy` and the format for the log was changed to:

```yaml
format: "touch x=%d (raw x=%d), y=%d (raw y=%d)"
args: ['touch.x', 'touch.x_raw', 'touch.y', 'touch.y_raw']
```

```
touch x=0 (raw x=1435), y=9 (raw y=1896)          # Top Left, Expect apx 0,0
touch x=973 (raw x=71), y=14 (raw y=1883)         # Top Right, Expect apx 758,0
touch x=1 (raw x=1434), y=748 (raw y=24)          # Bottom Left, Expect apx 0,1024
touch x=1023 (raw x=0), y=745 (raw y=31)          # Bottom Right, Expect apx 758,1024
```

It's fortunately not too difficult to calculate new touchpoints, but the prior method where touches were synced with the dimensions made it much easier.

droans, 2024-04-25:
Maybe it would be best if the touchscreen config was altered so you could set the dimensions of the touchscreen. Something like:

```yaml
touchscreen:
  - platform: XXX
    ...
    dimensions: 
      - [0,0]            # Possibly implied by default if not provided?
      - [758,1024]
    ...
```

Gadgit83, 2024-05-01:
Been searching and scratching my head for a couple of days on this one! Turns out it's not just me :). OK, so I think it's essentially these lines in touchscreen.cpp, which are designed to scale the 'normalised' i.e. 0 to 1 touch point coordinates to the display dimensions.

https://github.com/esphome/esphome/blob/c69cdec0521147ec3d1a594c9c5121cab9710c33/esphome/components/touchscreen/touchscreen.cpp#L97C1-L98C66 

However, the changes which were made 3 months ago - whichever version that was, reference the 'display' component here to get the height and width, which are incorrect (compared to the drawing x/y dimensions) if you have a display orientation of 90 and presumably 270:
https://github.com/esphome/esphome/blob/c69cdec0521147ec3d1a594c9c5121cab9710c33/esphome/components/touchscreen/touchscreen.cpp#L21

I found all this on an ili9xxx and xpt2046 touchscreen combo by the way, so it's not just EKTF2232. I've tried the transformation approach mentioned above with swap_xy instead of rotation 90, and that seems to make matters worse (the display completely mis-renders).

> Maybe it would be best if the touchscreen config was altered so you could set the dimensions of the touchscreen. Something like:
> 
> ```yaml
> touchscreen:
>   - platform: XXX
>     ...
>     dimensions: 
>       - [0,0]            # Possibly implied by default if not provided?
>       - [758,1024]
>     ...
> ```

@droans this would probably work, but I think that the touchscreen.cpp could just look at this->display rotation and swap the x/y dimensions for 90 and 270 degree rotations, and that would sort it out automatically.

berndkrannich, 2024-05-06:
I can confirm that a touchscreen that is rotated 90 or 270 degrees doesn't seem to work.

Running a xpt2046, I manually changed `x_raw` and `y_raw` at <https://github.com/esphome/esphome/blob/dev/esphome/components/xpt2046/touchscreen/xpt2046.cpp#L52-L53> and this resulted in the (calibrated) touch coordinates matching the screen coordinates, but without that change `x` and `y` values are swapped.

Gadgit83, 2024-07-13:
@nielsnl68  - would you suggest we have a go at fixing this and submitting a pull request? To be honest, I'm not sure where to start with it, but I could perhaps try making some changes locally (although I don't even know how to do that yet)