esphome/issues #5697: No PT1000 with MAX31865 at NodeMCU (ESP8266)

Itsch001, 2024-04-12:
### The problem

I try to use PT1000 with a MAX31865. No matter, what I set up, it only recognizes PT100. When I connect a 120Ω resistance as sensor, it gets 48°C. When I use a 1000Ω, it says “RTD High Threshold between measurements”.
The reference resistor is 4700Ω.

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

MAX31865

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO14
  miso_pin: GPIO12
  mosi_pin: GPIO13

sensor:
  - platform: max31865
    name: "BBQ Temperature"
    icon: "mdi:hamburger"
    cs_pin: GPIO15
    reference_resistance: "4300 Ω"
    rtd_nominal_resistance: "1000 Ω"
```

### Anything in the logs that might be useful for us?

```txt
[21:15:47][W][max31865:034]: RTD High Threshold between measurements
[21:15:47][E][max31865:101]: SPI bus read all 0 or all 1 (0xFFFF), check MAX31865 wiring & power.
[21:15:47][D][sensor:093]: 'BBQ Temperature': Sending state nan °C with 2 decimals of accuracy
```

### Additional information

the log output with the 120Ω connected is
`[21:29:42][D][sensor:093]: 'BBQ Temperature': Sending state 48.52332 °C with 2 decimals of accuracy`
changing the default values back to 470Ω / 100Ω doesn’t seem to have an effect.

DAVe3283, 2024-04-16:
The MAX31865 only measures the ratio between the probe and the reference resistor. The ratio is multiplied by `reference_resistance` to get the true measured sensor resistance. Then the `rtd_nominal_resistance` (resistance at 0°C) value and the measured resistance are fed into a [fairly complex formula](https://github.com/esphome/esphome/blob/01419822f73e8784ed7047950c437b5416efb861/esphome/components/max31865/max31865.cpp#L189) to determine the temperature.

The main error I see in your log is the `SPI bus read all 0 or all 1 (0xFFFF), check MAX31865 wiring & power.`. When you have the 1000Ω sensor connected, the SPI bus is being held high (hence the `0xFFFF`). The earlier warning about RTD High Threashold is because the error bit is set, but that is only set because the bus is stuck high. I would ignore that and focus on why the SPI bus is not working.

1. Is your PT1000 a 2-, 3-, or 4-wire sensor?
2. Did you solder the jumpers on the breakout board to match that setup?
3. Have you measured the resistance of the sensor with a standalone multimeter to verify you get nearly exactly 1000Ω with it in ice water?

sbrk-r, 2024-04-16:
Are you positive the physical reference resistor is 4700 ohms? Note that in the 4 digit notation for SMD resistors, 4700 = 470 ohms. So it should be either 472 or 4701.

The IC merely reports ratio between voltage across Rref and across the Pt sensor, and a few error conditions. If it had a proper bias resistor for Pt1000 I would expect an underrange error with 120R 'sensor'.

Itsch001, 2024-04-16:
I'm so sorry, you are totally right. I had both resistors in front of me and was misled by the number on it.

closed 2024-04-16 by Itsch001