esphome/issues #5698: Sensor readings not consistently reaching Home Assistant

stumpylog, 2024-04-13:
### The problem

It's hard to tell if this problem lies with Home Assistant or ESPHome, but given it is a single device, not all devices, I suspect ESPHome, prehaps due to being an ESP32-C3 device.

With a sensor configured for 60s intervals, I see in the serial logs the device believes it is sending the data, presumably over the API.  However, the sensor updates do not reach Home Assistant.  There are no errors in the serial logs when sending state.

Both the device and Home Assistant seem to believe they are connected, but the sensor value in the history updates only sporadically.

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32

### Board

adafruit_qtpy_esp32c3

### Component causing the issue

api

### Anything in the logs that might be useful for us?

```txt
[V][sensor:043]: 'Temperature': Received new state 20.086641
[D][sensor:093]: 'Temperature': Sending state 20.08664 °C with 1 decimals of accuracy
[V][sensor:043]: 'Pressure': Received new state 999.786316
[D][sensor:093]: 'Pressure': Sending state 999.78632 hPa with 1 decimals of accuracy
[V][sensor:043]: 'Relative Humidity': Received new state 53.348633
[D][sensor:093]: 'Relative Humidity': Sending state 53.34863 % with 1 decimals of accuracy
[V][sensor:043]: 'WiFi Signal': Received new state -79.000000
[D][sensor:093]: 'WiFi Signal': Sending state -79.00000 dBm with 0 decimals of accuracy
[V][bme280.sensor:210]: Sending conversion request...
[V][bme280.sensor:242]: Got temperature=20.1°C pressure=999.7hPa humidity=53.4%
[V][sensor:043]: 'Temperature': Received new state 20.107149
[D][sensor:093]: 'Temperature': Sending state 20.10715 °C with 1 decimals of accuracy

Occasionally:
```log
[V][esp-idf:000]: I (1656658) wifi:
[V][esp-idf:000]: bcn_timeout,ap_probe_send_start
[V][esp-idf:000]: 
```
```

### Additional information

![image](https://github.com/esphome/issues/assets/797416/7f22f52d-9613-42ea-9c31-35af3f12af09)

4 sensor updates over a roughly 20 minute period, despite 60s configured

sbrk-r, 2024-04-14:
HA ignores the updates if the value hasn't changed, maybe you are running into this. You can try force_update in esphome, or more decimals accuracy.

stumpylog, 2024-04-14:
The value is changing, even if only slightly.  Other sensors with the same interval also don't have this behavior. They present a smooth curve of history

ssieb, 2024-04-14:
```
[D][sensor:093]: 'Temperature': Sending state 20.08664 °C with 1 decimals of accuracy
[D][sensor:093]: 'Temperature': Sending state 20.10715 °C with 1 decimals of accuracy
```
These are the same value at 1 decimal of accuracy.

stumpylog, 2024-04-14:
Maybe I'm not understanding the decimal accuracy documentation.

`accuracy_decimals` says "This does not impact the actual value reported to Home Assistant, it just sets the number of decimals to use when displaying it"

Which I read to mean the actual value is sent with full accuracy, so a value like 20.08664, even if HA only uses 1 decimal to display it.  So HA would see the full value, and update "last updated" (not "last changed").  If that setting actual affects the reported value, so the precision doesn't leave the device, then the docs could clarify it.

sbrk-r, 2024-04-14:
I think HA receives 20.08664, rounds it to one decimal, 20.1, then compares to the last value, 20.1, and drops it as unchanged.

I've certainly run into behavior like that before anyway. "last updated" is a bit of a misnomer, yeah...

stumpylog, 2024-04-15:
Looks like after setting `accuracy_decimals` to 2, the updating is happening consistently now.  So it appears to be the rounding, wherever it is happening, compounded by the BME280 sensor defaulting to 1 decimal, vs SHT40 defaulting to 2 decimals.

While there's room for some documentation improvement, the actual issue is resolved to me.