esphome/issues #5699: Wifi Connection Loop after "Beacon Timeout"

stumpylog, 2024-04-13:
### The problem

After my ESP32-c3 device receives a "Beacon Timeout", it will never reconnect to the wifi, including after a reset of the device.  The wifi component will instead loop forever, reporting 'Probe Request Unsuccessful'

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Adafruit Qtpy C3

### Anything in the logs that might be useful for us?

```txt
V][sensor:043]: 'WiFi Signal': Received new state -82.000000
[D][sensor:093]: 'WiFi Signal': Sending state -82.00000 dBm with 0 decimals of accuracy
[D][time:048]: Synchronized time: 2024-04-13 13:04:57
[V][esp-idf:000]: I (1656658) wifi:
[V][esp-idf:000]: bcn_timeout,ap_probe_send_start
[V][esp-idf:000]: 

[V][bme280.sensor:210]: Sending conversion request...
[V][bme280.sensor:242]: Got temperature=20.1°C pressure=999.7hPa humidity=53.4%
[V][sensor:043]: 'Temperature': Received new state 20.117304
[D][sensor:093]: 'Temperature': Sending state 20.11730 °C with 1 decimals of accuracy
[V][sensor:043]: 'Pressure': Received new state 999.727173
[D][sensor:093]: 'Pressure': Sending state 999.72717 hPa with 1 decimals of accuracy
[V][sensor:043]: 'Relative Humidity': Received new state 53.362305
[D][sensor:093]: 'Relative Humidity': Sending state 53.36230 % with 1 decimals of accuracy
[V][esp-idf:000]: I (1659161) wifi:
[V][esp-idf:000]: ap_probe_send over, resett wifi status to disassoc
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1659162) wifi:
[V][esp-idf:000]: state: run -> init (c800)
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1659163) wifi:
[V][esp-idf:000]: pm stop, total sleep time: 0 us / 1604183979 us

[V][esp-idf:000]: 

[V][esp-idf:000]: I (1659164) wifi:
[V][esp-idf:000]: <ba-del>idx:1, tid:0
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1659164) wifi:
[V][esp-idf:000]: <ba-del>idx:0, tid:6
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1659165) wifi:
[V][esp-idf:000]: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[V][esp-idf:000]: 

[W][wifi_esp32:659]: Event: Disconnected ssid='myssidhere' bssid=E2:63:DA:22:45:29[redacted] reason='Beacon Timeout'
[W][wifi:153]: WiFi Connection lost... Reconnecting...
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:309]:   BSSID: E2:63:DA:22:45:29
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:334]:   Channel: 11
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[W][api.connection:078]: Home Assistant 2024.4.3 (192.168.1.13): Network unavailable, disconnecting
[W][wifi:152]: Warning set: unspecified
[V][api:115]: Removing connection to Home Assistant 2024.4.3
[W][api:152]: Warning set: unspecified
[V][text_sensor:013]: 'ESP IP Address': Received new state 0.0.0.0
[D][text_sensor:064]: 'ESP IP Address': Sending state '0.0.0.0'
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:311]:   BSSID: Not Set
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:336]:   Channel: Not Set
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[V][esp-idf:000]: W (1668101) wifi:
[V][esp-idf:000]: Haven't to connect to a suitable AP now!
[V][esp-idf:000]: 

[W][wifi_esp32:926]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_NOT_CONNECT
[V][sensor:043]: 'WiFi Signal': Received new state 0.000000
[D][sensor:093]: 'WiFi Signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:309]:   BSSID: E2:63:DA:22:45:29
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:334]:   Channel: 11
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:311]:   BSSID: Not Set
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:336]:   Channel: Not Set
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:309]:   BSSID: E2:63:DA:22:45:29
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:334]:   Channel: 11
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:311]:   BSSID: Not Set
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:336]:   Channel: Not Set
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[W][wifi:667]: Restarting WiFi adapter...
[V][wifi_esp32:219]: Disabling STA.
[V][esp-idf:000]: I (1688757) wifi:
[V][esp-idf:000]: flush txq
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1688758) wifi:
[V][esp-idf:000]: stop sw txq
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1688759) wifi:
[V][esp-idf:000]: lmac stop hw txq
[V][esp-idf:000]: 

[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:309]:   BSSID: E2:63:DA:22:45:29
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:334]:   Channel: 11
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
[V][wifi_esp32:217]: Enabling STA.
[V][esp-idf:000]: I (1688863) wifi:
[V][esp-idf:000]: mode : sta (34:b4:72:ea:e1:1c)
[V][esp-idf:000]: 

[V][esp-idf:000]: I (1688864) wifi:
[V][esp-idf:000]: enable tsf
[V][esp-idf:000]: 

[W][component:232]: Component wifi took a long time for an operation (123 ms).
[W][component:233]: Components should block for at most 30 ms.
[V][wifi_esp32:628]: Event: WiFi STA stop
[V][wifi_esp32:616]: Event: WiFi STA start
[V][esp-idf:000]: I (1688885) wifi:
[V][esp-idf:000]: Set ps type: 0, coexist: 0

[V][esp-idf:000]: 

[W][wifi_esp32:653]: Event: Disconnected ssid='myssidhere' reason='Probe Request Unsuccessful'
[W][wifi:630]: Error while connecting to network.
[D][wifi:673]: Retrying with hidden networks...
[I][wifi:303]: WiFi Connecting to 'myssidhere'...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: 'myssidhere'
[V][wifi:311]:   BSSID: Not Set
[V][wifi:329]:   Password: 'mypasswordhere'[redacted]
[V][wifi:336]:   Channel: Not Set
[V][wifi:343]:   Using DHCP IP
[V][wifi:345]:   Hidden: YES
```

huuub, 2024-05-21:
I bought 2 new ESP32-S3-BOX-3 devices. Both I am not able to install the homeassistant voice on them. I just cant enter a wi-fi network.

starschild86, 2024-05-27:
> I bought 2 new ESP32-S3-BOX-3 devices. Both I am not able to install the homeassistant voice on them. I just cant enter a wi-fi network.

same here my ESP32-S3-BOX-3 arrived after waiting 2 months but i'm not able to set it up properly :|

ssieb, 2024-05-27:
@huuub and @starschild86 that's an unrelated issue and has also been fixed.  Come ask on discord if you need help.