esphome/issues #5700: Connection error occurred: [Errno 104] Connection reset by peer

lasmaty07, 2024-04-14:
### The problem

Im trying to read the ambient light with a `bh1750` and using the shown `yaml`, every couple of readings, It becomes offline and shows the log presented.

tried changing the frequency, because I read [here](https://github.com/esphome/issues/issues/3455)

> There was a problem with establishing wifi connection when using low i2c frequencies. (The default is 50kHz)

any help is appreciated.

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

i2c/bh1750

### Example YAML snippet

```yaml
esphome:
  name: sensor-luz
  friendly_name: sensor_luz

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  encryption:
    key: "fp/Rfy71v3w2AHru1ea+j+amAe7XmfnA4fwf4TeHtuA="

ota:
  password: "some_password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 10.0.0.95
    gateway: 10.0.0.1
    subnet: 255.255.255.0  

i2c:
  sda: 21
  scl: 22
  # scan: true
  id: bus_a
  frequency: 800kHz

sensor:
  - platform: bh1750
    name: "Living Illuminance"
    address: 0x23
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
[21:24:43][I][app:102]: ESPHome version 2024.2.2 compiled on Apr 13 2024, 21:18:24
[21:24:43][C][wifi:577]: WiFi:
[21:24:43][C][wifi:409]:   Local MAC: D4:8A:FC:A9:D3:94
[21:24:43][C][wifi:414]:   SSID: [redacted]
[21:24:43][C][wifi:415]:   IP Address: 10.0.0.95
[21:24:43][C][wifi:417]:   BSSID: [redacted]
[21:24:43][C][wifi:418]:   Hostname: 'sensor-luz'
[21:24:43][C][wifi:420]:   Signal strength: -45 dB ▂▄▆█
[21:24:43][C][wifi:424]:   Channel: 4
[21:24:43][C][wifi:425]:   Subnet: 255.255.255.0
[21:24:43][C][wifi:426]:   Gateway: 10.0.0.1
[21:24:43][C][wifi:427]:   DNS1: 0.0.0.0
[21:24:43][C][wifi:428]:   DNS2: 0.0.0.0
[21:24:43][C][logger:447]: Logger:
[21:24:43][C][logger:448]:   Level: DEBUG
[21:24:43][C][logger:449]:   Log Baud Rate: 115200
[21:24:43][C][logger:451]:   Hardware UART: UART0
[21:24:43][C][i2c.arduino:053]: I2C Bus:
[21:24:43][C][i2c.arduino:054]:   SDA Pin: GPIO21
[21:24:43][C][i2c.arduino:055]:   SCL Pin: GPIO22
[21:24:43][C][i2c.arduino:056]:   Frequency: 800000 Hz
[21:24:43][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[21:24:43][I][i2c.arduino:069]: Results from i2c bus scan:
[21:24:43][I][i2c.arduino:075]: Found i2c device at address 0x23
[21:24:43][C][bh1750.sensor:118]: BH1750 'Living Illuminance'
[21:24:43][C][bh1750.sensor:118]:   Device Class: 'illuminance'
[21:24:43][C][bh1750.sensor:118]:   State Class: 'measurement'
[21:24:43][C][bh1750.sensor:118]:   Unit of Measurement: 'lx'
[21:24:43][C][bh1750.sensor:118]:   Accuracy Decimals: 1
[21:24:43][C][bh1750.sensor:119]:   Address: 0x23
[21:24:43][C][bh1750.sensor:124]:   Update Interval: 10.0s
[21:24:43][C][mdns:115]: mDNS:
[21:24:43][C][mdns:116]:   Hostname: sensor-luz
[21:24:43][C][ota:096]: Over-The-Air Updates:
[21:24:43][C][ota:097]:   Address: 10.0.0.95:3232
[21:24:43][C][ota:100]:   Using Password.
[21:24:43][C][ota:103]:   OTA version: 2.
[21:24:43][C][api:139]: API Server:
[21:24:43][C][api:140]:   Address: 10.0.0.95:6053
[21:24:43][C][api:142]:   Using noise encryption: YES
[21:21:53][D][bh1750.sensor:159]: 'Living Illuminance': Got illuminance=0.2lx
[21:21:53][D][sensor:094]: 'Living Illuminance': Sending state 0.22638 lx with 1 decimals of accuracy
[21:22:03][D][bh1750.sensor:159]: 'Living Illuminance': Got illuminance=0.2lx
[21:22:03][D][sensor:094]: 'Living Illuminance': Sending state 0.22638 lx with 1 decimals of accuracy
WARNING sensor-luz @ 10.0.0.95: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for sensor-luz @ 10.0.0.95
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for sensor-luz @ 10.0.0.95: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.0.0.95', port=6053))]: [Errno 111] Connect call failed ('10.0.0.95', 6053) (SocketAPIError)
INFO Trying to connect to sensor-luz @ 10.0.0.95 in the background
INFO Successfully connected to sensor-luz @ 10.0.0.95 in 0.005s
INFO Successful handshake with sensor-luz @ 10.0.0.95 in 0.096s
[21:22:53][D][bh1750.sensor:159]: 'Living Illuminance': Got illuminance=0.3lx
[21:22:53][D][sensor:094]: 'Living Illuminance': Sending state 0.33957 lx with 1 decimals of accuracy
[21:23:03][D][bh1750.sensor:159]: 'Living Illuminance': Got illuminance=0.2lx
[21:23:03][D][sensor:094]: 'Living Illuminance': Sending state 0.22638 lx with 1 decimals of accuracy
```

GobblerGuy, 2024-04-23:
I’m getting similar to this. Rock solid device for ages but had not update in a very very long time. Applied updates and boom getting device dropouts and similar messages to logs above.

brezuicabogdan, 2024-05-26:
Same issue here. Never had issues before and have been using ESPHome for a couple of years now. After upgrading to 2024.5.3
I started getting Connection reset by peer errors. 
I have investigated network connectivity/wifi issues but I can see no problems and also nothing changed in my network settings lately.
After the same upgrade I also started getting Out Of Memory errors with ArduinoJson on devices which were using both API and MQTT- could this also be an OOM issue just less visible?

I will have one of the affected devices with DEBUG logging but any iddeas would help.

Thanks.

Qhilm, 2024-05-26:
I have this issue on two ESP32-C3 since a couple of months.

Other error I see:

```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:190
First occurred: May 25, 2024 at 20:37:10 (1 occurrences)
Last logged: May 25, 2024 at 20:37:10

Can't connect to ESPHome API for REDACTED @ 192.168.18.18: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.18.18', port=6053))]: [Errno 111] Connect call failed ('192.168.18.18', 6053) (SocketAPIError)
```

One has web interface enabled, the other not, so that's not related to this at least.

I have one ESP8266 which does not have the issue so far.

Qhilm, 2024-06-03:
Thanks @stigvi but my 3 devices are listed only as "1 device", the root cause seems to be different for me at least.

SVH-Powel, 2024-06-13:
It lasted only a few days and it started to reset the connection again each 5 minutes. But I found out when this happened, it was connected to a wifi mesh node with a weak signal. And it seems to not connect back to the nearest wifi node until the esp32 was reset.

I am now forcing the esp32 to connect to a spesific wifi node and that seems to do the trick. It has been free from "connection reset by peer" for a few days now, but you never know.....

```
wifi:
  domain: .lan
  enable_btm: true
  enable_rrm: true
  networks: 
    - ssid: "HEIME5.ORG"
      password: !secret heime_wifi
      bssid: "70:3A:CB:F4:ED:F6"
```

MikeBP13, 2025-04-02:
I get the same error randomly on 2 BLE proxy devices (out of 4). All are based on the same FW, same HW device.

The Wi-Fi connection is good and the issue is not related to DHCP IP lease... The devices do not restart themselves as the uptime sensor confirms this.

Any ideas where to look / what to test?