esphome/issues #5705: LD2420 triggers boot loop due WDT on ESP32S3

andyvandesand, 2024-04-17:
### The problem

I'm trying to get an LD2420 to work with an ESP32-S3, the waveshare s3-zero version. As the specific board profile isn't available yet I've already tried changing some paramters, without success. I've tried just the UART without the LD2420, that's working fine. Other pins don't seem to change much either. When monitoring the uart I can see the sensor actually providing valid data, but then the WDT resets the chip.

Curiously, if I disable all the MHI-AC-Ctrl components, compilation fails with 
`Compiling .pioenvs/thatreacesp/src/esphome/components/ld2420/ld2420.cpp.o
src/esphome/components/ld2420/ld2420.cpp: In member function 'virtual void esphome::ld2420::LD2420Component::dump_config()':
src/esphome/components/ld2420/ld2420.cpp:88:45: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
                                             ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:88:3: error: 'LOG_SELECT' was not declared in this scope
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
   ^~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:88:3: note: suggested alternative: 'LOG_SECRET'
   LOG_SELECT(TAG, "  Operating Mode", this->operating_selector_);
   ^~~~~~~~~~
   LOG_SECRET
src/esphome/components/ld2420/ld2420.cpp: In member function 'virtual void esphome::ld2420::LD2420Component::setup()':
src/esphome/components/ld2420/ld2420.cpp:140:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_SIMPLE_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:145:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_NORMAL_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp: In member function 'void esphome::ld2420::LD2420Component::set_operating_mode(const string&)':
src/esphome/components/ld2420/ld2420.cpp:296:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(state);
           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:315:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
     this->operating_selector_->publish_state(OP_SIMPLE_MODE_STRING);
           ^~~~~~~~~~~~~~~~~~~
*** [.pioenvs/thatreacesp/src/esphome/components/ld2420/ld2420.cpp.o] Error 1
========================= [FAILED] Took 12.93 seconds =========================`
which doesn't make any sense to me!?

The sensor is working on an ESP32-WROOM-32E. 

Any help would be appreciated...

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.0

### What platform are you using?

ESP32

### Board

Waveshare ESP32-S3-Zero

### Component causing the issue

LD2420

### Example YAML snippet

```yaml
substitutions:
  name: esphome-web-529aa0
  friendly_name: "Theatre AC"
  # Unique device ID in HA
[... 213 lines cut ...]
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][ld2420:114]: Setting up LD2420...
[D][ld2420:566]: Sending set config enable command: FF
E (10682) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10682) task_wdt:  - loopTask (CPU 1)
E (10682) task_wdt: Tasks currently running:
E (10682) task_wdt: CPU 0: IDLE
E (10682) task_wdt: CPU 1: loopTask
E (10682) task_wdt: Aborting.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377590
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
entry 0x403c98c0
[I][logger:158]: Log initialized
[C][ota:483]: There have been 1 suspected unsuccessful boot attempts.
```

StreakingJerry, 2024-04-18:
Totally same issue with esp32c3

fragpic, 2024-04-22:
Is the LD2420 connected to the board when you boot it?
The LD2420 component has an issue where if no data is seen on the RX pin at boot, it will bootloop.
#6594 is supposed to fix it

andyvandesand, 2024-04-23:
I just tried without the board attached at all, and it does the same. 
Going off the fact that the LD actually sends data through the uart until the WDT resets when I monitor it I would assume this is purely a bug in the ld2420 library?

andyvandesand, 2024-05-02:
Sorry took me a while to get a local version setup. I can confirm this is at least a partial fix. There is still an issue, in that on every other cold power up the sensor just gets flagged as "failed". Restarting the board seems to fix this. 
I don't have the required insight to fix this, but I'm very happy to do some local testing if anyone has any info for me.

LBrinston, 2024-05-19:
I believe I too am experiencing this issue with the standard ESP32 dev board and the LD2420. 
I see the watchdog resetting the ESP part way through setting up the LD2420. 
This results in a boot loop during programming ultimately ending in the ESP booting into safe mode and thus being undiscovered by Home Assistant.

```
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[14:23:00]\x80\00x\x80\00\x80\00x\xf8\00x\x80x\x80\xf8x\00\x80x\x80x\xf8x\xf8\x80\00\x80\xc10,hd_drv:0x00,wp_drv:0x00
[14:23:00]mode:DIO, clock div:2
[14:23:00]load:0x3fff0030,len:1184
[14:23:00]load:0x40078000,len:13132
[14:23:00]load:0x40080400,len:3036
[14:23:00]entry 0x400805e4
[14:23:01][I][logger:158]: Log initialized
[14:23:01][C][ota:483]: There have been 0 suspected unsuccessful boot attempts.
[14:23:01][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:23:01][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[14:23:01][I][app:029]: Running through setup()...
[14:23:01][C][uart.arduino_esp32:077]: Setting up UART...
[14:23:01][C][ld2420:114]: Setting up LD2420...
[14:23:01][D][ld2420:566]: Sending set config enable command: FF
[14:23:01][D][ld2420:448]: LD2420 reply - set config enable: CMD = FF success
[14:23:01][D][ld2420:647]: Sending read gate min max and timeout command:  8
[14:23:01][D][ld2420:448]: LD2420 reply - set config enable: CMD = FF success
[14:23:01][D][number:012]: 'Detection Presence Timeout': Sending state 0.000000
[14:23:01][D][number:012]: 'Select Gate to Set': Sending state 0.000000
[14:23:01][D][number:012]: 'Detection Gate Minimum': Sending state 0.000000
[14:23:01][D][number:012]: 'Detection Gate Maximum': Sending state 0.000000
[14:23:01][D][number:012]: 'Set Still Threshold Value': Sending state 0.000000
[14:23:01][D][number:012]: 'Set Move Threshold Value': Sending state 0.000000
[14:23:01][D][ld2420:682]: Sending read firmware version command:  0
[14:23:01][D][ld2420:488]: LD2420 reply - module firmware version:  v0.0.0 success
[14:23:01][D][text_sensor:064]: 'LD2420 Firmware': Sending state 'v0.0.0'
[14:23:01][D][ld2420:622]: Sending read gate 0 high/low theshold command:  8
[14:23:04][D][sensor:094]: 'Moving Distance': Sending state 70.00000 cm with 0 decimals of accuracy
[14:23:04][D][binary_sensor:036]: 'Presence': Sending state ON
[14:23:05][D][ld2420:448]: LD2420 reply - set config enable: CMD = FF success
[14:23:05][D][ld2420:622]: Sending read gate 1 high/low theshold command:  8
[14:23:06]E (10402) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[14:23:06]E (10402) task_wdt:  - loopTask (CPU 1)
[14:23:06]E (10402) task_wdt: Tasks currently running:
[14:23:06]E (10402) task_wdt: CPU 0: IDLE
[14:23:06]E (10402) task_wdt: CPU 1: loopTask
[14:23:06]E (10402) task_wdt: Aborting.
[14:23:06]
[14:23:06]abort() was called at PC 0x400fa1dc on core 0
[14:23:06]
[14:23:06]
[14:23:06]Backtrace:0x400837cd:0x3ffbe9bc |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400837cd: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[14:23:06]
[14:23:06]
[14:23:06]
[14:23:06]
[14:23:06]ELF file SHA256: 0000000000000000
[14:23:06]
[14:23:06]Rebooting...
```

Above is the first instance of the ESP trying to communicate with the LD2420 right after flashing. It does this about 9 times before it boots into safe mode.

I've hooked both UART0 (programming/logger) and UART1(LD2420) up to a logic analyzer and can actually see the LD2420 response traffic on initial programming and if I power cycle the ESP after attempting to program. For some reason I can't seem to get the logger traffic to appear on a Serial monitor that isn't the one provided by ESPHome. 
For example : `tio -b 115200 /dev/ttyUSB0` just results in an empty monitor when I restart the ESP.

This is the `yaml` I am using to program the ESP:
```
esphome:
  name: office-presence
  friendly_name: Office_Presence

esp32:
  board: esp32dev

# Enable logging - Uses UART0 
logger:
  level: VERY_VERBOSE

api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "redacted"
    password: "redacted"

captive_portal:

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    # See "Number" section below for detail
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

button:
  - platform: ld2420
    apply_config:
      name: Apply Config
    factory_reset:
      name: Factory Reset
    restart_module:
      name: Restart Module
    revert_config:
      name: Undo Edits
```

It is the boiler plate given on the ESPHome page for the LD2420.

GelidusResearch, 2024-07-11:
Are you still experiencing issues on this issue?

paulgiordano, 2025-06-29:
I am experiencing the same issue. here's my yaml and logs

[workshop_logs.txt](https://github.com/user-attachments/files/20968215/workshop_logs.txt)

[workshop.yaml.txt](https://github.com/user-attachments/files/20968269/workshop.yaml.txt)

GelidusResearch, 2025-06-29:
> I am experiencing the same issue. well, not quite the same. I'm running on a generic esp32c3 mini. Here's my yaml and logs, it looks very similar to what's mentioned here.
> 
> [workshop_logs.txt](https://github.com/user-attachments/files/20968215/workshop_logs.txt)
> 
> [workshop.yaml.txt](https://github.com/user-attachments/files/20968269/workshop.yaml.txt)

These are not related issues, the esp32 vs esp32-c3 are very different.
You need to capture the serial esp32-c3 exception log for any possible support on this issue.
Also please redact the pw in your posted YAML file.