esphome/issues #5707: Tuya climate compilation error

bearpawmaxim, 2024-04-17:
### The problem

I'm getting the following compilation error when trying to update my ME-81H thermostat:
`src/main.cpp: In function 'void setup()':
src/main.cpp:360:41: error: invalid conversion from 'const char*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]
  360 |   tuya_tuyaclimate->set_active_state_id("datapoint");
      |                                         ^~~~~~~~~~~
      |                                         |
      |                                         const char*
In file included from src/esphome.h:54,
                 from src/main.cpp:3:
src/esphome/components/tuya/climate/tuya_climate.h:18:36: note:   initializing argument 1 of 'void esphome::tuya::TuyaClimate::set_active_state_id(uint8_t)'
   18 |   void set_active_state_id(uint8_t state_id) { this->active_state_id_ = state_id; }
      |                            ~~~~~~~~^~~~~~~~`

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP8266

### Component causing the issue

tuya climate

### Example YAML snippet

```yaml
substitutions:
  device_name: warm-floor

packages:
  device_common: !include .device_common.yaml

esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m

logger:
  baud_rate: 0

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:
  time_id: sntp_time
  status_pin:
    number: 14
    inverted: true

climate:
  - platform: tuya
    name: "Warm floor"
    switch_datapoint: 1
    target_temperature_datapoint: 16
    target_temperature_multiplier: 1.0
    current_temperature_datapoint: 24
    current_temperature_multiplier: 0.1
    active_state:
      datapoint: 36
      heating_value: 0

switch:
  - platform: "tuya"
    name: "Warm floor lock"
    switch_datapoint: 40

select:
  - platform: "tuya"
    name: "Warm floor sensor selection"
    enum_datapoint: 43
    options:
      0: Internal
      1: Floor
      2: Both
```

ssieb, 2024-04-17:
This was caused by https://github.com/esphome/esphome/pull/5159.  Clearly we don't have any tests for tuya climate.
https://github.com/esphome/esphome/blob/dev/esphome/components/tuya/climate/__init__.py#L211

closed 2024-04-17 by jesserockz

moriahmorgan, 2024-04-19:
I want to apologize to @ssieb @jesserockz and @zry98. Apparently I made the changes included in https://github.com/esphome/esphome/pull/6568 but I never uploaded them. 

Thanks @zry98 for helping.